bb-qq/aqc111 #1: only 1000Mbps link

Black6spdZ, 2019-10-06:
cannot achieve over 1Gbps Link speed. Verified 5Gbps connectivity on PC to PC using QNA-UC5G1T connected to Aquantia AQtion 10G Pro (AQN-107).

On DS918+:
root@TerraServer:~# ethtool eth2
Settings for eth2:
        Supported ports: [ TP MII ]
        Supported link modes:   100baseT/Full
                                1000baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: g
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
 

root@TerraServer:~# ethtool -s eth2 autoneg off speed 5000
Cannot set new settings: Invalid argument
  not setting speed
  not setting autoneg

root@TerraServer:~# ethtool -s eth2 speed 5000
Cannot advertise speed 5000

root@TerraServer:~# ethtool --version
ethtool version 3.13

bb-qq, 2019-10-06:
This driver does not support to disable auto-negotiation. (Above commands fail in my environment as well.)

So to disable auto-negotiation, we need to configure the PC or the switch.

````
# ethtool eth2
Settings for eth2:
        Supported ports: [ TP MII ]
        Supported link modes:   100baseT/Full
                                1000baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 5000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: g
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes

````

Black6spdZ, 2019-10-06:
I was able to disable auto negotiation but setting speed 2500 or 5000 says invalid command. If I forced 2.5 or 5Gbps on the PC the link just stays disconnected

bb-qq, 2019-10-08:
Hmm, it might be an issue of cable quality. NbaseT should work with Cat.5e cables, but using Cat.6e is ideal.

bb-qq, 2019-10-10:
Also, can you establish superspeed for USB? Please try lsusb command to show that.

Black6spdZ, 2019-10-10:
I used a 50ft cat5e cable between aq-107 pcie and aq111u adapter ang got 5gb link on PC.. But same cable on the syno has with aq111u adapter only gets 1gb link.. Ill have a new cat7 cable to test with tomorrow

Black6spdZ, 2019-10-10:
bb-qq, you got me wondering about if the adapter was only running at USB2 speeds and thus limitting link speed to 1Gb, lsusb proves just that:
1c04:0015:0101 00  2.10  480MBit/s 500mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 97I05619)
it is indeed plugged into a USB 3.0 port, why would the USB be falling back to USB2 speed?

closed 2019-10-10 by Black6spdZ

bb-qq, 2019-10-10:
Indeed, since It was confirmed that there was no problem with the PCs, cable seemed unlikely to be a problem...

Please paste ``demsg`` and `l`susb``  outputs if the issue is still occurred.