bb-qq/aqc111 #2: Support for DS1819+

Siutat, 2019-10-23:
Hi bb-qq,

Could you please make a version support for DS1819+? Million thanks.

bb-qq, 2019-10-25:
Please try [this package](https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-1/aqc111-denverton-1.3.3.0-1.spk)

Siutat, 2019-10-25:
installed the package, however cannot the see the additional Ethernet port :(

Siutat, 2019-10-25:
some output of dmsg

[   49.574999] usbcore: registered new interface driver aqc111
[   49.587151] aqc111 2-3:1.0 eth4: register 'aqc111' at usb-0000:00:15.0-3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:xx:xx:xx:xx:xx
[   50.133638] init: synoscheduler-vmtouch main process (4716) killed by TERM signal
[   50.577337] ip_tables: (C) 2000-2006 Netfilter Core Team
[   50.833544] init: synoscheduler-vmtouch main process (17136) killed by TERM signal
[   51.320444] init: nmbd main process (17631) killed by TERM signal
[   54.607255] ip_tables: (C) 2000-2006 Netfilter Core Team
[   55.222735] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   55.470587] aufs 4.4-20160328
[   55.566879] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[   55.624427] Bridge firewalling registered
[   55.706073] workqueue: max_active 1024 requested for vhost_scsi is out of range, clamping between 1 and 512
[   55.721222] IPVS: Registered protocols (TCP, UDP, SCTP, AH, ESP)
[   55.727539] IPVS: Connection hash table configured (size=4096, memory=64Kbytes)
[   55.735184] IPVS: Creating netns size=2096 id=0
[   55.739960] IPVS: ipvs loaded.
[   55.765676] IPVS: [rr] scheduler registered.
[   57.735728] init: upnpd main process (10990) killed by KILL signal
[   58.261577] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   68.067334] init: synocontentextractd main process ended, respawning
[   68.598576] init: iscsi_pluginserverd main process (22051) killed by TERM signal
[   68.645772] init: iscsi_pluginengined main process (21998) killed by TERM signal
[   72.517149] init: iscsi_pluginserverd main process (22299) terminated with status 255
[   72.526031] init: iscsi_pluginserverd main process ended, respawning
[  114.353155] usb 2-3: USB disconnect, device number 3
[  114.358351] aqc111 2-3:1.0 eth4: unregister 'aqc111' usb-0000:00:15.0-3, QNAP QNA-UC5G1T USB to 5GbE Adapter
[  114.372120] aqc111 2-3:1.0 eth4 (unregistered): Failed to write(0x1) reg index 0x0002: -19
[  114.380529] aqc111 2-3:1.0 eth4 (unregistered): Failed to write(0x1) reg index 0x0002: -19
[  114.389366] aqc111 2-3:1.0 eth4 (unregistered): Failed to write(0x61) reg index 0x0000: -19
[  124.386583] usb 2-2.2: new SuperSpeed USB device number 5 using xhci_hcd
[  124.418744] aqc111 2-2.2:1.0 eth4: register 'aqc111' at usb-0000:00:15.0-2.2, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:xx:xx:xx:xx:xx
[  149.709067] usbcore: deregistering interface driver aqc111
[  149.714726] aqc111 2-2.2:1.0 eth4: unregister 'aqc111' usb-0000:00:15.0-2.2, QNAP QNA-UC5G1T USB to 5GbE Adapter
[  180.321737] aqc111 2-2.2:1.0 eth4: register 'aqc111' at usb-0000:00:15.0-2.2, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:xx:xx:xx:xx:xx
[  180.334368] usbcore: registered new interface driver aqc111

Siutat, 2019-10-25:
output of lsusb:

```
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008991DF81C640)
  |__1-2         0bda:5411:0104 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub syno.ext.hub) hub
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:0411:0104 09  3.00 5000MBit/s 0mA 1IF  (Generic 4-Port USB 3.0 Hub syno.ext.hub) hub
    |__2-2.1     0080:a001:0203 00  3.00 5000MBit/s 896mA 1IF  (TOSHIBA External USB 3.0 201503310018E)
    |__2-2.2     1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 94I08396)
```

bb-qq, 2019-10-25:
Hmm... please try:

- disconnect the usb hard drive
- change usb port for the dongle

Siutat, 2019-10-25:
thx

just worked, but the speed not up to 5Gbps

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   181 MBytes  1.51 Gbits/sec
[  4]   1.00-2.00   sec   188 MBytes  1.58 Gbits/sec
[  4]   2.00-3.00   sec   185 MBytes  1.55 Gbits/sec
[  4]   3.00-4.00   sec   194 MBytes  1.63 Gbits/sec
[  4]   4.00-5.00   sec   197 MBytes  1.65 Gbits/sec
[  4]   5.00-6.00   sec   197 MBytes  1.65 Gbits/sec
[  4]   6.00-7.00   sec   186 MBytes  1.56 Gbits/sec
[  4]   7.00-8.00   sec   193 MBytes  1.62 Gbits/sec
[  4]   8.00-9.00   sec   190 MBytes  1.60 Gbits/sec
[  4]   9.00-10.00  sec   190 MBytes  1.59 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.86 GBytes  1.59 Gbits/sec                  sender
[  4]   0.00-10.00  sec  1.86 GBytes  1.59 Gbits/sec                  receiver

iperf Done.

bb-qq, 2019-10-26:
> just worked,

Which solution you could resolve the issue? I suspected it caused by power supply in previous comment.

> but the speed not up to 5Gbps

You must enable jumbo frame for the performance described in the readme. 
And two concurrent sessions are needed, which can be specified by -P 2 option in iperf.

closed 2019-11-01 by Siutat