bb-qq/aqc111 #4: DS916+への対応について

toku565, 2019-11-30:
bb-qq様

DS916+(Braswell)、DSM 6.2.2-24922 Update 4にて
aqc111-braswell-1.3.3.0-1.spkを適用したところ、
1000Mbpsリンクアップとなりました。

また、一定時間経つとリンクアップにも関わらず通信が切れるようです。
ifconfig eth2 down
ifconfig eth2 up
で復旧はしますが…。

対向はIntel X540-T2 @Windowsです。

何かアドバイス等ありましたらどうぞよろしくお願い致します。

bb-qq, 2019-12-01:
X540-T2は5Gbps/2.5Gbpsに対応していない気がします。Windowsであればデバイスマネージャからリンク速度を変更できるので一度確認してみてください。
https://ark.intel.com/content/www/jp/ja/ark/products/58954/intel-ethernet-converged-network-adapter-x540-t2.html

またリンクが切れてしまうとのことですが、他にUSB機器を使ってないでしょうか？使ってないのでしたら接続するUSBポートも変えて見てください。

toku565, 2019-12-01:
ご返信ありがとうございます。
仰る通りで、X540-T2では5GBASE-Tに設定できませんでした。
10GBASE-T対応していれば自動的に5GBASE-T、2.5GBASE-Tも対応していると勘違いしておりました。

5GBASE-TのNICに接続したところ、5Gbpsでアップ致しました。
（Aquantia Aqtion 10G Pro Nic）
ありがとうございます。

しばらく使って様子見してみます。

closed 2019-12-01 by toku565