bb-qq/aqc111 #7: On DS716+

Balooforever, 2020-01-24:
Hi,

The drivers work, I can ping or use SMB but after little test (SMB copy ..), the adapter disconnect and can t ping or move into smb folder

Same problem with the 2,5Gb adapter.

I set it by default. How I can get the log for debug ?

Thank a lot :)

Balooforever, 2020-01-24:
I have find log :

> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.420451] aqc111 4-2:1.0 eth2: Failed to read(0x1) reg index 0x0002: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.428285] aqc111 4-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.436427] aqc111 4-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.444356] aqc111 4-2:1.0 eth2: Failed to write(0x61) reg index 0x0000: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.465115] aqc111 4-2:1.0 (unregistered net_device): Failed to write(0x1) reg index 0x0002: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.475076] aqc111 4-2:1.0 (unregistered net_device): Failed to write(0x1) reg index 0x0002: -19
> 2020-01-24T11:34:45+01:00 DSM kernel: [407354.485140] aqc111 4-2:1.0 (unregistered net_device): Failed to write(0x61) reg index 0x0000: -19

bb-qq, 2020-01-26:
Cloud you fill out these questions?

---

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7) and straight / cross-link

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

Balooforever, 2020-01-26:
Hi,

Description of your products
DS716+ II
Linux DSM 3.10.105 #24922 SMP Fri May 10 02:50:17 CST 2019 x86_64 GNU/Linux synology_braswell_716+II
DSM version 6.2.2
Qnap-UC5G1T

Description of your environment
Using a hub : Netgear SX10 Pro (same problem with Netgear EM110x)
Cat.6

Output of dmesg command
[93125.221109] mii: module verification failed: signature and/or required key missing - tainting kernel
[93125.242020] usbcore: registered new interface driver aqc111
[93125.261288] aqc111 2-2:1.0 eth2: register 'aqc111' at usb-0000:00:14.0-2, QNAP QNA-UC5G1T USB to 5GbE Adapter, xxxxxxxxxx
[93128.068021] init: winbindd main process (19861) killed by TERM signal
[93129.884110] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[93137.466082] aqc111 2-2:1.0 eth2: Link Speed 5000, USB 3
[93137.478205] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[93139.244016] init: winbindd main process (14703) killed by TERM signal
[93140.992512] init: iscsi_pluginserverd main process (15663) killed by TERM signal
[93141.006025] init: iscsi_pluginengined main process (15653) killed by TERM signal
[93141.414839] init: winbindd main process (15956) killed by TERM signal
[93154.512527] init: winbindd main process (16376) killed by TERM signal

Output of lsusb command
lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation xxxxxxxxxxx)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 99I09255)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub

Output of ifconfig -a command

bond0     Link encap:Ethernet  HWaddr xxxxx
          inet addr:192.168.x.xxx  Bcast:192.168.x.255  Mask:255.255.255.0
          inet6 addr: fe80::xxx:xxx:xxxx:9e92/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:221101545 errors:0 dropped:2698 overruns:2698 frame:0
          TX packets:260626448 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:210008591071 (195.5 GiB)  TX bytes:252130627122 (234.8 GiB)

docker0   Link encap:Ethernet  HWaddr xxxxx
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::xxx:xxx:xxxx:c69e/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:328 (328.0 B)

docker-21 Link encap:Ethernet  HWaddr xxxxx
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          inet6 addr: fe80::xxx:xxx:xxxx:2bdc/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:211701 errors:0 dropped:0 overruns:0 frame:0
          TX packets:234804 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:29212950 (27.8 MiB)  TX bytes:70112609 (66.8 MiB)

docker03f Link encap:Ethernet  HWaddr xxxxx 
          inet6 addr: fe80::xxx:xxx:xxxx:ba1e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:176 errors:0 dropped:0 overruns:0 frame:0
          TX packets:230 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
[... 87 lines cut ...]

eth2      Link encap:Ethernet  HWaddr xxxxx
          inet addr:192.168.x.yy  Bcast:192.168.x.255  Mask:255.255.255.0
          inet6 addr: fe80::xxx:xxx:xxxx:e164/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1114 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:231140 (225.7 KiB)  TX bytes:51896 (50.6 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:611082 errors:0 dropped:0 overruns:0 frame:0
          TX packets:611082 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:110797223 (105.6 MiB)  TX bytes:110797223 (105.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

Balooforever, 2020-01-26:
Hi,

After nab ling "OpenVswitch", all work but not very good performance.
70Mb write (110+ with native eth) and 240 Read (a test on a un SSD MX500).
I don t understand why slow write speed ..

bb-qq, 2020-01-26:
Please try increasing MTU to 9k, and measuring transfer speed using iperf.

Balooforever, 2020-01-26:
And again disconnect after little transfert..
I'm disappointing ..

bb-qq, 2020-01-26:
I'm sorry to hear that...

I think it is good to confirm the behavior with a PC connecting the same cable and the same switch. I used to find a cable issue by this method when I setup DS918+.

Balooforever, 2020-01-26:
Same problem with another cable. No problem with my computer Mac and another Windows machine ..

bb-qq, 2020-01-27:
Probably this is a driver issue, but I don't have any idea about its cause... My DS918(apollolake) is not much different from DS716(braswell) and other braswell based product users did not claim about stability issues so far. (eg. A user of issue #4 uses DS916, but it seems to go well) 

How about with 1Gigabit connection? 

> Using a hub : (same problem with Netgear EM110x)

Is this a MultiGiga switch? (I Googled this product, but I could not find it)

Balooforever, 2020-01-27:
Hi,
It s a GS110MX (my bad :D)
With 1Gb, same problem .. After transfert, i can t ping and copy stop working :(
He work after 5 minutes and block again..

bb-qq, 2020-01-31:
> EDIT : The LACP Bond can be the faulty ?

I believe that LACP can not cause linking down of interfaces. If something of LACP is bad, it will just stop a transfer.

By the way, do you have a self-powered USB hub? If you have it, please try to connect the dongle via that hub with power supply. The last cause I come up with is a power shortage, but this possibility is very low...

Balooforever, 2020-01-31:
I don t have, I try to find it for test. Anyone have this issue ? :(

bb-qq, 2020-02-01:
No, I just guessed from the situation, which disconnection is occurred in L2 layer when load is high.
So it might not be worth the trouble of buying a new one.

```
[93129.884110] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[93137.466082] aqc111 2-2:1.0 eth2: Link Speed 5000, USB 3
[93137.478205] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
```

gaarou, 2020-02-07:
Hello,
First I want to thank you for your work!
I have exactly the same problem with DS1815 + with DSM 6.2.2.24922 update 4.

For information I also had the same problem with this adapter on a computer in Win7 after switching to W10 everything worked.

Between the NAS and the computer I have a Netgear GS110EMX Switch.

Thanks for your help

HomerJau, 2020-03-14:
I have same problem with DS2415+ with DSM 6.2.2.24922 update 4 using a Netgear MS510TX switch with new QNAP UC5G1T.

Works fine for a few seconds under load (moving a  large file) then stops and does not respond. Eventually it comes back, but unusable like this.

Another Synology NAS (DS1618+) with internal Synology 10GBE card with same switch no problem.

HomerJau, 2020-03-15:
Hi bb-qq.

I just tried a new QNAP UC5G1T on a DS1618+ using the Denverton driver and it works flawlessly (I copied a 70GB file without a bandwidth glitch).

Same QNAP on a DS2415+ using the Avoton driver fails after an initial high rate transfer rate (as previous posts here, dies after 2 or 3 seconds). Both NAS have same DSM version. So it does look like the Denverton driver is the issue.

If you need more info let me know, although I'm no network expert...

bb-qq, 2020-03-16:
Thank you for the information.

> If you need more info let me know,

I would really appreciate it if you could try a self-powered USB hub to isolate power supply problems.

I can't come up with other ideas to resolve this issue, because these products have very similar hardware configurations and exactly the same kernel/driver version.

work?|Product name | Platform    | SoC           | CPU architecture
-----|-------------|-------------|---------------|----------
 OK  |DS1618+      | Denverton   | Atom C3538    | Goldmont 
 OK  |DS918+       | Apollolake  | Celeron J3455 | Goldmont
 NG  |DS1815+      | Avoton      | Atom C2538    | Silvermont
 NG  |DS2415+      | Avoton      | Atom C2538    | Silvermont
 NG  |DS716+       | Braswell    | Celeron N3150 | Airmont

HomerJau, 2020-03-17:
Thanks bb-qq. You may be right about the USB power on the older Synology models. 

I don't have a powered USB hub but I just ordered one to try on the DS2415+. I won't get it for a couple of days but I'll report here as soon as I have test results.

Thanks!

HomerJau, 2020-03-19:
Bad news. I tried a powered USB HUB and it did not work.

Worse is the QNAP UC5G1T doesn't work at all in anything after I tried it in a USB powered hub. Look like its fried/faulty. It was working a couple days ago when I tried it on my DS1618+ so I assume the powered hub caused the problem.

The QNAP is no longer recognized by either NAS without the hub. My switch ports don't blink at the other end...

I think I'll give up on this now. but will keep my eye on this for any other input from users.

The hub is an Orico F4U-U3 with an external 5V USB supply.

bb-qq, 2020-03-19:
Thank you for reporting the result...

> The QNAP is no longer recognized by either NAS without the hub. My switch ports don't blink at the other end.

Can you use the dongle with Windows PC, or it was completely broken? I am suspecting that the driver of  the host controller of the NAS became invalid state by using the USB hub,

bb-qq, 2020-03-19:
To all, could you provide results of `sudo lspci -v` regarding the above devices? It may be caused by the difference in the USB host controller chip.

HomerJau, 2020-03-19:
Latest info:

The QNAP works ok on my Win10 PC (phew!)
The QNAP works ok on my DS1618+ after a NAS restart
The QNAP is not recognized by the DS1618+ when I connect the hub

I didn't try restarting the DS2415+ as its doing a rebuild. I'll do that later.

I ran sudo lspci -v on the DS1618+:
1. With nothing attached
2. With the Hub connected directly to a USB 3.0 port (no QNAP)
3. With the Hub connected directly to a USB 3.0 port and QNAP connected to HUb

See attached text file:
[SynNAS and Hub.txt](https://github.com/bb-qq/aqc111/files/4357362/SynNAS.and.Hub.txt)

HomerJau, 2020-03-20:
I've ordered a new powered hub that works on a Synology NAS (based on user feedback). Its uses a Realtek chip (different to the Orico). It could take a week to get here based on ebay's estimates.

bb-qq, 2020-03-21:
I'm glad the device is not broken, and thank you for the providing output of lspci. What is the model of it?

HomerJau, 2020-03-25:
I posted about this issue on another forum and was recommended to try a USB 3.0 Y cable. That should arrive tomorrow. Basically its a USB 3.0 extension cable with another USB lead to inject 5V power. The NAS should only see the QNAP adapter. Apparently this has worked for other Linux devices, so fingers crossed.

The 2nd USB 3.0 Hub (Realtek chipset) is due to arrive on 1st April.

@bb-qq You asked: What model? Are you asking about the 2nd USB Hub?

bb-qq, 2020-03-28:
> You asked: What model? Are you asking about the 2nd USB Hub?

Oh, sorry, I missed that you wrote the product name of txt files. Please ignore that question.

Balooforever, 2020-04-29:
It's ok for me when i use the front port !

bb-qq, 2020-05-22:
@Balooforever , thank you for sharing the information.
@HomerJau , what was your result?

jackhudson2001, 2020-07-20:
hi ive also got a ds1618+ which version of drivers works?

bb-qq, 2022-08-11:
Since the original author of this issue mentioned that his issue was resolved by using the front port, I close this issue. Create a new issue if something remains.

closed 2022-08-11 by bb-qq