bb-qq/aqc111 #8: Driver for DS2415+

rabbit1028, 2020-01-27:
Hi,

I want to use the driver on DS2415+, could you please help on this?

NAS mode: Synology DS2415+
DSM version: 6.2.2-24922 Update 4
ethernet dongle:  QNAP QNA-UC5G1T

Thank you!

bb-qq, 2020-01-31:
This driver should work with DS2415+.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-avoton-1.3.3.0-2.spk

bb-qq, 2020-03-21:
I would like to know whether the driver works for your DS2415+ because some users reported stability issues of this model recently.

rabbit1028, 2020-03-23:
I didn't use it for a long time. I tried on my both DS218+ and DS2514+, I was getting up to 300MB/s on DS218+ and only 200MB/s on DS2415+. So I rolled back to use 4x 1GB Multichannel.

bb-qq, 2020-03-23:
Thank you for the information.

closed 2020-03-23 by bb-qq