bb-qq/aqc111 #11: USB Ethernet not appearing / disappears a few mins after boot.

scyto, 2020-03-20:
### Description of the problem

Installed apollo lake variant on my DS918+
Eth2 (3rd ethernet port never showed up)
dmesg showed device detected
rebooted - eth3 was there, i could configure IP but a few minutes later it had disappeared
I see this in the  messages

```2020-03-20T08:20:33-07:00 SYN02 synoscgi_SYNO.Core.Network.Ethernet_1_set[16801]: net_lnx_set_ip_netmask.c:94 unable to stop dhcp-client instance on eth2
2020-03-20T08:20:34-07:00 SYN02 ipv4_change hook event: eth2 none->192.168.23.1 
2020-03-20T08:20:34-07:00 SYN02 gateway_change hook event: NEW 0.0.0.0 on eth2 
2020-03-20T08:20:34-07:00 SYN02 gateway_change hook event: DEL 192.168.1.1 on ovs_eth0 
2020-03-20T08:20:34-07:00 SYN02 gateway_change hook event: DEL 0.0.0.0 on eth2 
2020-03-20T08:20:34-07:00 SYN02 gateway_change hook event: NEW 192.168.1.1 on ovs_eth0 
2020-03-20T08:20:34-07:00 SYN02 [  732.326171] init: winbindd main process (14211) killed by TERM signal
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.048595] aqc111 2-2:1.0 eth2: Failed to read(0x1) reg index 0x0002: -19
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.056287] aqc111 2-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.064076] aqc111 2-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.071899] aqc111 2-2:1.0 eth2: Failed to write(0x61) reg index 0x0000: -19
2020-03-20T08:22:33-07:00 SYN02 synonetd: hook/ip_change_event.cpp:58 error on link detect
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.088664] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x1) reg index 0x0002: -19
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.097912] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x1) reg index 0x0002: -19
2020-03-20T08:22:33-07:00 SYN02 kernel: [  851.107155] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x61) reg index 0x0000: -19
2020-03-20T08:29:02-07:00 SYN02 kernel: [ 1239.985491] xhci_hcd 0000:00:15.0: port 1 resume PLC timeout```

### Description of your products

> DS918+
> Linux SYN02 4.4.59+ #24922 SMP PREEMPT Thu Apr 25 14:29:34 CST 2019 x86_64 GNU/Linux synology_apollolake_918+
> DSM 6.2.2-24922 Update 4
> TRENDnet TUC-ET5G  (connected using USB C to A adapter)

### Description of your environment

> - connect directly to a PC
>     - PC: another TRENDnet TUC-ET5G connected to USBC port
>     - Hub: model N/A
> - cable category Cat.6 

### Output of ``dmesg`` command
This output was from where i plug it back in and out, i also saw this during the initial test.

````
[  851.031627] usb 2-2: USB disconnect, device number 2
[  851.037287] aqc111 2-2:1.0 eth2: unregister 'aqc111' usb-0000:00:15.0-2, USB-C 3.1 to 5GBASE-T Ethernet Adapter
[  851.048595] aqc111 2-2:1.0 eth2: Failed to read(0x1) reg index 0x0002: -19
[  851.056287] aqc111 2-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
[  851.064076] aqc111 2-2:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
[  851.071899] aqc111 2-2:1.0 eth2: Failed to write(0x61) reg index 0x0000: -19
[  851.088664] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x1) reg index 0x0002: -19
[  851.097912] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x1) reg index 0x0002: -19
[  851.107155] aqc111 2-2:1.0 eth2 (unregistered): Failed to write(0x61) reg index 0x0000: -19
[ 1009.205016] usbcore: deregistering interface driver aqc111
[ 1013.438852] usbcore: registered new interface driver aqc111
[ 1239.985491] xhci_hcd 0000:00:15.0: port 1 resume PLC timeout
[ 1253.110490] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
[ 1253.140489] aqc111 2-2:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-2, USB-C 3.1 to 5GBASE-T Ethernet Adapter, 3c:8c:f8:f9:60:16
````

### Output of ``lsusb`` command

````
alexb@SYN02:/var/log$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008C48A8DA5A85)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         20f4:e05a:0101 00  3.20 5000MBit/s 896mA 1IF  (TRENDnet TRENDnet USB 5G Adapter 000000BF)
````

### Output of ``ifconfig -a`` command

````
alexb@SYN02:/var/log$ ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:E1:65:F1:DE  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:e1ff:fe65:f1de/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:82964 errors:0 dropped:0 overruns:0 frame:0
          TX packets:77659 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:609167530 (580.9 MiB)  TX bytes:758678951 (723.5 MiB)

docker1c6 Link encap:Ethernet  HWaddr 4E:82:75:12:76:6A  
          inet6 addr: fe80::4c82:75ff:fe12:766a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2614 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2394 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:199471 (194.7 KiB)  TX bytes:631186 (616.3 KiB)

docker336 Link encap:Ethernet  HWaddr D2:06:73:F3:EC:0B  
          inet6 addr: fe80::d006:73ff:fef3:ec0b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:80350 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75282 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:610129555 (581.8 MiB)  TX bytes:758049103 (722.9 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:A1:CA:DD  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:709137 errors:0 dropped:0 overruns:0 frame:0
          TX packets:488977 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:964256961 (919.5 MiB)  TX bytes:618720951 (590.0 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:A1:CA:DE  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 3C:8C:F8:F9:60:16  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8342 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8342 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1042895 (1018.4 KiB)  TX bytes:1042895 (1018.4 KiB)

ovs-syste Link encap:Ethernet  HWaddr 56:5F:89:62:65:3C  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:A1:CA:DD  
          inet addr:192.168.1.38  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:167456 errors:0 dropped:0 overruns:0 frame:0
          TX packets:97443 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:932653965 (889.4 MiB)  TX bytes:597578139 (569.8 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:A1:CA:DE  
          inet addr:192.168.1.39  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:362 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:56930 (55.5 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

scyto, 2020-03-20:
i have a QNAP QNA-UC5G1T  on order, but given amazon shipping at moment it isn't expected to arrive for a week - this will help identify if it is the usbc>a adapter or the trend net device itself.

scyto, 2020-03-20:
i rebooted to see if it was because i originally booted with out eth cable plugged in
it worked for a minute but after that i could no longer ping between the PC the DS918+

scyto, 2020-03-24:
I think this is looking more like it was the USBC > USBA adapters.  I have one iof these units working on my DS1815+ (HTML speed test reports 2gbps down but only 5Gbps up - but that is different issue so i will close as resolved for now.)

closed 2020-03-24 by scyto

suddy, 2020-04-01:
> i rebooted to see if it was because i originally booted with out eth cable plugged in
> it worked for a minute but after that i could no longer ping between the PC the DS918+
> I don't see the same logged items in messages or dmesg i posted above
> and it hasn't yet disappeared from the the UI
> 
> i guess wait for the QNAP one to arrive before spending any time on this,

I have the same error.
I had two QNAP one use typec to typec connect to my MacBook Pro 2018, the other is connect to ds918+ and connect with one line direct.
When I boot ds918+, I can ping from my MacBook
 Pro, but few minute later, I cannot ping between my MacBook Pro and ds918+
And  when I change ipaddr or netmask on ds918+(for this eth2) sometime it work for a minute, sometimes not. I had reinstall DS918+ ‘s system, but nothing change

scyto, 2020-04-01:
@suddy hmm sounds exactly like what i saw, i need to do some more longhaul testing

bb-qq, 2020-04-08:
Hi @suddy , could you create a new issue and attach the required information if you still have the problem.

avielc, 2022-05-14:
Kinda reviving the old topic.  - sorry for that.
I was wondering if you guys found the cause as I want to take a dive here and try to upgrade the nas to 2.5gbps (router has a supported port. and wifi is 2.5gbps in speed) 
I am thinking of getting the Trend ET5G only because it's under the  tax enabled rate for delivery. 

Whenever you can reply back, i'd appreciate it. 
Thanks!