bb-qq/aqc111 #13: No Hot Plug of adapters?

scyto, 2020-04-22:
### Description of the problem

With driver running plugging in a compatible USB adapter doesn't result in a new interface in DSM.
Restarting the driver package resolves the issue.

### Description of your products

Linux SYN01 3.10.105 #24922 SMP Wed Jul 3 16:37:24 CST 2019 x86_64 GNU/Linux synology_avoton_1815+

DSM 6.2.2-24922 Update 4
QNAP QNA-UC5G1T 
TRENDnet TUC-ET5G 

### Description of your environment

plugged into a multigig RJ45 SFP port and USB port 1

### Output of ``dmesg`` command

````
978.908041] init: synowsdiscoveryd main process (16960) killed by TERM signal
[6254578.795447] init: synowsdiscoveryd main process (425) killed by TERM signal
[6263387.797590] init: synoscheduler-vmtouch main process (28554) killed by TERM signal
[6263388.079431] init: synoscheduler-vmtouch main process (24795) killed by TERM signal
[6263408.482183] init: synoscheduler-vmtouch main process (24878) killed by TERM signal
[6263408.700496] init: synoscheduler-vmtouch main process (26099) killed by TERM signal
[6281462.218793] FAT-fs (synoboot2): fat_set_state: set=1 process=synoddsmd pid=17326
[6281462.239414] FAT-fs (synoboot2): fat_set_state: set=0 process=synoddsmd pid=17326
[6281462.365261] FAT-fs (synoboot2): fat_set_state: set=1 process=synoddsmd pid=17326
[6281462.382960] FAT-fs (synoboot2): fat_set_state: set=0 process=synoddsmd pid=17326
[6302334.977547] usbcore: deregistering interface driver aqc111
[6302334.983945] aqc111 3-4:1.0 eth4: unregister 'aqc111' usb-0000:04:00.0-4, QNAP QNA-UC5G1T USB to 5GbE Adapter
[6302354.573371] aqc111 3-4:1.0 eth4: register 'aqc111' at usb-0000:04:00.0-4, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:4d:a6:7f
[6302354.586691] usbcore: registered new interface driver aqc111
[6302355.317235] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[6302362.833089] aqc111 3-4:1.0 eth4: Link Speed 5000, USB 3
[6302362.852894] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302362.852905] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302362.866608] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302362.867666] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[6302365.339236] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302366.706665] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302366.844501] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302366.851766] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302369.281115] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302370.079961] init: winbindd main process (21471) killed by TERM signal
[6302371.618975] iSCSI:iscsi_target.c:520:iscsit_add_np CORE[0] - Added Network Portal: 192.168.1.111:3260 on iSCSI/TCP
[6302377.042452] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302379.366634] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302379.373478] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302379.448110] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302385.543398] init: iscsi_pluginengined main process (20867) killed by KILL signal
[6302386.177931] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.184770] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.191597] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.198421] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.284993] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.291811] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.298625] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302386.305439] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.842533] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.849360] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.856182] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.863002] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.869960] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.876789] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.883608] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.890494] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.895893] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[6302393.906132] aqc111 3-4:1.0 eth4: kevent 2 may have been dropped
[... 315 lines cut ...]

tap021132 Link encap:Ethernet  HWaddr 36:A1:BF:C1:27:D4  
          inet6 addr: fe80::34a1:bfff:fec1:27d4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6099483 errors:0 dropped:0 overruns:0 frame:0
          TX packets:103694538 errors:0 dropped:4581 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2156061164 (2.0 GiB)  TX bytes:15957562327 (14.8 GiB)

vx-001001 Link encap:Ethernet  HWaddr E6:18:A7:AC:E4:D6  
          inet6 addr: fe80::e418:a7ff:feac:e4d6/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1450  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:50884 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

vx-001002 Link encap:Ethernet  HWaddr AE:C1:89:69:08:E8  
          inet6 addr: fe80::acc1:89ff:fe69:8e8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1450  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:50855 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2020-05-22:
Currently, the script in driver package does not support hot plug, but we can enable the interface manually by typing commands such as  `ifconfig eth4 up && ifconfig eth4 inet 192,168.x.x netmask 255.255.255.0`

sbsrouteur, 2021-03-05:
Hello,

Package installed & started. Seems OK. synopkg status reports running status [255].

I did inconfig to up the sit0 interface and affect an IPV4 address.

No led show up on the interface, nor the switch on the other end of the ethernet cable. interface is not listed in DSM control panel.

Tried to unplug/replug the USB device.

Is a reboot required?

inconfig shows a weird encap, not ethernet : 

```
root@DS:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:11:32:32:DD:71
          inet addr:10.32.11.249  Bcast:10.32.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9817068324 errors:0 dropped:95670 overruns:95670 frame:0
          TX packets:14239753703 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6872918182734 (6.2 TiB)  TX bytes:18026618406287 (16.3 TiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:32:DD:72
          inet addr:169.254.80.85  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:32:DD:73
          inet addr:169.254.237.243  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 00:11:32:32:DD:74
          inet addr:169.254.21.235  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:22647551 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22647551 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3335710319 (3.1 GiB)  TX bytes:3335710319 (3.1 GiB)

sit0      Link encap:IPv6-in-IPv4
          inet addr:10.32.11.248  Mask:255.255.0.0
          inet6 addr: ::10.32.11.249/96 Scope:compatGlobal
          inet6 addr: ::10.32.11.248/96 Scope:compatGlobal
          inet6 addr: ::169.254.80.85/96 Scope:compatGlobal
          inet6 addr: ::127.0.0.1/96 Scope:compatHost
          inet6 addr: ::169.254.21.235/96 Scope:compatGlobal
          inet6 addr: ::169.254.237.243/96 Scope:compatGlobal
          UP RUNNING NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

```

bb-qq, 2022-08-11:
The hotplug was added in the latest release as an experimental feature. Try it, please.

closed 2022-08-11 by bb-qq