bb-qq/aqc111 #15: Successful install and first tests DS1819+ 2xQNap

fbwsol, 2020-05-17:
After a little research found your driver, it helped influence my buying choices - new DS1819+ to replace ageing DX1511+ and DX510 combo, 2x QNAP QNA-UC5G1T for direct link workstation-server.
Right off the bat, THANK YOU. After 3 days of benchmarking file systems and network protocols machine is now production, I'm super happy with it, your driver is a crucial part of that.

TL;DR? Working great, maxed it out on large single file test against 6-drive RAID6. A few tiny bugs during install. SMB kills NFS across all tests Win10 - DS.

DS1819+ is Denverton Celeron. Comes with a PCIE slot can be either 10Gb card or M2 cache. Since i'm only interested in improving point to point transfers between workstation and the DS, not providing >2G bond to rest of network, wanted to keep options open for expansion card, so opted to try pair of QNAP's.
Installation went fine, i'm n00b at the prompt so went through manual install in package centre. After reboot new interface appeared OK under CP/Network. At first, Manual MTA showed only 1500 option in dropdown.
Installed other adapter on USB3 port on TB3 dock for Lenovo X1, direct Cat6 cable to DS.
Forced link speed to 5G on workstation end, seems driver is reporting incorrect capabilities to DS system - possible link speeds, maybe range allowed MTU also?:

Output of `ethtool eth4` command
````
Settings for eth4:
        Supported ports: [ TP MII ]
        Supported link modes:   100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 5000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
Cannot get wake-on-lan settings: Operation not permitted
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
````
Forced MTU 9000 with `ip link set eth4 mtu 9000` by ssh at NAS.
After reboot (I'm sure could be done less brutally at shell but again, I'm terminal n00b) the Manual MTU option under Network-Network Interface-Edit shows fully populated list 1500-9000 with everything in between.
Although it's on USB, I'm pretty sure you can't hotplug the adapter, at least for me it misbehaved and I needed a reboot of the DS to get stable connection back. Might well be doable somehow from the shell but beyond my scope, someone more expert with synology can maybe clear that up.

With a stable connection, I set about benchmarking SMB and NFS for transfers to and from Win10 and DS. Note, I had upgraded to 32GB onboard memory - in prior tests I observed up to 30.5GB memory used for cache so my tests were:
1) very large single file (1x64gb zip, 1x28Gb zip) - used to definitely exceed / be just within max cache
2) folder with 28GB of audio files each 3-12MB  (20% of my data)
3) folder with 28GB of video files each 2-7GB  (50% of my data)

Repeatedly tested transfer back and forth of these, first over SMB then over NFS.

### Results: 
Large single file first copy approx 300-350MB/s range, mostly 320MB/s sustained
Second copy of 28GB file (fully cached by DS) 430MB/s sustained (Yes!)

Audio files first copy approx 80-170MB/s range
Cached copy approx 200-280MB/s range

Video files first approx 175-250MB/s range
Cached copy approx 200-300MB/S range

Across all file sizes, NFS only every managed 100-150MB/s range, cached or uncached.
CPU utilisation under NFS 17-24%
CPU utilisation under SMB 40-45%

These are just first benchmarks, happy to run something specific if anyone wants, and have screenshots during all the above tests also if requested... learning more shell every day but please be patient if you need me to run CLI commands, be explicit :)
I'll post a follow-on in a few weeks, by then should have some feedback on stability and real world usage on vid production and machine backups.

Thanks again @bb-qq, domo arigato gozaimasu
 
### Description of your products

> - DS1819+
>   output of ``uname -a`` command
``Linux DS1819 4.4.59+ #24922 SMP PREEMPT Thu Mar 12 13:02:11 CST 2020 x86_64 GNU/Linux synology_denverton_1819+``
> - On-board memory upgraded to 32GB
> - DSM 6.2.2-24922 Update 6
> - QNAP QNA-UC5G1T x2

### Description of your environment

> - direct connect Win10 Lenovo X1 Carbon Tablet 3rd Gen, TB3 Dock, QNAP to QNAP on DS
> - cable Cat.6

### Output of ``ifconfig -a`` command

````
eth4      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet addr:10.0.0.1  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::265e:beff:xxxx:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:25477 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21775 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3277016 (3.1 MiB)  TX bytes:7667712 (7.3 MiB)````

fbwsol, 2020-05-17:
[image]

bb-qq, 2020-05-22:
Thank you for the performance report and I'm looking forward to hearing the stability report.

Please note that some users reported instability issues when they use rear ports.