bb-qq/aqc111 #16: 在DS1815+  DSM 6.2.3-25426安装失败，提示 套件与DSM不兼容

yimengmedia, 2020-05-19:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2020-05-22:
Please fill the above form in English.

bb-qq, 2020-06-27:
Please reopen the issue when you write additional information.

closed 2020-06-27 by bb-qq