bb-qq/aqc111 #17: Active/Passive Bond for Fail safe does not work

Podden, 2020-05-27:
### Description of the problem

I tried to Create an Active/Passive Bond with my QNAP Adapter and one of the internal Network interface for Fail Safe purpose. It says "Applying Network Settings" and reloads DSM Site but the Bond does not get created. Is this even possible with mixed speed Interfaces?

### Description of your products

- DS916+ Linux 3.10.105 #24922 SMP synology_braswell_916+
- DSM 6.2.2-24922 Update 6
- QNAP QNA-UC5G1T USB 3.0 auf 5 GbE Netzwe  

### Description of your environment

> Via Netgear 5Gbe Switch
>     - PC: Asus 10Gbe Card (Aquantia)
>     Netgear XS508M-100 Switch
> Cat.7 Cable

### Output of ``dmesg`` command

````
[   76.737100] MODSIGN: Loaded cert 'Synology Kernel Module Signing Certification Authority: 600839b5d127e0e11d817a31f0575d323a7c0e28'
[   76.750360] MODSIGN: Loaded cert 'Synology kernel module signing key: 4646ce54489669338118a3b1286da156ac366fa5'
[   76.761657] registered taskstats version 1
[   76.766963] rtc_cmos 00:05: setting system clock to 2020-04-12 01:29:22 UTC (1586654962)
[   76.777452] Freeing unused kernel memory: 772k freed
[   76.783269] Write protecting the kernel read-only data: 8192k
[   76.789974] Freeing unused kernel memory: 1272k freed
[   76.798472] Freeing unused kernel memory: 504k freed
[   76.919124] ACPI: bus type USB registered
[   76.924097] usbcore: registered new interface driver usbfs
[   76.930497] usbcore: registered new interface driver hub
[   76.936736] usbcore: registered new interface driver ethub
[   76.943199] usbcore: registered new device driver usb
[   76.989583] xhci_hcd 0000:00:14.0: setting latency timer to 64
[   76.989598] xhci_hcd 0000:00:14.0: xHCI Host Controller
[   76.995476] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 1
[   77.203902] xhci_hcd 0000:00:14.0: cache line size of 64 is not supported
[   77.203944] xhci_hcd 0000:00:14.0: irq 133 for MSI/MSI-X
[   77.204427] xHCI xhci_add_endpoint called for root hub
[   77.204432] xHCI xhci_check_bandwidth called for root hub
[   77.204681] hub 1-0:1.0: USB hub found
[   77.208911] hub 1-0:1.0: 7 ports detected
[   77.214596] xhci_hcd 0000:00:14.0: xHCI Host Controller
[   77.220460] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[   77.229202] xHCI xhci_add_endpoint called for root hub
[   77.229209] xHCI xhci_check_bandwidth called for root hub
[   77.229435] hub 2-0:1.0: USB hub found
[   77.233659] hub 2-0:1.0: 6 ports detected
[   77.280883] etxhci_hcd-170202 0000:04:00.0: Etron xHCI Host Controller
[   77.288204] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 3
[   77.309714] etxhci_hcd-170202 0000:04:00.0: irq 134 for MSI/MSI-X
[   77.310645] xHCI etxhci_add_endpoint called for root hub
[   77.310650] xHCI etxhci_check_bandwidth called for root hub
[   77.310895] ethub 3-0:1.0: USB hub found
[   77.315322] ethub 3-0:1.0: 2 ports detected
[   77.320159] etxhci_hcd-170202 0000:04:00.0: Etron xHCI Host Controller
[   77.327478] etxhci_hcd-170202 0000:04:00.0: new USB bus registered, assigned bus number 4
[   77.337559] xHCI etxhci_add_endpoint called for root hub
[   77.337567] xHCI etxhci_check_bandwidth called for root hub
[   77.337809] ethub 4-0:1.0: USB hub found
[   77.342245] ethub 4-0:1.0: 2 ports detected
[   77.355170] braswell_synobios: module license 'Synology Inc.' taints kernel.
[   77.363063] Disabling lock debugging due to kernel taint
[   77.369793] ttyS1 - failed to request DMA
[   77.374420] 2020-4-12 1:29:23 UTC
[   77.378142] synobios: load, major number 201
[   77.382926] Brand: Synology
[   77.386061] Model: DS-916+
[   77.389096] This is default settings: set group disks wakeup number to 1, spinup time deno 1
[   77.398553] synobios cpu_arch proc entry initialized
[... 1681 lines cut ...]
ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:62:70:35  
          inet addr:192.168.1.181  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9895 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10399 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:4209185 (4.0 MiB)  TX bytes:363982170 (347.1 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:62:70:36  
          inet addr:169.254.93.33  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: 2a02:8106:22b:d100:211:32ff:fe62:7036/64 Scope:Global
          inet6 addr: fe80::211:32ff:fe62:7036/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6475 errors:0 dropped:0 overruns:0 frame:0
          TX packets:253 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:3167564 (3.0 MiB)  TX bytes:71426 (69.7 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2020-06-21:
Because the script to configure Open vSwitches is executed before loading the driver, `ovs_*` devices for USB ethernet are not created. 

Could you try these steps to create `ovs_eth2`? 

* Add eth2 to `/usr/syno/etc/synoovs/ovs_interface.conf` 
* run `/etc/rc.network restart`

If this method works for you, you can run these commands on startup by configuring the Task scheduler.
Please note that I didn't try the above steps yet because I don't enable Open vSwitch for my NAS.

Podden, 2020-11-04:
does not work unfortunatly. Stil no ovs_eth2 is generated. I've tried restarting the whole device, but does not work either.

`
admin@nebukadnezar:~$ sudo vi /usr/syno/etc/synoovs/ovs_interface.conf
Password:
admin@nebukadnezar:~$ sudo /etc/rc.network restart
stop: Unknown instance: eth0
stop: Unknown instance: eth0
stop: Unknown instance: eth0/client
stop: Unknown instance: eth0/info
stop: Unknown instance: eth0/pd
stop: Unknown instance: eth1
stop: Unknown instance: eth1
stop: Unknown instance: eth1/client
stop: Unknown instance: eth1/info
stop: Unknown instance: eth1/pd
network-interface stop/waiting
stop: Unknown instance: eth2
stop: Unknown instance: eth2/client
stop: Unknown instance: eth2/info
stop: Unknown instance: eth2/pd
network-interface stop/waiting
stop: Unknown instance: lo
stop: Unknown instance: lo/client
stop: Unknown instance: lo/info
stop: Unknown instance: lo/pd
network-interface stop/waiting
stop: Unknown instance: ovs_eth0
stop: Unknown instance: ovs_eth0/client
stop: Unknown instance: ovs_eth0/info
stop: Unknown instance: ovs_eth0/pd
network-interface stop/waiting
dhcp-client stop/waiting
stop: Unknown instance: ovs_eth1/client
stop: Unknown instance: ovs_eth1/info
stop: Unknown instance: ovs_eth1/pd
stop: Unknown instance: tun
stop: Unknown instance: tun
stop: Unknown instance: tun/client
stop: Unknown instance: tun/info
stop: Unknown instance: tun/pd
SIOCSIFADDR: No such device
ovs_eth0: ERROR while getting interface flags: No such device
ovs_eth0: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
ovs_eth1: ERROR while getting interface flags: No such device
ovs_eth1: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
tun: ERROR while getting interface flags: No such device
tun: ERROR while getting interface flags: No such device
docker0   Link encap:Ethernet  HWaddr 02:42:26:47:B8:D6
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

status: Unknown instance: ovs_eth1
dhcp-client (ovs_eth1) start/running, process 21585
docker0   Link encap:Ethernet  HWaddr 02:42:26:47:B8:D6
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:62:70:35
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:178419 errors:0 dropped:0 overruns:0 frame:0
          TX packets:293687 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:21291878 (20.3 MiB)  TX bytes:412337574 (393.2 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:62:70:36
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:4E:0B:1B
          inet addr:192.168.1.180  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:272478 errors:1 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:14007425 (13.3 MiB)  TX bytes:1281657184 (1.1 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:156848 errors:0 dropped:0 overruns:0 frame:0
          TX packets:156848 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:21956273 (20.9 MiB)  TX bytes:21956273 (20.9 MiB)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:62:70:35
          inet addr:192.168.1.181  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:4530 (4.4 KiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:62:70:36
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:90 (90.0 B)

/bin/cat: /etc/dhcpd/dhcpd-eth2-static.conf: No such file or directory
port=0
`

riverscn, 2021-07-19:
I have this problem, too.

bb-qq, 2022-01-09:
Open vSwitch support is added in the latest release. Could you try it?

bb-qq, 2022-08-11:
I close this issue due to no reply. Please feel free to re-open the issue.

closed 2022-08-11 by bb-qq

marcel151, 2023-10-09:
Hello, I also want to create an active/passive bond. On DSM 7.2, the bond is created, but it is using the Gbit connection from the internal adapter. When I disconnect the cable from internal adapter, the connection doesn't work anymore.