bb-qq/aqc111 #18: Openvswitch interface generation

Podden, 2020-06-02:
### Description of the problem

Installing the Driver does not install an openvswitch interface. The new LAN Interface cannot be used with virtual machine manager. Is there a command to activate or generate an "ovr_eth2" interface?

My current workaround is deinstalling virtual machine manager, disable and enable openvswitch again.

Description of your products

    DS916+ Linux 3.10.105 #24922 SMP synology_braswell_916+
    DSM 6.2.2-24922 Update 6
    QNAP QNA-UC5G1T USB 3.0 auf 5 GbE Netzwe

Description of your environment

    Via Netgear 5Gbe Switch
    PC: Asus 10Gbe Card (Aquantia)
    Netgear XS508M-100 Switch
    Cat.7 Cable

### Output of ``ifconfig -a`` command

````

docker0   Link encap:Ethernet  HWaddr 02:42:53:52:5C:0F
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:62:70:35
          inet addr:192.168.1.20  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10382 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12184 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2584552 (2.4 MiB)  TX bytes:9360004 (8.9 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:62:70:36
          inet addr:192.168.1.21  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4163 errors:0 dropped:0 overruns:0 frame:0
          TX packets:730 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:722237 (705.3 KiB)  TX bytes:249691 (243.8 KiB)

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:4E:0B:1B
          inet addr:192.168.1.180  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3741 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:679793 (663.8 KiB)  TX bytes:937336 (915.3 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:56214 errors:0 dropped:0 overruns:0 frame:0
          TX packets:56214 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8800917 (8.3 MiB)  TX bytes:8800917 (8.3 MiB)

admin@nebukadnezar:~$ ifconfig
docker0   Link encap:Ethernet  HWaddr 02:42:53:52:5C:0F
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:62:70:35
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11332 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13626 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2712614 (2.5 MiB)  TX bytes:10155968 (9.6 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:62:70:36
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4370 errors:0 dropped:0 overruns:0 frame:0
          TX packets:770 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:756191 (738.4 KiB)  TX bytes:261795 (255.6 KiB)

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:4E:0B:1B
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:4214 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:776689 (758.4 KiB)  TX bytes:990096 (966.8 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:56439 errors:0 dropped:0 overruns:0 frame:0
          TX packets:56439 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8834180 (8.4 MiB)  TX bytes:8834180 (8.4 MiB)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:62:70:35
          inet addr:192.168.1.20  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:859 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1371 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:104883 (102.4 KiB)  TX bytes:782971 (764.6 KiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:62:70:36
          inet addr:192.168.1.21  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:169 errors:0 dropped:0 overruns:0 frame:0
          TX packets:40 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:31148 (30.4 KiB)  TX bytes:12140 (11.8 KiB)

ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:4E:0B:1B
          inet addr:192.168.1.180  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:208 errors:0 dropped:0 overruns:0 frame:0
          TX packets:107 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:50001 (48.8 KiB)  TX bytes:44445 (43.4 KiB)

````

bb-qq, 2020-06-27:
Please refer my this comment. (I should have written this comment here but I didn't notice that.)
https://github.com/bb-qq/aqc111/issues/17#issuecomment-647065184

Podden, 2020-11-04:
Okay, I've managed to manually add a ovr_eth2 interface to my configuration via putty. I've used 
`ovs-vsctl add-br ovs_eth2`
`ovs-vsctl add-port ovs_eth2 eth2`
and them manually added configuration files in `/etc/sysconfig/network-scripts/`
after `/etc/rc.network restart` the interface shows up correctly in ifconfig, and I can see it in DSM (I can see it's ovs_eth2 and not eth2 because when I want to configure DHCP via DSM it says "The DHCP range of interfaces ovs_eth2 and eth2 are in conflict". I can work this out, but LAN3 (ovs_eth2) still does not show up in Virtual Machine Manager for selecting in the network tab :(. Would be thankful for any guesses

Podden, 2020-11-04:
Okay, seems I had to wait a little bit. After fiddling around for a few minutes, pulling out and plugging in network cables and restarting interfaces, it finally shows up in Virtual Machine Manager. So no problem here anymore.

Now I only got the hassle of Directory Server hard pulling it's DNS Server IP from LAN, but thats another story....

AltB, 2020-12-10:
On DS1515+ with 4 ethernet ports and the USB 2.5G dongle as 'eth4' with static IP:
I have tried  `ovs-vsctl add-br ovs_eth4` and `ovs-vsctl add-port ovs_eth4 eth4` and added ifcfg-ovs_eth4 file in /etc/sysconfig/network-scripts (ovs-vsctl add-br doesnt created it). Tried restart network, up eth4, restart virtualization service, etc.
The eth4 (USB dongle) works but never appears in 'Virtual Machine Manager' network!

An idea ?

riverscn, 2021-07-19:
`disable and enable openvswitch` trick doesn't work on DSM 7 any more.

riverscn, 2021-07-19:
On DSM 7 I run these command and it works:

```
sudo cp /etc/sysconfig/network-scripts/ifcfg-ovs_eth1 /etc/sysconfig/network-scripts/ifcfg-ovs_eth2
sudo sed -i.bak 's/eth1/eth2/g' /etc/sysconfig/network-scripts/ifcfg-ovs_eth2
sudo /etc/rc.network restart
```
and then `ifconfig`

```
eth0      Link encap:Ethernet  HWaddr 00:11:32:DF:A9:CD
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:100367 errors:0 dropped:0 overruns:0 frame:0
          TX packets:79132 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:49029338 (46.7 MiB)  TX bytes:42582421 (40.6 MiB)
          Interrupt:97 base 0xc000

eth1      Link encap:Ethernet  HWaddr 00:11:32:DF:A9:CE
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:96 base 0x8000

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:09:B9
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1242417 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1147630 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:904380866 (862.4 MiB)  TX bytes:1548279863 (1.4 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:75443 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75443 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:32952906 (31.4 MiB)  TX bytes:32952906 (31.4 MiB)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:DF:A9:CD
          inet addr:192.168.1.***  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fedf:a9cd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:751 errors:0 dropped:0 overruns:0 frame:0
          TX packets:456 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:260660 (254.5 KiB)  TX bytes:145012 (141.6 KiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:DF:A9:CE
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 00:E0:4C:68:09:B9
          inet addr:192.168.1.***  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:9b9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:692 errors:0 dropped:0 overruns:0 frame:0
          TX packets:498 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:192800 (188.2 KiB)  TX bytes:98419 (96.1 KiB)
```

bb-qq, 2021-07-20:
@riverscn , thank you for creating [the PR](https://github.com/bb-qq/r8152/pull/134) to support OVS. I will try this.

Can anyone else try [the script](https://github.com/bb-qq/r8152/blob/26bed0cb7418c0963c25b10c71c31af4892259b4/scripts/start-stop-status) on the PR and help us verify that it works fine in various environments?

anhaehne, 2021-11-12:
I've tried the script now for a month and had no issues. All VM's have network access and the network connection is correctly reesteblished after reboots and hard resets.

bb-qq, 2021-11-13:
Thank you for your report. I will merge this PR and release drivers for both platforms.

bb-qq, 2022-01-09:
I released new releases that include this change.
https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-5
https://github.com/bb-qq/r8152/releases/tag/2.15.0-5

bb-qq, 2022-08-11:
I close this issue since open vSwich support is working correctly.

closed 2022-08-11 by bb-qq