bb-qq/aqc111 #19: DS216PLAY(MONAKO) CAN'T WORK

b4ll5oh4rd, 2020-06-29:
DSM 6.2.3-25426 Update 1,CAN'T WORK.
BUT DSM 6.2.3-25426 IS OK!
I GUESS SYNOLOGY BLOCK 2 INTERFACE CARD.ONLY ONE WORK.
I CAN'T DOWNGRADE TO DSM 6.2.3-25426 VERISON.

bb-qq, 2020-07-02:
Could you paste the result of `dmesg` here?

b4ll5oh4rd, 2020-07-03:
[15410.880741] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15414.878698] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15420.875641] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15440.865242] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15442.864232] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15450.860088] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15452.859110] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15454.857993] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15456.857027] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15464.852897] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15468.850845] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15472.848830] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15474.847754] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15480.844657] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[15482.843662] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[243738.243230] swapper/0: page allocation failure: order:0, mode:0x20
[243738.249493] CPU: 0 PID: 0 Comm: swapper/0 Tainted: PF          O 3.10.105 #25426
[243738.249520] [<80015320>] (unwind_backtrace+0x0/0xec) from [<8001193c>] (show_stack+0x10/0x14)
[243738.249535] [<8001193c>] (show_stack+0x10/0x14) from [<800920f8>] (warn_alloc_failed+0xe4/0x11c)
[243738.249546] [<800920f8>] (warn_alloc_failed+0xe4/0x11c) from [<8009443c>] (__alloc_pages_nodemask+0x514/0x7a4)
[243738.249560] [<8009443c>] (__alloc_pages_nodemask+0x514/0x7a4) from [<803fe344>] (__netdev_alloc_frag+0x140/0x158)
[243738.249571] [<803fe344>] (__netdev_alloc_frag+0x140/0x158) from [<8040155c>] (__netdev_alloc_skb+0x48/0xfc)
[243738.249581] [<8040155c>] (__netdev_alloc_skb+0x48/0xfc) from [<8032c57c>] (stmmac_rx+0x4ec/0x718)
[243738.249590] [<8032c57c>] (stmmac_rx+0x4ec/0x718) from [<8032e8a4>] (stmmac_poll+0x30/0x60)
[243738.249602] [<8032e8a4>] (stmmac_poll+0x30/0x60) from [<8040bec8>] (net_rx_action+0x110/0x184)
[243738.249612] [<8040bec8>] (net_rx_action+0x110/0x184) from [<80025f08>] (__do_softirq+0x110/0x1e8)
[243738.249620] [<80025f08>] (__do_softirq+0x110/0x1e8) from [<80026080>] (do_softirq+0x50/0x58)
[243738.249627] [<80026080>] (do_softirq+0x50/0x58) from [<80026308>] (irq_exit+0x9c/0xd0)
[243738.249635] [<80026308>] (irq_exit+0x9c/0xd0) from [<8000ebc4>] (handle_IRQ+0x44/0x90)
[243738.249643] [<8000ebc4>] (handle_IRQ+0x44/0x90) from [<80008560>] (gic_handle_irq+0x2c/0x5c)
[243738.249653] [<80008560>] (gic_handle_irq+0x2c/0x5c) from [<804c4680>] (__irq_svc+0x40/0x50)
[243738.249658] Exception stack(0x806a5f70 to 0x806a5fb8)
[243738.249663] 5f60:                                     80b2b800 00000000 2e0dcf02 00000000
[243738.249670] 5f80: 806a4000 806ac4cc 806ac53c 804ca6dc 806e098d 806e098d 806a4000 00000000
[243738.249675] 5fa0: 01000000 806a5fb8 8000eedc 8000eee0 60060013 ffffffff
[243738.249684] [<804c4680>] (__irq_svc+0x40/0x50) from [<8000eee0>] (arch_cpu_idle+0x2c/0x30)
[243738.249694] [<8000eee0>] (arch_cpu_idle+0x2c/0x30) from [<8005c854>] (cpu_startup_entry+0x100/0x144)
[243738.249705] [<8005c854>] (cpu_startup_entry+0x100/0x144) from [<80668ab4>] (start_kernel+0x328/0x334)
[243738.249710] Mem-info:
[243738.249714] Normal per-cpu:
[243738.249718] CPU    0: hi:  186, btch:  31 usd:  31
[243738.249723] CPU    1: hi:  186, btch:  31 usd:   1
[243738.249733] active_anon:27692 inactive_anon:27686 isolated_anon:160
                 active_file:4200 inactive_file:7042 isolated_file:0
                 unevictable:1024 dirty:576 writeback:10672 unstable:0
                 free:3960 slab_reclaimable:3526 slab_unreclaimable:8068
                 mapped:6017 shmem:2785 pagetables:2796 bounce:0
                 free_cma:3489
[243738.249753] Normal free:15840kB min:8192kB low:10240kB high:12288kB active_anon:110768kB inactive_anon:110744kB active_file:16800kB inactive_file:28168kB unevictable:4096kB isolated(anon):640kB isolated(file):0kB present:475136kB managed:393556kB mlocked:4096kB dirty:2304kB writeback:42688kB mapped:24068kB shmem:11140kB slab_reclaimable:14104kB slab_unreclaimable:32272kB kernel_stack:3336kB pagetables:11184kB unstable:0kB bounce:0kB free_cma:13956kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[243738.249757] lowmem_reserve[]: 0 0
[243738.249763] Normal: 458*4kB (UERC) 363*8kB (UERC) 302*16kB (RC) 196*32kB (C) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 15840kB
[243738.249786] 22460 total pagecache pages
[243738.249793] 8366 pages in swap cache
[243738.249797] Swap cache stats: add 40742, delete 32376, find 6380979/6382004
[243738.249800] Free swap  = 1979044kB
[243738.249803] Total swap = 2097084kB
[243738.250146] stmmaceth 9630000.dwmac: fail to alloc skb entry 2
[243738.256761] stmmaceth 9630000.dwmac: packet dropped
[243738.261718] stmmaceth 9630000.dwmac: fail to alloc skb entry 2
[243738.268530] stmmaceth 9630000.dwmac: packet dropped
[243738.273600] stmmaceth 9630000.dwmac: fail to alloc skb entry 2
[243738.279695] stmmaceth 9630000.dwmac: packet dropped
[243738.284737] stmmaceth 9630000.dwmac: fail to alloc skb entry 2
[243738.860203] stmmaceth 9630000.dwmac: packet dropped
[243738.865203] stmmaceth 9630000.dwmac: packet dropped
[243738.896840] stmmaceth 9630000.dwmac: packet dropped
[243743.327041] net_ratelimit: 2 callbacks suppressed
[243743.331829] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531571.273504] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[... cut ...]
[531591.171119] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531592.170675] net_ratelimit: 1 callbacks suppressed
[531592.175470] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531593.170109] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531595.169057] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531597.168019] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531599.167081] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped
[531601.258053] aqc111 4-1:1.0 eth1: kevent 2 may have been dropped

b4ll5oh4rd, 2020-07-03:
b4ll5oh4rd@ds216play:~$ ifconfig 
eth0      Link encap:Ethernet  HWaddr 00:11:32:55:FF:0C  
          inet addr:192.168.100.64  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11510766 errors:0 dropped:10 overruns:0 frame:0
          TX packets:7997209 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3693042251 (3.4 GiB)  TX bytes:1406391294 (1.3 GiB)
          Interrupt:130 

eth1      Link encap:Ethernet  HWaddr 24:5E:BE:4D:9A:48  
          inet addr:192.168.100.24  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe4d:9a48/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1501899 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:69632224 (66.4 MiB)  TX bytes:1711456 (1.6 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:355237 errors:0 dropped:0 overruns:0 frame:0
          TX packets:355237 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:25061719 (23.9 MiB)  TX bytes:25061719 (23.9 MiB)

bb-qq, 2020-07-04:
Hmm, there are no messages about the second adapter... Is this "full" output from the beginning? And the output of `lsusb` also may help.

b4ll5oh4rd, 2020-07-05:
not full,but same output,may be 10 thousand output.eth1 is usb adapter
b4ll5ox3h4rd@ds216play:~$ lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd ehci-platform.0) hub
|__usb2          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (Linux 3.10.105 ohci_hcd Generic Platform OHCI Controller ohci-platform.0) hub
  |__2-1         051d:0002:0106 00  2.00   12MBit/s 100mA 1IF  (American Power Conversion BK650M2-CH 9B2004A07552)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller xhci-hcd.0.auto) hub
  |__4-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 02I08005)

b4ll5oh4rd, 2020-07-05:
sorry,my fault!everything is ok,now.thank u !!!

closed 2020-07-05 by b4ll5oh4rd