bb-qq/aqc111 #20: Support for Geminilake CPUs

kkuhler, 2020-07-02:
I have a DS920+ with an Intel Celeron J4125 CPU (Geminilake) with a QNAP QNA-UC5G1T installed.  

I tried the to install the ApolloLake (aqc111-apollolake-1.3.3.0-2.spk) build on DSM 6.2.3-25426 and it says:

This package is not supported on the platform of Synology NAS or is incompatible with the current DSM version.

If you create a build for this platform I would be happy to test it for you

Thanks

bb-qq, 2020-07-02:
Currently, building packages for Geminilake architecture is not supported, because toolchains for them are not released yet.
https://sourceforge.net/projects/dsgpl/files/DSM%206.2.2%20Tool%20Chains/

marco712, 2020-07-31:
when will the update be released DS920+???

marco712, 2020-08-01:
If you create a build for this platform I would be happy to test it for you

Thanks

Evo609, 2020-08-03:
> Currently, building packages for Geminilake architecture is not supported, because toolchains for them are not released yet.
> https://sourceforge.net/projects/dsgpl/files/DSM%206.2.2%20Tool%20Chains/

Hi,

Geminilake is already there 😊

marco712, 2020-08-04:
I have a DS920+ with an Intel Celeron J4125 CPU (Geminilake) 
can someone make the drivers

kkuhler, 2020-08-05:
Awesome.  The toolchain was released on 8/3/Covid.  Hopefully the dev gets a chance sometime soon to release a build.

bb-qq, 2020-08-14:
Toolchains for geminilake is certainly released, but the build environment including kernel sources is not yet...

```
pkgscripts-ng/EnvDeploy -v 6.2 --list
Available platforms: 6281 alpine alpine4k apollolake armada370 armada375 armada37xx armada38x armadaxp avoton braswell broadwell broadwellnk bromolow cedarview comcerto2k denverton dockerx64 evansport grantley hi3535 kvmx64 monaco qoriq rtd1296 x64
```

marco712, 2020-08-14:
I simply renamed the line
arch= "apollolake"
with
arch="geminilake"
in the aqc111-apollolake-1.3.3.0-1.spk file with NotePad++ and saved the file.

After Installation on the DS920+ the driver worked well and I have proper LAN connection
(see Daniels Video: https://www.youtube.com/watch?v=AyU7bB7BOsc)

bb-qq, 2020-10-12:
I uploaded [the driver for the geminilake platform](https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-geminilake-1.3.3.0-2.spk) using the official toolchains and parameters. (thanks @silverko2)
I guess that this driver is almost the same as one created by the rename method.

wujin941005, 2020-10-12:
> I uploaded [the driver for the geminilake platform](https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-geminilake-1.3.3.0-2.spk) using the official toolchains and parameters. (thanks @silverko2)
> I guess that this driver is almost the same as one created by the rename method.

I am so luckily, literally got the adapter hours ago and about to install the driver.

haoranxu1989, 2020-10-28:
> I uploaded [the driver for the geminilake platform](https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-geminilake-1.3.3.0-2.spk) using the official toolchains and parameters. (thanks @silverko2)
> I guess that this driver is almost the same as one created by the rename method.

Thanks！

bb-qq, 2020-12-13:
Closing. Please create a new ticket If you found issues.

closed 2020-12-13 by bb-qq