bb-qq/aqc111 #22: DS718+ Report

jmrickerby, 2020-08-18:
First off - thanks for making this driver! While it would be great if Synology offered a faster interface on their 2-bay and 4-bay models, the x20 series indicates that they are in no hurry to do this.

**Equipment list**
Synology DS718+ with:
   - 16GB G.SKILL F3-1866C10D-16GRSL memory kit
   - 2x Western Digital 6TB Red Pro hard drives (WD6002FFWX-68TZ4N0)
   - QNAP QVA-UC6G1T USB network adapter
   - Filesystem:  Btrfs
   - MTU 9000 on Synology 1Gbps (LACP) interface and QNAP QVA-UC6G1T USB network adapter

Ryzen 7 3800X PC with:
   - Samsung SSD 970 EVO Plus 1TB
   - ASUS XG-C100C 10G Ethernet Adapter

Cisco 2960-X024TD-L network switch with:
   - 2x FS 10GBASE-T SFP+ Copper RJ-45 30m Transceiver (SFP-10G-T) - One SFP+ to the PC the other to the QNAP QVA-UC6G1T

Upon receiving the QNAP QVA-UC6G1T USB network adapter, I immediately removed the metal enclosure and added an old motherboard VRM heatsink to the bottom thermal interface of the adapter. Within the QVA-UC6G1T, the included heatsink looks reasonable, however it has no airflow or direct connection to the metal enclosure to perform properly. The adapter is currently sitting on the shelf, without an enclosure, in front of the Synology DS718+.

CrystalDiskMark used with default settings.

CrystalDiskMark with mapped drive from **Synology 1Gbps** Ethernet interface:
[image](url)

CrystalDiskMark with mapped drive from **QNAP QVA-UC6G1T 5Gbps** Ethernet interface:
[image](url)

jmrickerby, 2020-11-26:
Recently seeing the performance of the 5Gpbs link, drop to tens of megabits. Checking configuration, there is nothing to indicate a change in the interface status. The adapter is out of it's enclosure, with an additional heat sink, and the device does not seems to be hot, so I do not believe this is thermal throttling. Stopping and restarting the ACQ111 driver package temporarily fixes the issue.

bb-qq, 2020-11-29:
> It appears data inbound to the DS718+ is peaking at 1.1Gbps.
I would guess that disk write speed is the bottleneck.
Measuring pure communication speed using iperf would help isolate the problem.

> Recently seeing the performance of the 5Gpbs link, drop to tens of megabits.

* Are you using the front USB port?
* What was the output of dmesg when the problem occurred?

bb-qq, 2020-11-29:
I replied in issue #32.