bb-qq/aqc111 #24: DS1019+ losing connections

jgibo1, 2020-08-31:
### Description of the problem

While copying data the network connection randomly drops. I can go to package center and turn off and on to get it to work again. May or not be related to my 10Gb rj45 sfp+ in my Mikrotik.

### Description of your products

> DS1019+ with the Qnap QNA-UC5G1T

>     - DSM version 6.2.3 update 2

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: Mikrotik 5 port desktop switch crs305-1g-4s with a ipolex 10g sfp+ rj45 copper transceiver
> - cable category (Cat.6)

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````Not sure how to do this part below
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

jmrickerby, 2020-09-12:
The QNAP QNA-UC5G1T can get very hot, then drop it's connection rate. I would suggest removing the adapter from its enclosure, to see if it still appears to drop the connection with better ventilation. On the QNA-UC6G1T, underneath the Mac address/serial number sticker, there is a small screw. If you remove this screw, you can slide the circuit board out of the enclosure. The black finned heat sink, pictured below, should performs better with some air circulation. The copper heat sink is from a retired motherboard VRM.

[image]

wheatiesthecereal, 2020-10-28:
@jmrickerby - Glad to see that they use a screw. I always end up damaging glue-heavy pieces.

devedse, 2021-07-20:
@jmrickerby I'd like to know how the QNAP with heat sink has performed up untill now :))

jmrickerby, 2021-07-20:
> 
> 
> @jmrickerby I'd like to know how the QNAP with heat sink has performed up untill now :))

@devedse - So far so good! No performance slow-downs due to thermal throttling that I'm aware of. On an unrelated note, having some issues with 9K Jumbo packets, but I think this ma be related to a Windows update.

DamianEdwards, 2022-11-23:
I'm experiencing connection drops and subsequent SMB instability after copying files over the QNAP QNA-UC5G1T. Client is 2.5GbE. How would I go about diagnosing if this is due to overheating?

bb-qq, 2023-04-23:
Low power mode is enabled by default in recent versions. With this setting, thermal issues are almost non-existent.