bb-qq/aqc111 #25: DS1819+ issues

cyrus104, 2020-09-08:
### Description of the problem

> I have tested the QNA-UC5G1T in all of the ports on the DS1819+ and have been getting errors on all devices. I thought that power might be an issue but no port solved the issue. 

### Description of your products

> - DS1819+
>     - Linux notadrobo 4.4.59+ #25426
>     - DSM version x.x
> - Qnap QNA-UC5G1T

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: Switch is a Unfi 10Gb switch
> - Cat.7

### Output of ``dmesg`` command

````
[   55.488743] usb 2-2-port3: logical disconnect on port 3 [kworker/3:5]
[   55.670308] usb 2-2.3: USB disconnect, device number 62
[   55.984463] usb 2-2.3: new SuperSpeed USB device number 63 using xhci_hcd
[   56.712340] usb 2-2.3: usb_reset_and_verify_device Failed to disable LTM
               .
[   56.720749] usb 2-2-port3: logical disconnect on port 3 [kworker/3:1]
[   56.908170] usb 2-2.3: USB disconnect, device number 63
[   57.005499] usb 2-2.3: new SuperSpeed USB device number 64 using xhci_hcd
[   57.144328] usb 2-2.3: usb_reset_and_verify_device Failed to disable LTM
               .
[   57.160301] usb 2-2-port3: logical disconnect on port 3 [kworker/3:6]
[   57.384300] usb 2-2.3: USB disconnect, device number 64
[   57.700801] usb 2-2.3: Device not responding to setup address.
[   57.910814] usb 2-2.3: Device not responding to setup address.
[   58.067242] mii: module verification failed: signature and/or required key missing - tainting kernel
[   58.122301] usb 2-2.3: device not accepting address 65, error -71
[   58.172119] usbcore: registered new interface driver aqc111
[   58.500811] usb 2-2.3: Device not responding to setup address.
[   58.710811] usb 2-2.3: Device not responding to setup address.
[   59.036760] usb 2-2.3: device not accepting address 66, error -71
[   59.407823] usb 2-2.3: Device not responding to setup address.
[   59.617824] usb 2-2.3: Device not responding to setup address.
[   59.825319] usb 2-2.3: device not accepting address 67, error -71
[   60.204411] ip_tables: (C) 2000-2006 Netfilter Core Team
[   60.509321] usb 2-2.3: new SuperSpeed USB device number 68 using xhci_hcd
[   60.543000] aqc111 2-2.3:1.0 eth5: register 'aqc111' at usb-0000:00:15.0-2.3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:4f:87:dd
[   60.577664] ip6_tables: (C) 2000-2006 Netfilter Core Team
[   60.772847] aqc111 2-2.3:1.0 eth5: unregister 'aqc111' usb-0000:00:15.0-2.3, QNAP QNA-UC5G1T USB to 5GbE Adapter
[   60.807439] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[   60.868108] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[   60.927938] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x61) reg index 0x0000: -71
[   60.985924] usb 2-2.3: usb_reset_and_verify_device Failed to disable LTM
               .
[   61.020822] usb 2-2-port3: logical disconnect on port 3 [kworker/3:1]
[   61.049184] aqc111 2-2.3:1.0 (unnamed net_device) (uninitialized): Failed to write(0xb0) reg index 0x0001: -19
[   61.078284] aufs 4.4-20160328
[   61.166799] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[   61.256333] usb 2-2.3: USB disconnect, device number 68
[   61.310256] Bridge firewalling registered
[   61.365759] usb 2-2.3: new SuperSpeed USB device number 69 using xhci_hcd
[   61.402965] aqc111 2-2.3:1.0 eth5: register 'aqc111' at usb-0000:00:15.0-2.3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:4f:87:dd
[   61.492861] aqc111 2-2.3:1.0 eth5: unregister 'aqc111' usb-0000:00:15.0-2.3, QNAP QNA-UC5G1T USB to 5GbE Adapter
[   61.573219] IPVS: Registered protocols (TCP, UDP, SCTP, AH, ESP)
[   61.577109] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[   61.580590] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[   61.584098] aqc111 2-2.3:1.0 eth5 (unregistered): Failed to write(0x61) reg index 0x0000: -71
[   61.588366] usb 2-2.3: usb_reset_and_verify_device Failed to disable LTM
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008ABB20CA6010)
  |__1-2         0bda:5411:0104 09  2.10  480MBit/s 0mA 1IF  (Generic 4-Port USB 2.0 Hub syno.ext.hub) hub
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:0411:0104 09  3.00 5000MBit/s 0mA 1IF  (Generic 4-Port USB 3.0 Hub syno.ext.hub) hub
    |__2-2.3     1c04:0015:0101 00  3.20 5000MBit/s  0IFs (QNAP▒ Pacific▒ 04I22793)
````

### Output of ``ifconfig -a`` command

````
eth0,eth1,eth2,eth3,eth4 (Internal Synology 10GBe card), no eth5 shows up
````

scyto, 2020-09-12:
this is what my lsusb command looks like on my DS918+ (admittedly with DSM7 on it).
But the device listing looks very strange compare to mine... the QNAP entries doesn't look right, try different ports if you can, make sure you try different usb cable itoo (make sure it is a short usb c>A cable and don't use a plug adapter).  

alexb@SYN02:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00943A38DA5A16)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 2IFs (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 02I08844)

cyrus104, 2020-09-14:
I tested all the ports and a couple of cables with the same results. One
port must have lower power because it showed less.

scyto, 2020-09-15:
@cyrus104 then maybe you have faulty adapter or faulty synology - the adapter a)is not drawing any power and it's device string looks corrupted, i don't think you have an aqc111 driver issue as that info in lsusb is independent of the driver ever loading, AFAIK.