bb-qq/aqc111 #26: Doesn't work on DSM7  - no surprise, just FYI

scyto, 2020-09-12:
### Description of the problem

Driver won't start at boot or using run on DSM7 preview.

### Description of your products

Linux SYN02 4.4.180+ #40850 SMP Wed Sep 2 17:03:14 CST 2020 x86_64 GNU/Linux synology_apollolake_918+
DS 1815+
DSM 7.0-40850
QNAP UC5G1T
--

### Description of your environment

CAT5 connect to 1g port on a switch

### Output of ``dmesg`` command

````
[161752.847455] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[161768.749960] mii: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
[161768.761983] usbnet: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
[161768.774312] aqc111: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
[161777.941873] mii: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
[161777.954066] usbnet: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
[161777.966435] aqc111: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.180+ SMP mod_unload '
````

### Output of ``lsusb`` command

````
alexb@SYN02:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00943A38DA5A16)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 2IFs (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 02I08844)
````

### Output of ``ifconfig -a`` command

````
alexb@SYN02:~$ ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:22:12:2B:59  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:22ff:fe12:2b59/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:394317 errors:0 dropped:0 overruns:0 frame:0
          TX packets:388391 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:28530272 (27.2 MiB)  TX bytes:84844276 (80.9 MiB)

docker_gw Link encap:Ethernet  HWaddr 02:42:6D:3E:65:A2  
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

dockerb12 Link encap:Ethernet  HWaddr 3E:2C:16:CA:3E:18  
          inet6 addr: fe80::3c2c:16ff:feca:3e18/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:195957 errors:0 dropped:0 overruns:0 frame:0
          TX packets:170902 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:15701335 (14.9 MiB)  TX bytes:27338771 (26.0 MiB)

dockerbd0 Link encap:Ethernet  HWaddr A6:90:5C:9E:8D:77  
          inet6 addr: fe80::a490:5cff:fe9e:8d77/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:10710 (10.4 KiB)

dockere11 Link encap:Ethernet  HWaddr BE:D7:82:68:DD:8E  
          inet6 addr: fe80::bcd7:82ff:fe68:dd8e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:198360 errors:0 dropped:0 overruns:0 frame:0
          TX packets:217504 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:18349375 (17.4 MiB)  TX bytes:57506591 (54.8 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:AE:28:4F  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:7232402 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1666519 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2424063246 (2.2 GiB)  TX bytes:379412153 (361.8 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:AE:28:50  
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:7185316 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7185316 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1094434122 (1.0 GiB)  TX bytes:1094434122 (1.0 GiB)

ovs-syste Link encap:Ethernet  HWaddr A2:FA:D2:65:1A:EA  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:AE:28:4F  
          inet addr:192.168.1.38  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5707722 errors:0 dropped:0 overruns:0 frame:0
          TX packets:667528 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1630325220 (1.5 GiB)  TX bytes:237238960 (226.2 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:AE:28:50  
          inet addr:192.168.1.39  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 52:9B:C9:FF:B9:8A  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:178993 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5225146 errors:0 dropped:1193 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:27151146 (25.8 MiB)  TX bytes:1598347817 (1.4 GiB)

tap021132 Link encap:Ethernet  HWaddr BA:FB:70:AB:05:A4  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:723652 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5727662 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:109792556 (104.7 MiB)  TX bytes:1137271995 (1.0 GiB)

````

bb-qq, 2020-09-13:
Thank you for the report. I will update my drivers after toolchains for DSM7 will be released.

mlc130104, 2020-10-17:
Seems DSM7 preview toolchains was released.

PureWhiteWu, 2020-12-08:
Is there any update on this issue?

Benniphx, 2020-12-17:
now with 7 beta out is there any update

RickGanKL, 2021-01-15:
I hope it supoorted DSM7

cmdcirdec, 2021-02-19:
Yes me too! because I'm desperately waiting for dsm 7

nighoa, 2021-04-29:
Hi community ! 

I'm still waiting for it also. Please update, or give compiling instructions for the latest DSM 7.0 Beta SDK.

patrick-87, 2021-05-20:
@bb-qq First of all, thank you very much for the driver, it has been running with me since I had my DS918 + without any problems :-). I know DSM 7 is not yet available, but is there any information whether the driver will run on DSM 7 (Beta) and if so when?

ihvar, 2021-05-27:
I'm also waiting for driver to install it on DSM 7 for 418play

bb-qq, 2021-05-29:
Sorry, currently I do not have any plans to support DSM7. And I know installation via the Package Center is no longer supported in DSM 7 due to policy change of Synology. I will investigate this issue when I move to DSM7.

jcavalheri, 2021-06-03:
Hello Team,

Any plans to upgrade this to DSM 7???
I wish I would have read this tread before!!! I just bough the QNAP modules :-(

Any idea how community can support here?
Thanks Jose

scyto, 2021-06-03:
See https://github.com/bb-qq/aqc111/issues/26#issuecomment-691594616

bb-qq, 2021-07-10:
I uploaded drivers for DSM7. 
It is still loaded from the Package Manager and it needs additional steps with the root shell.

closed 2022-01-09 by bb-qq