bb-qq/aqc111 #29: DS1520+ geminilake/QNA-UC5G1T connection failure on restart

noblemd00, 2020-10-29:
### Description of the problem

5GbE Qnap connection failure occurs every shutdown --> startup, restart, or stop/run the driver package. The IP4 connection does not retain the Subnet mask, Gateway and DNS server settings resulting in connection failure. 

Reconnection can be reestablish using manual configuration if failure occurred using DHCP or if the failure occurred with a manual connection it can be reestablish using automatic (DHCP)  - reconnection is reestablished via the opposite IP4 connection type which was active when the connection failure occurred.

### Description of your products

DS1520+ 
uname -a output:
Linux TM5 4.4.59+ #25426 SMP PREEMPT Thu Jul 16 17:09:17 CST 2020 x86_64 GNU/Linux synology_geminilake_1520+

DSM 6.2.3-25426 Update 2

aqc111-geminilake-1.3.3.0-2.spk
QNA-UC5G1T

### Description of your environment

Connected to a 10GbE switch (NETGEAR XS508M) with a Cat 8 cable

The problem has nothing to do with the currently connected computer - it is an internal - Synology/QNA-UC5G1T problem.

X570 TUF GAMING PLUS
AMD Ryzen 9 3950X 16-Core Processor 3.49GHz
RAM: 16.0 GB
NIC: QNAP QXG-10G1T Single-Port (10Gbase-T) 10GbE Network Expansion PCIe

lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0007207481C658)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 06I08145)

I will submit the other Linux command outputs via an additional comment when I restart the DS1520+ and the connection failure problem occurs.

noblemd00, 2020-10-29:
BTW: RAID setup: 

32TB RAID 5 (8Tb 7200 x 5)
2x NVME 500Gb SSD Cache

noblemd00, 2020-10-30:
Putty Command outputs - after startup - connection failure:

[Output ACQ111.txt](https://github.com/bb-qq/aqc111/files/5465332/Output.ACQ111.txt)

bb-qq, 2021-02-23:
Did you try to use the front port?

bb-qq, 2022-08-11:
The startup script of the recent release is different from past releases. So I close this issue. If the issue remains, please re-open the issue.

closed 2022-08-11 by bb-qq