bb-qq/aqc111 #31: DS 1091+ not seeing any improvement over 2.5 gbpe

gordoabc, 2020-11-05:
### Description of the problem

I see from other reports iperf3 doesn't give the full 5 gbpe throughput - presumably limited by the usb3 port but I am not seeing any improvement over 2.5 gbpe and am below what others have posted

### Description of your products
Synology DS1019+
uname -a 
Linux sourdust 4.4.59+ #25426 SMP PREEMPT Wed Jul 8 03:21:29 CST 2020 x86_64 GNU/Linux synology_apollolake_1019+

DSM version 6.2.3-25426 Update 2

USB 3.0 Type-C to 5GbE Adapter (QNA-UC5G1T) by QNAP

### Description of your environment

Connected to imac with 10 gbpe through OWC thunderbolt 3 hub
netgear MS510TXPP switch

### Output of ``dmesg`` command

[   92.326032] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:50:d1:ef
[   92.791796] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  100.420242] aqc111 2-1:1.0 eth2: Link Speed 5000, USB 3

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0002AE89345A85)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 06I14522)
  |__2-2         059f:1090:1001 00  3.00 5000MBit/s 896mA 1IF  (LaCie P9227 Slim 000000020CEC)

### Output of ``ifconfig -a`` command

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:50:D1:EF  
          inet addr:192.168.86.35  Bcast:192.168.86.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe50:d1ef/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:38739580 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14861208 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:55811895187 (51.9 GiB)  TX bytes:13855890440 (12.9 GiB)

### Output of ``iperf3 -c sourdust.macgmason.us -P 4`` command

Connecting to host sourdust.macgmason.us, port 5201
[  7] local 192.168.86.54 port 59301 connected to 192.168.86.35 port 5201
[  9] local 192.168.86.54 port 59302 connected to 192.168.86.35 port 5201
[ 11] local 192.168.86.54 port 59303 connected to 192.168.86.35 port 5201
[ 13] local 192.168.86.54 port 59304 connected to 192.168.86.35 port 5201
[ ID] Interval           Transfer     Bitrate
[  7]   0.00-1.00   sec  69.8 MBytes   585 Mbits/sec                  
[  9]   0.00-1.00   sec   136 MBytes  1.14 Gbits/sec                  
[ 11]   0.00-1.00   sec  25.9 MBytes   218 Mbits/sec                  
[ 13]   0.00-1.00   sec  30.9 MBytes   260 Mbits/sec                  
[SUM]   0.00-1.00   sec   262 MBytes  2.20 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   1.00-2.00   sec  51.6 MBytes   432 Mbits/sec                  
[  9]   1.00-2.00   sec   112 MBytes   935 Mbits/sec                  
[ 11]   1.00-2.00   sec  49.2 MBytes   413 Mbits/sec                  
[ 13]   1.00-2.00   sec  78.3 MBytes   657 Mbits/sec                  
[SUM]   1.00-2.00   sec   291 MBytes  2.44 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   2.00-3.00   sec  92.7 MBytes   778 Mbits/sec                  
[  9]   2.00-3.00   sec  10.1 MBytes  84.8 Mbits/sec                  
[ 11]   2.00-3.00   sec   104 MBytes   874 Mbits/sec                  
[ 13]   2.00-3.00   sec  94.0 MBytes   789 Mbits/sec                  
[SUM]   2.00-3.00   sec   301 MBytes  2.53 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   3.00-4.00   sec  71.0 MBytes   595 Mbits/sec                  
[  9]   3.00-4.00   sec  82.0 MBytes   687 Mbits/sec                  
[ 11]   3.00-4.00   sec  66.6 MBytes   559 Mbits/sec                  
[ 13]   3.00-4.00   sec  64.3 MBytes   540 Mbits/sec                  
[SUM]   3.00-4.00   sec   284 MBytes  2.38 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   4.00-5.00   sec  82.2 MBytes   690 Mbits/sec                  
[  9]   4.00-5.00   sec  66.7 MBytes   560 Mbits/sec                  
[ 11]   4.00-5.00   sec  52.1 MBytes   437 Mbits/sec                  
[ 13]   4.00-5.00   sec  77.1 MBytes   647 Mbits/sec                  
[SUM]   4.00-5.00   sec   278 MBytes  2.33 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   5.00-6.00   sec  81.9 MBytes   687 Mbits/sec                  
[  9]   5.00-6.00   sec  89.2 MBytes   748 Mbits/sec                  
[ 11]   5.00-6.00   sec  67.7 MBytes   568 Mbits/sec                  
[ 13]   5.00-6.00   sec  63.3 MBytes   531 Mbits/sec                  
[SUM]   5.00-6.00   sec   302 MBytes  2.53 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   6.00-7.00   sec  80.6 MBytes   676 Mbits/sec                  
[  9]   6.00-7.00   sec  78.0 MBytes   654 Mbits/sec                  
[ 11]   6.00-7.00   sec  56.9 MBytes   477 Mbits/sec                  
[ 13]   6.00-7.00   sec  69.0 MBytes   579 Mbits/sec                  
[SUM]   6.00-7.00   sec   284 MBytes  2.39 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   7.00-8.00   sec  88.5 MBytes   742 Mbits/sec                  
[  9]   7.00-8.00   sec  65.2 MBytes   547 Mbits/sec                  
[ 11]   7.00-8.00   sec  84.2 MBytes   706 Mbits/sec                  
[ 13]   7.00-8.00   sec  61.5 MBytes   516 Mbits/sec                  
[SUM]   7.00-8.00   sec   299 MBytes  2.51 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   8.00-9.00   sec  94.5 MBytes   793 Mbits/sec                  
[  9]   8.00-9.00   sec  62.0 MBytes   520 Mbits/sec                  
[ 11]   8.00-9.00   sec  49.7 MBytes   417 Mbits/sec                  
[ 13]   8.00-9.00   sec  85.2 MBytes   715 Mbits/sec                  
[SUM]   8.00-9.00   sec   291 MBytes  2.44 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  7]   9.00-10.00  sec  79.3 MBytes   665 Mbits/sec                  
[  9]   9.00-10.00  sec  76.1 MBytes   638 Mbits/sec                  
[ 11]   9.00-10.00  sec  70.9 MBytes   595 Mbits/sec                  
[ 13]   9.00-10.00  sec  70.0 MBytes   587 Mbits/sec                  
[SUM]   9.00-10.00  sec   296 MBytes  2.49 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  7]   0.00-10.00  sec   792 MBytes   665 Mbits/sec                  sender
[  7]   0.00-10.00  sec   792 MBytes   664 Mbits/sec                  receiver
[  9]   0.00-10.00  sec   777 MBytes   651 Mbits/sec                  sender
[  9]   0.00-10.00  sec   776 MBytes   651 Mbits/sec                  receiver
[ 11]   0.00-10.00  sec   628 MBytes   526 Mbits/sec                  sender
[ 11]   0.00-10.00  sec   627 MBytes   526 Mbits/sec                  receiver
[ 13]   0.00-10.00  sec   694 MBytes   582 Mbits/sec                  sender
[ 13]   0.00-10.00  sec   694 MBytes   582 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  2.82 GBytes  2.42 Gbits/sec                  sender
[SUM]   0.00-10.00  sec  2.82 GBytes  2.42 Gbits/sec                  receiver

iperf Done.

wheatiesthecereal, 2020-11-05:
per my my thread / issue I am using the front USB port and have never seen it go past about 3Gbps

Perhaps I missed it in your post but what kind of drives do you have and what type of RAID / SHR are you using? 🤔

gordoabc, 2020-11-05:
I have Seagate Ironwolf drives in SHR along with an SSD read cache.  iperf3 doesn't access the hard drive so it shouldn't be impacted one way or the other.  The other things that is odd is the variability - I see between 1 and 3 Gbits/sec on this adaptor while on the 2.5 gbpe adaptor it was a very solid 2.35 GBits/sec

wheatiesthecereal, 2020-11-05:
**Iperf** 

D'oh ... I wasn't thinking 🤦🏽‍♂️

**USB NICs**

Also, what 2.5GbE were you using?  I had ordered the Realtek 8156- based Pluggable 2.GBE's but returned them in favor of these ACQ 5GGbE based models and never got to test them. If they are stable in function on both USB ports, consistent in bandwidth, and could be teamed that might have been the better option for my MPIO iSCSI based workloads 🤔 on well 🤷🏽‍♂️

It looks like the 2.5GbE Realtek repo is getting a lot more TL.🥰 I wonder if that is out of favor 🤔 or out of necessity 😅

gordoabc, 2020-11-05:
The 2.5 gbpe is Asustor | AS-U2.5G | USB3.2 Gen 1 Type-C to 2.5GBase-T (RJ-45 2.5 Gigabit LAN)

noblemd00, 2020-11-06:
> **Iperf**
>
> D'oh ... I wasn't thinking 🤦🏽‍♂️
>
> **USB NICs**
>
> Also, what 2.5GbE were you using? I had ordered the Realtek 8156- based Pluggable 2.GBE's but returned them in favor of these ACQ 5GGbE [...]

I use this 2.5 GbE USB 3.0 adapter: https://smile.amazon.com/gp/product/B07VNFLTLD on a second NAS
with good results. 

This is the Computer NIC I use: https://smile.amazon.com/gp/product/B07Y2GWVB8

bb-qq, 2020-11-29:
> UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1

How about changing MTU to 9000?