bb-qq/aqc111 #33: Receiving error "The operation failed. Please log in to DSM again and retry" when attempting to install spk file on DS918+

Schm1tty, 2020-12-04:
### Description of the problem

I have tried both releases and both give me the same error message "The operation failed. Please log in to DSM again and retry". I updated DSM and even attempted this with while logged in with the built in admin account, still getting the same results. I have "Any publisher" set in Trust Level.  I have a DS918+, and my device has an Apollo lake CPU (according to the Synology documentation), and I am trying to use the apollolake spk file. lsusb shows the qnap nic, so I know the system at least detects that it exists...any help is appreciated. Thanks in advanced!

Here is an image of the error:
https://imgur.com/fCv89QQ

### Description of your products

> - Synology DS918+
>     - Linux DataStore 4.4.59+ #25426 SMP PREEMPT Wed Jul 8 03:21:29 CST 2020 x86_64 GNU/Linux synology_apollolake_918+
>     - DSM Version 6.2.3-25426 Update 2
> - USB NIC - QNAP UC5G1T

### Description of your environment

> - QNAP NIC plugged into USB3 port on NAS

### Output of ``dmesg`` command

````
There are no logs for the attempted install showing in dmesg, surprisingly.
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008AA89E3A7D07)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 2IFs (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 07I11374)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:9E:84:69
          inet6 addr: fe80::211:32ff:fe9e:8469/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:20776 errors:0 dropped:0 overruns:0 frame:0
          TX packets:25770 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6204734 (5.9 MiB)  TX bytes:23352017 (22.2 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:9E:84:6A
          inet6 addr: fe80::211:32ff:fe9e:846a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6722 errors:0 dropped:0 overruns:0 frame:0
          TX packets:82 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1679586 (1.6 MiB)  TX bytes:21639 (21.1 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:13871 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13871 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1876931 (1.7 MiB)  TX bytes:1876931 (1.7 MiB)

ovs-syste Link encap:Ethernet  HWaddr 4E:26:B3:B8:B1:4B
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_bond0 Link encap:Ethernet  HWaddr 00:11:32:9E:84:69
          inet addr:192.168.1.40  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe9e:8469/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:20207 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14527 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:5890198 (5.6 MiB)  TX bytes:22728038 (21.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

Schm1tty, 2020-12-05:
I figured it out from a plex forum post. Apparently people are having trouble manually installing spk files that are downloaded using chrome??? I downloaded the spk on one of my vm's that still has the pre-chrome engine version of edge and I was able to upload the file from there. Weird.........

https://forums.plex.tv/t/manual-install-failure-operation-failed-please-log-in-to-dsm-and-try-again/618952/17

closed 2020-12-05 by Schm1tty

tholsan, 2021-07-05:
I have the same issue on DS916+ Braswell but I have tried every browser there is, unblocked the file in windows, placed the file on Diskstation, update all packages, restart DSM, set DNS to my router's ip, logged in from quickconnect and local, https and not, using full-acces admin. Nothing in dmesg. Installing through synopkg returns error 276, which I cannot find an explanation for, synopkg.log says nothing about it. Maybe I should have opened a new ticket for this? (Too bad since only one LAN port is not completely dead and that can only get 100-half-duplex no matter what.)

bb-qq, 2021-07-11:
Could you try to install other third-party packages? If this problem can be reproduced with various packages, the support desk would have a better answer.

By the way, [this post](https://community.synology.com/enu/forum/8/post/127205) indicates that the database of Package Manager is corrupted.

tholsan, 2021-07-19:
I found that all packages fail to install manually (sorry I didnt' test that first). So my issue is not related to this package and can be ignored. I opened a support ticket with Synology but since all packages they support can be installed automatically (which works for me), they would not help me. I found a [similar issue with a proposed solution](
https://www.reddit.com/r/synology/comments/a24d1u/universal_search_the_operation_failed_please_log/), but that didn't help. So when I get the time, I guess I will do a reinstall of DSM.

bb-qq, 2021-07-20:
Ok, if you find a solution, I would appreciate it if you could share it here for the other users.