bb-qq/aqc111 #34: support for v1000 arch

ccooney, 2020-12-21:
### Description of the problem

i checked the architecture list for my DS (DS1621+) - it's v1000 (amd embedded) and that isn't on the release list

### Description of your products

using a DS1621+, NIC is a QNAP qxg-10g1t

### Description of your environment

using a cat 6 connection to 10G switch

### Output of ``dmesg`` command

````
[    2.148827] [Firmware Bug]: ACPI: No _BQC method, cannot determine initial brightness
[    2.157110] [Firmware Bug]: ACPI: No _BQC method, cannot determine initial brightness
[    2.165600] ACPI: Power Resource [P0ST] (on)
[    2.169728] ACPI: Power Resource [P3ST] (on)
[    2.176763] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    2.182770] acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI]
[    2.191084] acpi PNP0A08:00: _OSC: OS now controls [PCIeHotplug PME AER PCIeCapability]
[    2.198910] acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-3f] only partially covers this bridge
[    2.209506] PCI host bridge to bus 0000:00
[    2.213468] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    2.220226] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    2.226989] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    2.234440] pci_bus 0000:00: root bus resource [mem 0x000c0000-0x000c3fff window]
[    2.241891] pci_bus 0000:00: root bus resource [mem 0x000c4000-0x000c7fff window]
[    2.249345] pci_bus 0000:00: root bus resource [mem 0x000c8000-0x000cbfff window]
[    2.256798] pci_bus 0000:00: root bus resource [mem 0x000cc000-0x000cffff window]
[    2.264251] pci_bus 0000:00: root bus resource [mem 0x000d0000-0x000d3fff window]
[    2.271705] pci_bus 0000:00: root bus resource [mem 0x000d4000-0x000d7fff window]
[    2.279158] pci_bus 0000:00: root bus resource [mem 0x000d8000-0x000dbfff window]
[    2.286612] pci_bus 0000:00: root bus resource [mem 0x000dc000-0x000dffff window]
[    2.294065] pci_bus 0000:00: root bus resource [mem 0x000e0000-0x000e3fff window]
[    2.301518] pci_bus 0000:00: root bus resource [mem 0x000e4000-0x000e7fff window]
[    2.308971] pci_bus 0000:00: root bus resource [mem 0x000e8000-0x000ebfff window]
[    2.316425] pci_bus 0000:00: root bus resource [mem 0x000ec000-0x000effff window]
[    2.323877] pci_bus 0000:00: root bus resource [mem 0xd0000000-0xf7ffffff window]
[    2.331331] pci_bus 0000:00: root bus resource [mem 0xfc000000-0xfed3ffff window]
[    2.338784] pci_bus 0000:00: root bus resource [bus 00-ff]
[    2.344253] pci 0000:00:00.0: [1022:15d0] type 00 class 0x060000
[    2.344376] pci 0000:00:00.2: [1022:15d1] type 00 class 0x080600
[    2.344532] pci 0000:00:01.0: [1022:1452] type 00 class 0x060000
[    2.344638] pci 0000:00:01.1: [1022:15d3] type 01 class 0x060400
[    2.344742] pci 0000:00:01.1: PME# supported from D0 D3hot D3cold
[    2.344800] pci 0000:00:01.1: System wakeup disabled by ACPI
[    2.350319] pci 0000:00:01.2: [1022:15d3] type 01 class 0x060400
[    2.350422] pci 0000:00:01.2: PME# supported from D0 D3hot D3cold
[    2.350477] pci 0000:00:01.2: System wakeup disabled by ACPI
[    2.355999] pci 0000:00:01.4: [1022:15d3] type 01 class 0x060400
[    2.356102] pci 0000:00:01.4: PME# supported from D0 D3hot D3cold
[    2.356158] pci 0000:00:01.4: System wakeup disabled by ACPI
[    2.361677] pci 0000:00:01.5: [1022:15d3] type 01 class 0x060400
[    2.361783] pci 0000:00:01.5: PME# supported from D0 D3hot D3cold
[    2.361837] pci 0000:00:01.5: System wakeup disabled by ACPI
[    2.367354] pci 0000:00:01.6: [1022:15d3] type 01 class 0x060400
[    2.367458] pci 0000:00:01.6: PME# supported from D0 D3hot D3cold
[    2.367512] pci 0000:00:01.6: System wakeup disabled by ACPI
[    2.373040] pci 0000:00:08.0: [1022:1452] type 00 class 0x060000
[    2.373146] pci 0000:00:08.1: [1022:15db] type 01 class 0x060400
[    2.373249] pci 0000:00:08.1: PME# supported from D0 D3hot D3cold
[    2.373295] pci 0000:00:08.1: System wakeup disabled by ACPI
[    2.378849] pci 0000:00:14.0: [1022:790b] type 00 class 0x0c0500
[    2.379027] pci 0000:00:14.3: [1022:790e] type 00 class 0x060100
[    2.379209] pci 0000:00:18.0: [1022:15e8] type 00 class 0x060000
[    2.379263] pci 0000:00:18.1: [1022:15e9] type 00 class 0x060000
[    2.379317] pci 0000:00:18.2: [1022:15ea] type 00 class 0x060000
[    2.379371] pci 0000:00:18.3: [1022:15eb] type 00 class 0x060000
[... 983 lines cut ...]
eth3      Link encap:Ethernet  HWaddr 00:11:32:D8:51:EF
          inet addr:169.254.56.197  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:87 base 0xa000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2399321 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2399321 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:341214356 (325.4 MiB)  TX bytes:341214356 (325.4 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2021-01-04:
I uploaded the driver for the v1000 platform.
https://github.com/bb-qq/r8152/releases/download/2.14.0-2/r8152-v1000-2.14.0-2.spk

I would appreciate it if you report whether it works or not.

ccooney, 2021-01-20:
i tried loading it and got the result

> 2021-01-19T16:25:04-08:00 diskydisk kernel: [ 1820.174836] r8152: version magic '4.4.59+ SMP preempt mod_unload ' should be '4.4.59+ SMP mod_unload '

i suspect it's not finding the device

bb-qq, 2021-02-20:
It seems to be an issue on the DSM toolkit side. I applied a workaround to my build environment and rebuild the package.
Could you retry this package?
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-v1000-1.3.3.0-2a.spk

zachberger, 2021-03-15:
@bb-qq could you share the workaround? I'm struggling to build a different kernel module due to this. I've also filed SynologyOpenSource/pkgscripts-ng#30 upstream to fix the toolkit.

zachberger, 2021-03-18:
A new toolkit was released today that fixes this. Please try rebuilding with the latest toolkit.

bb-qq, 2021-03-21:
@zachberger 
Thank you for reporting the issue to Synology. It is very helpful. I have hand-edited the "config.h" for a workaround, but it seemed not to work... 

The new package is here.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-v1000-1.3.3.0-2b.spk

kudos, 2021-04-18:
I can confirm it's working for me on a DS1821+.

kudos, 2021-04-18:
Clarification, it only works if I stop and start the service after plugging in the nic, or booting server.

bb-qq, 2021-05-29:
Thank you for your confirmation. I close this issue.
Please create a new issue if the driver has other problems.

closed 2021-05-29 by bb-qq