bb-qq/aqc111 #35: Support for RS816

scalponi, 2021-01-27:
### Description of the problem

I can't find a Package for MARVELL Armada 385. Can you compile a package for this?

### Description of your products

- Synology RS816 
- Linux rackstation01 3.10.105 #25426 SMP Mon Dec 14 18:44:29 CST 2020 armv7l GNU/Linux synology_armada38x_rs816
- DSM version6.2.3-25426 Update 3
- QNAP-UC5G1T

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
  |__2-1         1058:25a3:1013 00  3.00 5000MBit/s 8mA 1IF  (Western Digital Elements 25A3 575832314432374653325055)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.105 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
  |__4-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 2IFs (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 0AI18305)
|__usb5          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd f1058000.usb) hub

````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX
          inet addr:192.168.2.30  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe60:3dd1/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:1245332718 errors:0 dropped:0 overruns:0 frame:0
          TX packets:731360315 errors:0 dropped:11 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1306607119 (1.2 GiB)  TX bytes:2757265716 (2.5 GiB)

eth0      Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:104194192 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11759112 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:485110825 (462.6 MiB)  TX bytes:600481583 (572.6 MiB)
          Interrupt:193

eth1      Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:1141138526 errors:0 dropped:0 overruns:0 frame:0
          TX packets:719601203 errors:0 dropped:11 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:821496294 (783.4 MiB)  TX bytes:2156784133 (2.0 GiB)
          Interrupt:194

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6690955 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6690955 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:846708609 (807.4 MiB)  TX bytes:846708609 (807.4 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2021-02-23:
I created the driver for armada38x based platforms. Please try it.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-armada38x-1.3.3.0-2.spk

scalponi, 2021-02-23:
Thank you! I can install the driver, the link between switch (EdgeSwitch 48 750W) and QNAP-UC5G1T cames up: 1 GBIT is shown in switch-dashboard and also by the leds of QNAP-UC5G1T. The driver starts in DSM without errors. In info-center is 'LAN3' shown with the correct MAC-address. BUT no link ist established in DSM:

ethtool shows none "PHYAD"/mac-address
````
ethtool eth2
Settings for eth2:
        Supported ports: [ TP MII ]
        Supported link modes:   100baseT/Full
                                1000baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: Unknown!
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: g
        Wake-on: d
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: no
````
ip link shows correct mac-address (I've replaced the original mac-address with zz:zz:zz:zz:zz:zz), but also no link.
````
ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP> mtu 1500 qdisc mq master bond0 state UP mode DEFAULT qlen 1024
    link/ether zz:zz:zz:zz:zz:zzbrd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP> mtu 1500 qdisc mq master bond0 state UP mode DEFAULT qlen 1024
    link/ether zz:zz:zz:zz:zz:zzbrd ff:ff:ff:ff:ff:ff
4: sit0: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT
    link/sit 0.0.0.0 brd 0.0.0.0
5: bond0: <BROADCAST,MULTICAST,MASTER,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT
    link/ether 00:11:32:60:3d:d1 brd ff:ff:ff:ff:ff:ff
47: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT qlen 1000
    link/ether zz:zz:zz:zz:zz:zzbrd ff:ff:ff:ff:ff:ff
````
With same cables and connections on a windows everything works fine. Could it be a version-problem between ethtools and usbnet.c?

RobbieTT, 2021-02-24:
Did you try stopping and restarting the package in Package Center?

scalponi, 2021-02-24:
Yes, I've, many times...
By stopping of the package all leds of QNAP-UC5G1T gone black and by starting the leds came back - with the correct color for 1000 MBit- connection (or by using a 10GBIT-Switch the correct Color for 5 GBIT). 
So the package/driver can start and stop the usb-networkadapter and read the MAC, but no other informations like linkstatus or statistics.

bb-qq, 2021-02-26:
* Could you try to detach the external HDD? `  |__2-1         1058:25a3:1013 00  3.00 5000MBit/s 8mA 1IF  (Western Digital Elements 25A3 575832314432374653325055)`
* Does dmesg output show something after attaching the dongle?

scalponi, 2021-02-26:
New try: 
1. Package aqc111 installed.
2. external hd dismounted and physically disconnected.
3. Package aqc111 restarted.
4. Use the other usb-port for QNAP-UC5G1T (here comes errors in dmesg. Time: 132439.480441 to 132439.580127)
5. Package aqc111 restarted.
Nothing has changed :-(

dmesg:
````
[131368.150407] aqc111 4-1:1.0 eth2: register 'aqc111' at usb-f10f8000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, zz:zz:zz:zz:zz:zz
[131368.162297] usbcore: registered new interface driver aqc111
[131368.513191] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[131402.213208] EXT4-fs (sdq1): re-mounted. Opts: (null)
[131459.690622] usb 2-1: USB disconnect, device number 2
[131569.992834] usbcore: deregistering interface driver aqc111
[131569.998450] aqc111 4-1:1.0 eth2: unregister 'aqc111' usb-f10f8000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[131579.370670] aqc111 4-1:1.0 eth2: register 'aqc111' at usb-f10f8000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, zz:zz:zz:zz:zz:zz
[131579.382535] usbcore: registered new interface driver aqc111
[131579.711377] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[132439.475166] usb 4-1: USB disconnect, device number 2
[132439.480441] aqc111 4-1:1.0 eth2: unregister 'aqc111' usb-f10f8000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[132439.490513] aqc111 4-1:1.0 eth2: Failed to read(0x1) reg index 0x0002: -19
[132439.497514] aqc111 4-1:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
[132439.504598] aqc111 4-1:1.0 eth2: Failed to write(0x1) reg index 0x0002: -19
[132439.511674] aqc111 4-1:1.0 eth2: Failed to write(0x61) reg index 0x0000: -19
[132439.562291] aqc111 4-1:1.0 (unregistered net_device): Failed to write(0x1) reg index 0x0002: -19
[132439.571204] aqc111 4-1:1.0 (unregistered net_device): Failed to write(0x1) reg index 0x0002: -19
[132439.580127] aqc111 4-1:1.0 (unregistered net_device): Failed to write(0x61) reg index 0x0000: -19
[132442.412443] usb 2-1: new SuperSpeed USB device number 3 using xhci-hcd
[132442.433396] usb 2-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[132442.454390] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-f10f0000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, zz:zz:zz:zz:zz:zz
[132783.416994] usbcore: deregistering interface driver aqc111
[132783.422623] aqc111 2-1:1.0 eth2: unregister 'aqc111' usb-f10f0000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[132793.394795] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-f10f0000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, zz:zz:zz:zz:zz:zz
[132793.406663] usbcore: registered new interface driver aqc111
[132793.807895] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready

````
After reconnect on the 2nd usb-port I found someting in messages:
````
2021-02-26T08:14:15+01:00 rackstation01 kernel: [132442.433396] usb 2-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
````
Could it be a problem with a missing latency info? The external harddisk use an own power supply.

bb-qq, 2021-02-27:
It might be that the root hub of RS816 does not supply enough power to work for the dongle. Using a self-powered USB hub with an AC adapter may resolve the issue, but I don't have confidence.

scalponi, 2021-02-27:
I have tried y-cable and self-powered USB hub - both doesn't help.

The usb-networkadapter has the latest firmware 3.1.6 installed. In the release-notes I found:

````
FW v.3.1.6
  Enhancements:
    * [FIJI-353] Improved power consumpsion. LEDs are disabled with unpluged cable.
  Bug fix:
    * [FIJI-349] Fixed issue when firmware report incorrect link status to the driver`
````

Maybe the firmware doesn't compatible to the current driver for kernel 3.10? 
I found a way to compile the package by myself and trying now debugging.

bb-qq, 2021-03-05:
The driver should be compatible with firmware 3.1.6.
This issue may be a problem with the USB host controller or kernel version.

bb-qq, 2024-06-15:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

If this is an auto-negotiation problem, this procedure may solve it.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#disable-auto-negotiation

closed 2024-06-15 by bb-qq