bb-qq/aqc111 #36: Synology RS819 - Realtek RTD1296

RobbieTT, 2021-02-20:
Is there a package that works with the RS819 and its Realtek RTD1296 cpu?

[I did look, honest, so happy to be forcibly pointed in the right direction]

Regards to all.

bb-qq, 2021-02-23:
I uploaded the driver for RTD1296 platforms.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-rtd1296-1.3.3.0-2.spk

> Incidentally I got this running on a colleagues NAS (already supported) and found the performance to be erratic. Looking in more detail with a USB analyser and saw the voltage drop when transfer speed and current draw increased until a momentary drop completely with the speed then building again (a sawtooth pattern). Issue cured by using a Y-type USB adapter to provide power from 2 USB ports. No more throttled speed or sawtooth transfer pattern.

Thank you for the information. I will share this tip in other threads.

RobbieTT, 2021-02-23:
@bb-qq Thank-you and I should add that the current limit seems to vary between Synology models and even between USB3 ports on the same units (eg front vs rear), so if in doubt trying a different USB3 port is a valid option.

closed 2021-02-23 by RobbieTT

marcosscriven, 2021-05-14:
Can I ask if you tried the RS819 package @RobbieTT? Thinking of trying this on DS220j.

RobbieTT, 2021-05-14:
@marcosscriven My efforts were thwarted by my multi-gig switch dying and no replacement available from UniFi / UI.  Being stuck on 1GbE is a painful experience when 10GbE had become the norm.