bb-qq/aqc111 #37: In depth testing on Synology apollolake_1019+ Linux msp 4.4.59+

dornquast, 2021-02-22:
### Description of the problem

Tested on my synology with 4 different 1, 2.5, 5, 10Gbps switches.
Out of the box with the aqc111 driver installed, it only supports 100Mbps.
If you turn off thermal throttling, you can get it to connect at 1000Mbps 
(However it was new out of box, in cool room, with no data flowing so temperature could not be issue)

Attempting to set speed to 2500 results in no speed change, and removing autonegotiation prevents link.

### Description of your products

> - Synology 1019+
>     - Linux msp 4.4.59+ #25426 SMP PREEMPT Tue May 12 04:54:55 CST 2020 x86_64 GNU/Linux synology_apollolake_1019+
>     - DSM version 6.2.3-25426
> - Sonnet Solo5G USB-C to 5 GBASE-T Ethernet Adapter

### Description of your environment
> - connect directly to a PC or using a hub
>     - PC: N/A - can't get it to connect to switch at right speed.
>     - Hub: Unifi Dream Machine PRO with 10Gtek SFP+ adapter 
>     - Hub: Linksys GS110EMX
> - Cat 6.  2 Meter.

### Output of ``dmesg`` command

````
ethtool eth2
Settings for eth2:
	Supported ports: [ TP MII ]
	Supported link modes:   100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Speed: 100Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: g
	Wake-on: d
	Current message level: 0x00000007 (7)
			       drv probe link
	Link detected: yes

````

````
ethtool --show-priv-flags eth2
Private flags for eth2:
Low Power 5G      : off
Thermal throttling: on
ethtool -i eth2 #driver info
driver: aqc111
version: 1.3.3.0
firmware-version: 3.1.6
bus-info: usb-0000:00:15.0-1
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes

````

````
ethtool -s eth2 speed 5000 duplex full autoneg on
Cannot advertise speed 5000 duplex full
````

bb-qq, 2021-02-23:
It seems to be an auto-negotiation issue. Did these combinations of the dongle and the switches work correctly with your PC? I would like to know about behavior when the dongle is attached to the PC with the official windows driver.

dornquast, 2021-02-24:
Well, I spent some time with this device.  Frankly it's not all that impressive.   Their mac drivers are horrible - (E.g. it's connecting at 5Gbps to switch automatically but reporting 1Gbps to OS and preventing me from change MTU to 9000).. Also despite 5Gbps connection, speed tops out at around 2.26 receive.  

I'm concerned about heat.. the unit is super hot.   So I'm going to buy another product and see if that performs better.
I cannot at this time recommend the Sonnet Solo5G USB-C to 5 GBASE-T Ethernet Adapter.

Doesn't work with their own driver form website on mac, doesn't work with your driver.

````
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   270 MBytes  2.26 Gbits/sec                  
[  5]   1.00-2.00   sec   269 MBytes  2.25 Gbits/sec                  
[  5]   2.00-3.00   sec   263 MBytes  2.21 Gbits/sec                  
[  5]   3.00-4.00   sec   260 MBytes  2.18 Gbits/sec                  
[  5]   4.00-5.00   sec   264 MBytes  2.21 Gbits/sec                  
[  5]   5.00-6.00   sec   262 MBytes  2.20 Gbits/sec                  
[  5]   6.00-7.00   sec   264 MBytes  2.22 Gbits/sec                  
[  5]   7.00-8.00   sec   266 MBytes  2.23 Gbits/sec                  
[  5]   8.00-9.00   sec   267 MBytes  2.24 Gbits/sec                  
[  5]   9.00-10.00  sec   269 MBytes  2.26 Gbits/sec                  
[  5]  10.00-10.00  sec   440 KBytes  2.27 Gbits/sec   
````

dornquast, 2021-03-01:
WOW!  It's been an adventure.   TLDR; this driver isn't reliable / doesn't work with my synology.

Over the past few days I installed linux clean on a stand alone PC and purchased 2 other AQC based drivers.   I now own a StarTech, a QNA, and the Sonnet solo5g.

On my clean linux build with latest ACQ drivers.. I tested with iPerf3, letting it move terabytes of data up and down to test how reliable the driver was or the unit would experience thermal shutdown or slowdown.  None of the units experienced thermal shutdown and all of them performed +1/-1 a few %.  Basically, it's the same interface minus physical characteristics.

Personally I think the startech looks the best as it has venting holes for cooling.  The runner up is QNA because it's a giant thermal heat sync (AND DOES GET HOT).

I digress, as it was I can reliably sustain 3.59Gbps TO the USB device.  When the USB device is transmitting, it runs 2.66Gbps.  This is all on my little linux box.  (NOTE:  9000 MTU as that helps performance)

All I do is then move the USB ethernet device to my synology, reboot it, and try and use it.   Suddenly the very same device can't auto-negotiate reliably, and when it does connect at 5Gbps (auto), it will fail within 2-3 minutes. 

The ONLY change is my linux host with driver that comes with kernel and your driver on synology.

This has lead me to be 100% sure it's not working on my model of synology.  Anything else I can provide? Would really like this to work!

~Matthew

dornquast, 2021-03-04:
Following up on an observation, I compared the linux kernel's AQC111 usb driver and yours, they are significantly different which confirms it's probably software.  It looks like you've significantly refactored the negotiation and connection logic.

All 3 hardware interfaces I purchased for testing based on AQC111 worked with Linux system I built but do not work on Synology with driver you built.

What was the reason for refactoring/rewriting the connection logic for driver?

bb-qq, 2021-03-05:
Thank you for your investigation.
Did you try the front port? Some users reported stability issues when they use rear ports.

> What was the reason for refactoring/rewriting the connection logic for driver?

I did not touch the logic of codes anything such as initialization or negotiation. I just port official driver that can be downloaded from the website to the Synology platform.

dornquast, 2021-03-05:
That was the front port.   I have not tested with rear.

I thought you ported official driver from Linux - do you have link to driver you ported?   The driver in linux kernel is from the AQC folks.

bb-qq, 2021-03-21:
I downloaded the driver from the Aquantia Official website. Aquantia has now been acquired by Marvell, so you can download the same one from here.
https://www.marvell.com/support/downloads.html

And you can find the commit of that driver from here.
https://github.com/bb-qq/aqc111/commit/64a17cd38d79508307c12d8e2518f3ac09f6a1b9

marcosscriven, 2021-05-16:
Interesting - I wonder if it would be worth trying the code from mainline Linux instead?

did you get something working @dornquast?

dornquast, 2021-05-16:
No.  I tested it with Linux and worked great, but this driver does not work with my synology.

TGSAN, 2021-08-10:
Same issue in apollolake.

bb-qq, 2021-09-24:
Hmm, it's weird. Currently, I am using QNA-UC5G1T connected to the front port of DS918+ at the production environment and I have not encountered any issues so far.

And according to the report in ticket #23, there are some environments where the driver works without any problems, so I have no idea what the difference is between those environments and those where it doesn't work...

bb-qq, 2024-06-15:
Even now, the drivers officially released by Marvell have not been updated. I am considering using the driver that came with Linux, as reported in this thread, but have not gotten around to it due to lack of time.
I'm also wondering why it works consistently in my environment so far.