bb-qq/aqc111 #38: What spk to use for bromolow architecture?

sbsrouteur, 2021-03-03:
### Description of the problem

> Would like to use an AQC111 on our RS3614xs, but arch bromolow does not seem supported. Is there a compatible driver? If not what is required to build one for Bromolow arch?

### Description of your products

> - the product name of your NAS model
RS3614xs
>     - and output of ``uname -a`` command
Linux DS 3.10.102 #15132 SMP Thu Jun 29 07:35:20 CST 2017 x86_64 GNU/Linux synology_bromolow_rs3614xs
>     - DSM version x.x
DSM 6.1.2-15132
> - the product name of the ethernet dongle
(Aquantia Aquantia 5G USB Ethernet Adapter 0000006c
### Description of your environment

> - connect directly to a PC or using a hub
DSM behind switch DLink DGS48
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)
Cat 5e

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.0) hub
  |__1-1         8087:8008:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.2     2eca:c101:0101 00  2.10  480MBit/s 500mA 2IFs (Aquantia Aquantia 5G USB Ethernet Adapter 0000006c)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.0) hub
  |__2-1         8087:8000:0005 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa0) hub
    |__2-1.3     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500456D8D915266)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub

````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:32:DD:71
          inet addr:10.32.11.249  Bcast:10.32.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9759265789 errors:0 dropped:95009 overruns:95009 frame:0
          TX packets:14164592319 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6833945891018 (6.2 TiB)  TX bytes:17932592349524 (16.3 TiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:32:DD:72
          inet addr:169.254.80.85  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:32:DD:73
          inet addr:169.254.237.243  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 00:11:32:32:DD:74
          inet addr:169.254.21.235  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:22323347 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22323347 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3287819995 (3.0 GiB)  TX bytes:3287819995 (3.0 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2021-03-05:
I created the driver for bromolow platforms. Please try it.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-2/aqc111-bromolow-1.3.3.0-2.spk

sbsrouteur, 2021-03-05:
Hello,

Package installed & started. Seems OK. synopkg status reports running status [255].

I did inconfig to up the sit0 interface and affect an IPV4 address.

No led show up on the interface, nor the switch on the other end of the ethernet cable. interface is not listed in DSM control panel.

Tried to unplug/replug the USB device.

Is a reboot required?

inconfig shows a weird encap, not ethernet : 

```
root@DS:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:11:32:32:DD:71
          inet addr:10.32.11.249  Bcast:10.32.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9817068324 errors:0 dropped:95670 overruns:95670 frame:0
          TX packets:14239753703 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6872918182734 (6.2 TiB)  TX bytes:18026618406287 (16.3 TiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:32:DD:72
          inet addr:169.254.80.85  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:32:DD:73
          inet addr:169.254.237.243  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 00:11:32:32:DD:74
          inet addr:169.254.21.235  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:22647551 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22647551 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3335710319 (3.1 GiB)  TX bytes:3335710319 (3.1 GiB)

sit0      Link encap:IPv6-in-IPv4
          inet addr:10.32.11.248  Mask:255.255.0.0
          inet6 addr: ::10.32.11.249/96 Scope:compatGlobal
          inet6 addr: ::10.32.11.248/96 Scope:compatGlobal
          inet6 addr: ::169.254.80.85/96 Scope:compatGlobal
          inet6 addr: ::127.0.0.1/96 Scope:compatHost
          inet6 addr: ::169.254.21.235/96 Scope:compatGlobal
          inet6 addr: ::169.254.237.243/96 Scope:compatGlobal
          UP RUNNING NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

```

sbsrouteur, 2021-03-05:
inquire a little more and found that the aqc111 folder in syno package folder is empty.

Manually copying the ko files to /lib/modules and trying inmod mii.ko gets me the following error : 
root@DS:/usr/syno/etc/packages# insmod /lib/modules/mii.ko
insmod: ERROR: could not insert module /lib/modules/mii.ko: Invalid module format

What am I missing here?

sbsrouteur, 2021-03-10:
Ok, for the record, the drivers in the release are for DSM6.2 I am on DSM 6.1. thus the issues that the package would not load.
I have rebuilt the drivers via the DSM6.1 tool chain and the driver now loads.

I will post here the performance findings, once the 5G switches have been installed.

sbsrouteur, 2022-03-01:
Adding to my previous post, because upon reboot, the driver did not load and restart.
In order to restart the network I had to 
```
cd /lib/modules
insmod mii.ko
insmod usbnet.ko
insmod aqc111.ko

ifconfig eth4 up

then configure the network interface via the GUI.
```

BUut did not activate 5000 speed (yet)

bb-qq, 2022-08-11:
Recent releases might fix your issue.

closed 2024-06-15 by bb-qq