bb-qq/aqc111 #39: Driver issue with UGREEN 5GBE Adapter

ilovemissjni, 2021-03-13:
I just brought the ugreen 5gbe adapter from china. When I plug into my 920+, I can't see LAN 3 with driver version 1.3.3.0-2.

I had asked google.com, he said this adapter use the same chip with UC5G1T.

Is it possible to support this adapter?

THX!!!

bb-qq, 2021-05-29:
Could you paste the result of `lsusb`?

ilovemissjni, 2021-05-29:
> Could you paste the result of `lsusb`?

`/$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000B5B2A13F201)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         2eca:c101:0101 00  3.20 5000MBit/s 896mA 2IFs (Aquantia Aquantia 5G USB Ethernet Adapter 00000072)
ilovemissjni@ilovemissjni:/$
`

bb-qq, 2021-07-11:
I guess that driver is working correctly, but NIC is not configured properly due to the DHCP-related bug in my driver package.
Using  1.3.3.0-3 will resolve the issue.

bb-qq, 2022-08-11:
I close the issue due to no response. Please feel free to re-open the issue.

closed 2022-08-11 by bb-qq