bb-qq/aqc111 #41: DS620Slim - Report - Working correctly with 3.2Gbps file transfer read performance

alexissusset, 2021-04-23:
Hello everyone
I am happy to report that the DS620Slim NAS works well with QNA-UC5G1T Adapter

Many thanks to @bb-qq for creating the drivers, this kind of performance on such as small NAS wouldn't be possible without it!

### Equipment list
## NAS
Synology DS620Slim

# RAM
Original 2GB RAM
SSD Cache (Read-only)

# Storage and Volumes
VOLUME 1 - 2 x SAMSUNG SSD 870 QVO 4TB
** SHR
** File system (NAS): Btrfs
** Encryption: None

# Networking
QNAP QNA-UC5G1T USD 3 Adapter at 5Gbps and MTU 1500

## Client
M1 MacBook Air 

# Networking
QNAP QNA-T310G1S Thunderbolt 3 Adapter at 10Gbps SFP+ Fiber and MTU 1500 (More on MTU bellow)

## Network
# Switch
Ubiquiti UniFi Switch Aggregation 8x 10Gbps SFP+ ports

# Modules
NAS connected to Wiitek 10Gbps SFP+ module with link negotiated at 5Gbps (But Ubiquiti Switch shows it at 10Gbps as it doens't officially support 5Gbps, more on that bellow)
Client connected to Ubiquiti 10Gbps SFP+ module with link negotiated at 10Gbps

### Test results
While tranferring a 50GB file over AFP
165MB/s Write
440MB/s Read

While testing with Blackmagic Disk Speed Test over AFP
101MB/s Write
390MB/s Read

### Notes
There are couple of small issues here and there but nothing critical and these are likely related to Ubiquiti's lack of official 5Gbps support:
- When chaging MTU settings on the QNAP network card, the NAS looses network connection until it is rebooted
- When running MTU at 9000 the performance is in fact degraded to:
 28MB/s Write
400MB/s Read
This is likely due to Ubiquiti trying to send data at 10Gbps rate but hitting the NAS SFP+ at 10Gbps while it's in fact running at 5Gbps

### Further testing
I have ordered another set of QNAP 5Gbps and Wiitek SFP module so that I can test and deploy either of the following:
both NAS and MacBook with 5Gbps networking
NAS with 2x 5Gbps Aggregate network with MacBook at 10Gbps Fiber networking

Will report back once I have completed these tests

alexissusset, 2021-05-08:
# Further testing report
I've now tested with 2x QNAP QNA-UC5G1T bond with LACP mode, the bonding works well but it needs to be configured after the NAS has started and is connected, here's the process:
1. Start the NAS, wait for it to be connected on LAN1 or LAN2 and have eth2 (LAN3) and eth3 (LAN4) online
2. Enable OVS
3. Create bond with LAN3 and LAN4
4. Your NAS now be connected with LACP bond

With this setup I still couldn't get past 3.5Gbps speed even when transferring multiple files at the same time. Will keep testing and report back if there's any way to improve speed

[image]

alexissusset, 2021-05-08:
# Setting up and automating USB Ethernet bond during reboot

Here is the proces to make sure USB Ethernet bond works correctly even after rebooting 
1. Once everything is running on your NAS, enable OVS in: Control Panel > Network > Network Interface > Manage > Open vSwitch Settings
2. Configure your switch how you wish, LACP or other
3. Create bond with LAN 3 and LAN 4 interfaces, let it connect and verify that it works, in my case I used LACP
4. Create a task to restart network service in: Control Panel > Task Scheduler, click Create, and select Triggered Task.
4.1 Run it on Boot-up Event, Run the command in the User-defined script `/etc/rc.network restart`
5. Disconnect LAN 1 and/or LAN 2 cables (Otherwise they will be registered as default interface and bonjour hostname during reboot)
6. Reboot the NAS

[image]
[image]

jocamero, 2021-09-11:
@alexissusset , were you ever able to get speeds greater than ~3.5Gbps? I have a similar setup as you and I'm wondering if the Ubiquity Agg switch is the issue.

alexissusset, 2021-09-12:
@jocamero this is the best I managed to test with this setup.
It may indeed be a limitation due to Ubiquiti

As a reference I managed to get 9.2Gbps between my computer and Ubiquiti UDM Pro through the 10Gbps switch aggregate