bb-qq/aqc111 #42: DS1813 + with QNAP QNA-UC5G1T : lost connection

higs76, 2021-05-07:
### Description of the problem

I'm french, so sorry for my english.
Qnap connection lost after a time, no ping. in Synology "Connected"
i've set fixed IPv4, jumbo frame to 9000, like other LAN. and it's strange.
desactivate IPV6, but i've got Ipv6 adress (in interface & you see in bottom of this page)

i've got cedarview drivers.

Qnap Lan worked Up/Down, a time and stop.
the same after reboot and not used (using one of my internal LAN)

### Description of your products

Synology DS1813+

and output of uname -a command :
Linux DS1813 3.10.105 #25556 SMP Thu Mar 18 12:51:35 CST 2021 x86_64 GNU/Linux synology_cedarview_1813+
DSM Version : 6.2.4-25556

the product name of the ethernet dongle : QNAP QNA-UC5G1T (firmware 3.1.6)

### Description of your environment

DS1813+  & DS414 > Netgear XS508M (bottom house) > Netgear XS508M (top house) > PC1 10Gbe & PC2 10Gbe ...

XS508M is not manageable and RMA changed to have no bug jumbo frame

cable category : Cat.6a & Cat.7

Router is "a ISP french box : Freebox Delta", connection : 10Gbps plugged on XS508M SFP+

### Output of ``dmesg`` command

````
[    0.000000] efi: mem06: type=7, attr=0xf, range=[0x0000000000100000-0x000000000151d000) (20MB)
[    0.000000] efi: mem07: type=4, attr=0xf, range=[0x000000000151d000-0x0000000001900000) (3MB)
[    0.000000] efi: mem08: type=7, attr=0xf, range=[0x0000000001900000-0x000000007f38a000) (2010MB)
[    0.000000] efi: mem09: type=2, attr=0xf, range=[0x000000007f38a000-0x0000000080000000) (12MB)
[    0.000000] efi: mem10: type=7, attr=0xf, range=[0x0000000080000000-0x00000000de736000) (1511MB)
[    0.000000] efi: mem11: type=10, attr=0xf, range=[0x00000000de736000-0x00000000de73e000) (0MB)
[    0.000000] efi: mem12: type=0, attr=0xf, range=[0x00000000de73e000-0x00000000de746000) (0MB)
[    0.000000] efi: mem13: type=4, attr=0xf, range=[0x00000000de746000-0x00000000de754000) (0MB)
[    0.000000] efi: mem14: type=0, attr=0xf, range=[0x00000000de754000-0x00000000de755000) (0MB)
[    0.000000] efi: mem15: type=4, attr=0xf, range=[0x00000000de755000-0x00000000de758000) (0MB)
[    0.000000] efi: mem16: type=0, attr=0xf, range=[0x00000000de758000-0x00000000de759000) (0MB)
[    0.000000] efi: mem17: type=10, attr=0xf, range=[0x00000000de759000-0x00000000de75a000) (0MB)
[    0.000000] efi: mem18: type=4, attr=0xf, range=[0x00000000de75a000-0x00000000de769000) (0MB)
[    0.000000] efi: mem19: type=5, attr=0x800000000000000f, range=[0x00000000de769000-0x00000000de76b000) (0MB)
[    0.000000] efi: mem20: type=4, attr=0xf, range=[0x00000000de76b000-0x00000000df0b5000) (9MB)
[    0.000000] efi: mem21: type=10, attr=0xf, range=[0x00000000df0b5000-0x00000000df0b8000) (0MB)
[    0.000000] efi: mem22: type=3, attr=0xf, range=[0x00000000df0b8000-0x00000000df0bd000) (0MB)
[    0.000000] efi: mem23: type=9, attr=0xf, range=[0x00000000df0bd000-0x00000000df0c1000) (0MB)
[    0.000000] efi: mem24: type=4, attr=0xf, range=[0x00000000df0c1000-0x00000000df0cd000) (0MB)
[    0.000000] efi: mem25: type=9, attr=0xf, range=[0x00000000df0cd000-0x00000000df0d0000) (0MB)
[    0.000000] efi: mem26: type=10, attr=0xf, range=[0x00000000df0d0000-0x00000000df0d1000) (0MB)
[    0.000000] efi: mem27: type=9, attr=0xf, range=[0x00000000df0d1000-0x00000000df0d2000) (0MB)
[    0.000000] efi: mem28: type=10, attr=0xf, range=[0x00000000df0d2000-0x00000000df0d3000) (0MB)
[    0.000000] efi: mem29: type=3, attr=0xf, range=[0x00000000df0d3000-0x00000000df0e2000) (0MB)
[    0.000000] efi: mem30: type=4, attr=0xf, range=[0x00000000df0e2000-0x00000000df0e4000) (0MB)
[    0.000000] efi: mem31: type=3, attr=0xf, range=[0x00000000df0e4000-0x00000000df0ea000) (0MB)
[    0.000000] efi: mem32: type=4, attr=0xf, range=[0x00000000df0ea000-0x00000000df0ec000) (0MB)
[    0.000000] efi: mem33: type=3, attr=0xf, range=[0x00000000df0ec000-0x00000000df0ed000) (0MB)
[    0.000000] efi: mem34: type=4, attr=0xf, range=[0x00000000df0ed000-0x00000000df0ee000) (0MB)
[    0.000000] efi: mem35: type=3, attr=0xf, range=[0x00000000df0ee000-0x00000000df0f3000) (0MB)
[    0.000000] efi: mem36: type=4, attr=0xf, range=[0x00000000df0f3000-0x00000000df0f7000) (0MB)
[    0.000000] efi: mem37: type=3, attr=0xf, range=[0x00000000df0f7000-0x00000000df0fc000) (0MB)
[    0.000000] efi: mem38: type=4, attr=0xf, range=[0x00000000df0fc000-0x00000000df0fd000) (0MB)
[    0.000000] efi: mem39: type=3, attr=0xf, range=[0x00000000df0fd000-0x00000000df10b000) (0MB)
[    0.000000] efi: mem40: type=4, attr=0xf, range=[0x00000000df10b000-0x00000000df10d000) (0MB)
[... 1596 lines cut ...]
eth4      Link encap:Ethernet  HWaddr 24:5E:BE:54:18:29
          inet addr:192.168.1.4  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe54:1829/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:2272315 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9983174607 (9.2 GiB)  TX bytes:5996280504 (5.5 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:318837 errors:0 dropped:0 overruns:0 frame:0
          TX packets:318837 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:37623750 (35.8 MiB)  TX bytes:37623750 (35.8 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2021-05-16:
Could you try to detach the USB cable of UPS? Also changing the USB port of the ethernet dongle might help.

higs76, 2021-05-22:
Remove my UPS (it' was on usb 2), DS1813+ have 4usb 2 ports and 2 usb 3 ports.

dongle is on usb3 port.

and i've the same thing : 

[image]
[image]

2oxygen2, 2021-06-28:
I have a similar issue with a rs2214rp+, using cedarview. works okay after starting, but transfer some data and then it goes down until the driver is stopped and started. what logs would help?

bb-qq, 2024-06-15:
Is the situation the same with the latest drivers?
The contents of dmesg are sufficient for logging. However, the UPS is giving a lot of errors in the log posted at the first, so it would be helpful to have another one.