bb-qq/aqc111 #43: Error installing "mii: exports duplicate symbol generic_mii_ioctl (owned by kernel)"

marcosscriven, 2021-05-16:
### Description of the problem

When installing the rtd2196 driver, I see this in the logs:

```
[54241.874719] mii: module verification failed: signature and/or required key missing - tainting kernel
[54241.884140] mii: exports duplicate symbol generic_mii_ioctl (owned by kernel)
[54241.989801] usbcore: registered new interface driver aqc111
```

### Description of your products

Synology DS220j
Linux nas 4.4.59+ #25556 SMP Thu Mar 4 17:52:53 CST 2021 aarch64 GNU/Linux synology_rtd1296_ds220j
DSM 6.2.4-25556
QNAP QNA-UC5G1T

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
[54241.874719] mii: module verification failed: signature and/or required key missing - tainting kernel
[54241.884140] mii: exports duplicate symbol generic_mii_ioctl (owned by kernel)
[54241.989801] usbcore: registered new interface driver aqc111
````

### Output of ``lsusb`` command

````
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
  |__2-1         0764:0501:0001 00  1.10  1.5MBit/s 50mA 1IF  (CPS Amazon550 84b184ea85ff9ef3b2)
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:E4:92:A9
          inet addr:192.168.1.204  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fee4:92a9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:474247 errors:0 dropped:0 overruns:0 frame:0
          TX packets:300562 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:97322145 (92.8 MiB)  TX bytes:77442719 (73.8 MiB)
          Interrupt:11

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:33283 errors:0 dropped:0 overruns:0 frame:0
          TX packets:33283 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2294694 (2.1 MiB)  TX bytes:2294694 (2.1 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

marcosscriven, 2021-05-16:
Searching for this symbol in the repo I find https://github.com/bb-qq/aqc111/blob/507c1545b3b76a756ff7816153ff2e1191e30fb8/linux-4.4.59/mii.c#L393

marcosscriven, 2021-05-16:
Just a bit of extra info, I diffed the loaded mods before/after install:
```
1a2,3
> aqc111                 16498  0
> usbnet                 22173  1 aqc111
77c79
< usbcore               203625  10 usblp,uhci_hcd,usb_storage,ehci_hcd,ehci_pci,xhci_plat_hcd,usbhid,xhci_hcd,xhci_pci
---
> usbcore               203625  12 usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,xhci_plat_hcd,usbhid,usbnet,xhci_hcd,xhci_pci
```

bb-qq, 2021-05-16:
Could you try to edit `/var/packages/aqc111/scripts/start-stop-status` manually not to load the `mii.ko`?
This module is usually not included on most platforms, but rtd2196 platforms seem to do.

marcosscriven, 2021-05-16:
Thanks @bb-qq - just got round to trying this, and it fixed that error.

Unfortunately the next error I get is:

```
[27578.571845] usb 5-1: new SuperSpeed USB device number 2 using xhci-hcd
[27578.717586] aqc111 5-1:1.0 eth1: register 'aqc111' at usb-xhci-hcd.5.auto-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:57:0f:39
[27584.117072] init: nmbd main process (5322) killed by TERM signal
```

marcosscriven, 2021-05-16:
I just tried restarting the NAS, and bringing up the interface with:

```
sudo ip link set eth1 up
```

And get the same error:

```
[  498.132810] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[  505.507794] aqc111 5-1:1.0 eth1: Link Speed 5000, USB 3
[  505.518955] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  511.453418] init: nmbd main process (13332) killed by TERM signal
[  526.070360] usb 2-1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
```

Curiously, this also seems to affect the USB connection to the UPS, and the Synology claims it's in battery mode (which it isn't).

marcosscriven, 2021-05-16:
I've disabled UPS, and unplugged it. I restarted the NAS.

On startup I now see:

```
[  126.199717] aqc111 5-1:1.0 eth1: unregister 'aqc111' usb-xhci-hcd.5.auto-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[  126.235229] aqc111 5-1:1.0 eth1 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[  126.243721] aqc111 5-1:1.0 eth1 (unregistered): Failed to write(0x1) reg index 0x0002: -71
[  126.252279] aqc111 5-1:1.0 eth1 (unregistered): Failed to write(0x61) reg index 0x0000: -71
[  126.260943] usb 5-1: usb_reset_and_verify_device Failed to disable LTM
               .
[  126.269244] usb usb5-port1: logical disconnect on port 1 [kworker/3:1]
[  126.276073] aqc111 5-1:1.0 (unnamed net_device) (uninitialized): Failed to write(0xb0) reg index 0x0001: -19
[  126.535313] usb 5-1: USB disconnect, device number 2
[  126.695584] usb 5-1: new SuperSpeed USB device number 3 using xhci-hcd
[  126.842269] aqc111 5-1:1.0 eth1: register 'aqc111' at usb-xhci-hcd.5.auto-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:57:0f:39
```

marcosscriven, 2021-05-16:
Tried all sorts now - I have a USB 3.1 Gen 2 powered hub, and tried using the device though that. Still having problem.

I tried setting throttling off, and low power 5g on, and still not working.

When I plug in the ethernet cable the Synology has a fit:

```
[  245.385978] aqc111 5-1.3:1.0 eth1: register 'aqc111' at usb-xhci-hcd.5.auto-1.3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:57:0f:39
[  293.452555] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[  383.115375] init: nmbd main process (11886) killed by TERM signal
[  428.840640] aqc111 5-1.3:1.0 eth1: Link Speed 1000, USB 3
[  428.848411] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.854676] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.860923] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.867215] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.873482] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  428.879912] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.886222] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.892508] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.898791] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.905058] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  428.911317] aqc111 5-1.3:1.0 eth1: kevent 2 may have been dropped
[  430.211799] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[  434.088839] aqc111 5-1.3:1.0 eth1: Link Speed 1000, USB 3
[  434.097740] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  436.016681] init: nmbd main process (13310) killed by TERM signal
[  437.706358] init: nmbd main process (13826) killed by TERM signal
[  440.106217] init: nmbd main process (14707) killed by TERM signal
```

I did, once, get it to run iperf3, and connect at 5G. But it seems extremely unstable.

I also upgraded both the dongles to firmware 3.1.6.

marcosscriven, 2021-05-17:
Just to rule out problems, I took a step back and tested two of these QNA-UC5G1T attached to two Linux VMs, both with their own PCI USB controller passed through.

uname for both VMs:
```
Linux linux2 5.11.0-17-generic #18-Ubuntu SMP Thu May 6 20:10:11 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

In dmesg for both I see:

```
[  464.218304] aqc111 10-1.4:1.0 enx245ebe570f39: Link Speed 5000, USB 3
[  464.230405] IPv6: ADDRCONF(NETDEV_CHANGE): enx245ebe570f39: link becomes ready
```

And nothing else. No dramas. No errors of any sort.

Iperf:
```
iperf3 -c 192.168.2.1
Connecting to host 192.168.2.1, port 5201
[  5] local 192.168.2.2 port 48676 connected to 192.168.2.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   410 MBytes  3.44 Gbits/sec    3    786 KBytes
[  5]   1.00-2.00   sec   415 MBytes  3.48 Gbits/sec    0    926 KBytes
[  5]   2.00-3.00   sec   414 MBytes  3.47 Gbits/sec    0   1.19 MBytes
[  5]   3.00-4.00   sec   414 MBytes  3.47 Gbits/sec    0   1.43 MBytes
[  5]   4.00-5.00   sec   414 MBytes  3.47 Gbits/sec    0   1.63 MBytes
[  5]   5.00-6.00   sec   415 MBytes  3.48 Gbits/sec    0   1.81 MBytes
[  5]   6.00-7.00   sec   414 MBytes  3.47 Gbits/sec    0   1.97 MBytes
[  5]   7.00-8.00   sec   415 MBytes  3.48 Gbits/sec    0   2.11 MBytes
[  5]   8.00-9.00   sec   414 MBytes  3.47 Gbits/sec    0   2.26 MBytes
[  5]   9.00-10.00  sec   414 MBytes  3.47 Gbits/sec    0   2.38 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  4.04 GBytes  3.47 Gbits/sec    3             sender
[  5]   0.00-10.01  sec  4.04 GBytes  3.47 Gbits/sec                  receiver
```

Transferring 10GB file with python SimpleHTTPServer / wget

```
Saving to: ‘10G.img’

10G.img                         100%[======================================================>]  10.00G   316MB/s    in 30s
```

Did this lots of times with no throttling, kernel errors, dmesg errors.

So at least I've confirmed that's what's going wrong is *only* on the Synology.

marcosscriven, 2021-05-17:
More info on the dongle:
```
lsusb -vvv -d 1c04:0015

Bus 010 Device 003: ID 1c04:0015 QNAP System Inc. Pacific￿
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x1c04 QNAP System Inc.
  idProduct          0x0015
  bcdDevice            1.01
  iManufacturer           1 QNAP￿
  iProduct                2 Pacific￿
  iSerial                 3 <redacted>
  bNumConfigurations      2
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0
      iInterface              5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              11
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
[... 97 lines cut ...]
      bInterfaceProtocol      0
      iInterface             12
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
```

bb-qq, 2021-05-17:
Could you paste the results of these commands executed on your DS220j, with the dongle connected via the self-powered hub?

* lsusb
* ls -l /sys/bus/usb/devices/*/power/autosuspend
* cat /sys/bus/usb/devices/*/power/autosuspend
* ls -l /sys/bus/usb/devices/*/power/level
* cat /sys/bus/usb/devices/*/power/level

marcosscriven, 2021-05-17:
@bb-qq  - I'll try that. At the moment though I've just noticed that even with your driver removed and the device reset, and nothing plugged into USB, I'm seeing this in my logs:

```
43066.064055] r8169 98016000.gmac eth0: link down
[43068.107960] init: dhcp-client (eth0) main process (4941) killed by TERM signal
[43071.089022] init: nmbd main process (26780) killed by TERM signal
[43125.812987] init: nmbd main process (26877) killed by TERM signal
[43126.992852] init: nmbd main process (27805) killed by TERM signal
[43141.717912] r8169 98016000.gmac eth0: link up 1 Gbps
[43141.723022] r8169_reset_task: reset with slow event.
[43141.765166] r8169 98016000.gmac eth0: rtl_csiar_cond == 0 (loop: 100, delay: 10).
[43141.774096] r8169 98016000.gmac eth0: rtl_csiar_cond == 1 (loop: 100, delay: 10).
[43141.783023] r8169 98016000.gmac eth0: link up 1 Gbps
[43143.143174] init: nmbd main process (27897) terminated with status 1
[43143.149945] init: nmbd main process ended, respawning
[43143.571571] init: dhcp-client (eth0) main process (26678) killed by TERM signal
[43145.042674] r8169 98016000.gmac eth0: link down
[43145.351186] init: nmbd main process (28559) killed by TERM signal
[43146.495966] init: nmbd main process (29058) killed by TERM signal
[43146.971807] init: dhcp-client (eth0) main process (28317) killed by TERM signal
[43147.097814] init: nmbd main process (29249) killed by TERM signal
[43148.572552] init: nmbd main process (29539) killed by TERM signal
[43148.578152] r8169 98016000.gmac eth0: link up 1 Gbps
[43148.578164] r8169_reset_task: reset with slow event.
[43148.645723] r8169 98016000.gmac eth0: rtl_csiar_cond == 0 (loop: 100, delay: 10).
[43148.654707] r8169 98016000.gmac eth0: rtl_csiar_cond == 1 (loop: 100, delay: 10).
[43148.663706] r8169 98016000.gmac eth0: link up 1 Gbps
[43150.311688] init: dhcp-client (eth0) main process (29268) killed by TERM signal
[43161.839121] init: nmbd main process (32400) killed by TERM signal
```

So it seems maybe my DS220j is not in a good state to start with?

marcosscriven, 2021-05-17:
As requested:

```
marcosscriven@nas:/lib/modules$ lsusb
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
  |__3-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP￿ Pacific￿ 0CI07797)
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
marcosscriven@nas:/lib/modules$ ls -l /sys/bus/usb/devices/*/power/autosuspend
-rw-r--r-- 1 root root 4096 May 17 18:02 /sys/bus/usb/devices/3-1/power/autosuspend
-rw-r--r-- 1 root root 4096 May 17 18:02 /sys/bus/usb/devices/usb2/power/autosuspend
-rw-r--r-- 1 root root 4096 May 17 18:02 /sys/bus/usb/devices/usb3/power/autosuspend
-rw-r--r-- 1 root root 4096 May 17 18:02 /sys/bus/usb/devices/usb4/power/autosuspend
-rw-r--r-- 1 root root 4096 May 17 18:02 /sys/bus/usb/devices/usb5/power/autosuspend
marcosscriven@nas:/lib/modules$ cat /sys/bus/usb/devices/*/power/autosuspend
2
0
0
0
0
marcosscriven@nas:/lib/modules$ ls -l /sys/bus/usb/devices/*/power/level
-rw-r--r-- 1 root root 4096 May 17 18:03 /sys/bus/usb/devices/3-1/power/level
-rw-r--r-- 1 root root 4096 May 17 18:03 /sys/bus/usb/devices/usb2/power/level
-rw-r--r-- 1 root root 4096 May 17 18:03 /sys/bus/usb/devices/usb3/power/level
-rw-r--r-- 1 root root 4096 May 17 18:03 /sys/bus/usb/devices/usb4/power/level
-rw-r--r-- 1 root root 4096 May 17 18:03 /sys/bus/usb/devices/usb5/power/level
marcosscriven@nas:/lib/modules$ cat /sys/bus/usb/devices/*/power/level
on
auto
auto
auto
auto
```

marcosscriven, 2021-05-17:
Another error I'm seeing on the occasion I get iperf3 running:

```
[12039.385852] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
[12039.394154] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
[12039.402438] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
[12039.410721] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
[12039.419004] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
```

bb-qq, 2021-05-20:
Could you execute this command to disable the auto-suspend function? It is unlikely this setting change will fix the problem, but please try it just in case. (starting/stopping the driver may be required to take effect)
```
sudo echo -1 > /sys/bus/usb/devices/3-1/power/autosuspend
```

> Another error I'm seeing on the occasion I get iperf3 running:
```
[12039.385852] xhci-hcd xhci-hcd.2.auto: WARN Event TRB for slot 1 ep 5 with no TDs queued?
```
I think these WARNings can be ignored, but the `xhci-hcd xhci-hcd.2.auto: xHCI host not responding to stop endpoint command.` error is fatal and this is the cause of instability.
Typically, the Host Controller should respond to any commands even if these commands issued by class drivers contain invalid sequences. So I guess the Host Controller has some kind of problem.

Can you confirm stability with other USB3 devices like external hard drives? I remember that one of the users in this issue tracker also observed similar errors in the dmesg when he uses hard drives.

If the other devices work correctly, we can conclude that the behavior of the Realtek driver is causing the host controller to hang. But it will be tough to fix it...

bb-qq, 2022-12-04:
This symptom appears to be caused by the rtd1296 platform.
I decided to merge topics on rtd1296 into the issue [#275](https://github.com/bb-qq/r8152/issues/275) in the r8152 project.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq