bb-qq/aqc111 #44: Including upstream driver fixes?

marcosscriven, 2021-05-17:
Hi @bb-qq 

As an aside to the separate issue I'm trying to get this working on the rtd2196 platform, I have a couple of questions about the three modules in your repo.

Looking at https://github.com/bb-qq/aqc111/blob/master/aqc111.c vs https://github.com/torvalds/linux/blob/master/drivers/net/usb/aqc111.c It looks like yours has everything up to 27th Nov 2018 https://github.com/torvalds/linux/commits/master/drivers/net/usb/aqc111.c. 

Do you think any of these are critical? If so, how are you building this? As a package in https://github.com/SynoCommunity?

Regarding your usbnet module, where did you get that from? I tried simply not loading it to easily see the differences, and got this:

```
[ 9476.414146] aqc111: module verification failed: signature and/or required key missing - tainting kernel
[ 9476.423901] aqc111: Unknown symbol usbnet_set_msglevel (err 0)
[ 9476.429956] aqc111: Unknown symbol usbnet_write_cmd_async (err 0)
[ 9476.436239] aqc111: Unknown symbol usbnet_get_msglevel (err 0)
[ 9476.442256] aqc111: Unknown symbol usbnet_open (err 0)
[ 9476.447550] aqc111: Unknown symbol usbnet_skb_return (err 0)
[ 9476.453381] aqc111: Unknown symbol usbnet_tx_timeout (err 0)
[ 9476.459207] aqc111: Unknown symbol usbnet_read_cmd (err 0)
[ 9476.464846] aqc111: Unknown symbol usbnet_suspend (err 0)
[ 9476.470407] aqc111: Unknown symbol usbnet_start_xmit (err 0)
[ 9476.476225] aqc111: Unknown symbol usbnet_read_cmd_nopm (err 0)
[ 9476.482322] aqc111: Unknown symbol usbnet_get_drvinfo (err 0)
[ 9476.488262] aqc111: Unknown symbol usbnet_get_endpoints (err 0)
[ 9476.494391] aqc111: Unknown symbol usbnet_stop (err 0)
[ 9476.499716] aqc111: Unknown symbol usbnet_defer_kevent (err 0)
[ 9476.505727] aqc111: Unknown symbol usbnet_disconnect (err 0)
[ 9476.511558] aqc111: Unknown symbol usbnet_probe (err 0)
[ 9476.516927] aqc111: Unknown symbol usbnet_resume (err 0)
```

Did you manually merge in those missing methods from a newer driver? I'm wondering if significant such signficant changes to the usbnet  module already present could also be a cause of issues?

bb-qq, 2021-05-29:
The usbnet.ko is built from [usbnet.c|https://github.com/bb-qq/aqc111/tree/master/linux-4.4.59/usbnet.c] in this repository, that source was copied from the original Linux repository, at the same version to the target platform.

I checked recent changes in the Linux repository, but only MTU related changes are meaningful. Other changes are fix to follow version up of the kernel interfaces.

So I guess that putting effort into following upstream is not going to improve stability, unfortunately...

bb-qq, 2022-08-11:
For recent releases, I use a newer version of `usbnet.c` than the kernel version used by Synology. (The branches used are the same)

I am not sure if this solves the problem with this issue, but I would like to close this issue as there is nothing more I can do. If you have any questions, feel free to re-open it.

closed 2022-08-11 by bb-qq