bb-qq/aqc111 #46: Report FYI DSM7 - DS1520+  (geminilake)

noblemd00, 2021-07-11:
Working good...

DSM 7.0-41890
aqc111-geminilake-1.3.3.0-3

Sabrent USB Type-A or Type-C to 5-Gigabit Ethernet Adapter [10/100/1000/2500/5000 Mbps] (NT-SS5G)
[amazon.com/gp/product/B08977K9D2](url)

Jumbo Frame Enabled - MTU 9000
Network Status: | 5000 Mbps, Full duplex, MTU 9000

Have not experimented with other values.

Getting 250-300 MB/s upload and 480 MB/s download settling half way through to 340 MB/s (3.5 GB file). 

Uploading to 5 Disk 6TB RAID 5 (22TB Total)

noblemd00, 2021-07-12:
Spoke too soon...

Checked status after restart and all was well, however...

After power off the package runs, but the adapter does not show up.

Package has to be stopped and re-started then the adapter shows up and connects.

bb-qq, 2021-07-13:
Are you using DHCP or static address?

ihvar, 2021-07-13:
Try this by sshimko - https://github.com/bb-qq/aqc111/issues/45#issuecomment-877728430

noblemd00, 2021-07-14:
> Are you using DHCP or static address?

Tried both. Same result.

Also tried #45 maxlanport 5 (4 built-in lan ports +1 5G lan port)  solution - did not work.

Temporary fix: Running scheduled task (as ROOT) to automatically stop and start the service 2 minutes after scheduled morning startup. I tried running this as a Boot-up task, but that did not work. The scheduled task seems to work as a temporary fix. Also works when manually run.

User defined script:

/usr/syno/bin/synopkg stop aqc111
/usr/syno/bin/synopkg start aqc111
exit 0

bb-qq, 2021-07-15:
Could you try to add some wait to the startup script?

```
vi /var/packages/aqc111/scripts/start-stop-status
```

 ```

case $1 in
    start)
        /sbin/insmod $SYNOPKG_PKGDEST/aqc111/mii.ko
        /sbin/insmod $SYNOPKG_PKGDEST/aqc111/usbnet.ko
        /sbin/insmod $SYNOPKG_PKGDEST/aqc111/aqc111.ko
        ###############
        sleep 10 # add this
        ###############
        set_interfaces up
        exit 0
    ;;
    stop)
        set_interfaces down
        /sbin/rmmod $SYNOPKG_PKGDEST/aqc111/aqc111.ko
        /sbin/rmmod $SYNOPKG_PKGDEST/aqc111/usbnet.ko
        /sbin/rmmod $SYNOPKG_PKGDEST/aqc111/mii.ko
        exit 0
    ;;
    status)
        /sbin/lsmod | grep aqc111 && exit 0 || exit 3
    ;;
    killall)
        ;;
    log)
        exit 0
    ;;
esac
```

noblemd00, 2021-07-15:
> Could you try to add some wait to the startup script?
>
> ```
> vi /var/packages/aqc111/scripts/start-stop-status
> ```
>
> ```
>
> case $1 in
>    start)
>        /sbin/insmod $SYNOPKG_PKGDEST/aqc111/mii.ko [...]

Deactivated my temp fix scheduled task.
_**Shut down and restarted several times and your fix seems to work.**_
Will monitor and report if your added delay fix stops working.