bb-qq/aqc111 #48: DS415+ slow performance with QNAP QNA-UC5G1T USB 3.2 Gen 1 to 5GbE

devedse, 2021-07-20:
### Description of the problem

Hi all, I just ordered the QNAP QNA-UC5G1T USB 3.2 Gen 1 to 5GbE adapter. Before testing I first hooked it up to a windows machine and did an iperf3 from that machine to my own PC. That worked fine with 1 thread steadily transferring at 3.5gbit / sec.

After that I put the QNAP adapter in one of the rear USB ports of my NAS. Since I'm running DSM 7.0 I couldn't find a version of iperf3 to run natively though so I just tested with the docker version.

Running a single thread on my **1gbit line** results in the following performance (**as expected**):

[image]

Running a single thread on my **5gbit line** through the QNAP adapter results in the following performance (**slower then 1gbit line**):

[image]

Running a 10 thread test on my **5gbit line** through the QNAP adapter results in the following performance (**A bit faster but still not getting the expected speed**):

[image]

So it seems that in my case the QNAP adapter works perfectly fine from windows (getting 3.5 gbit) but not from my DS415+. Any suggestions? (On windows I did disable `Low Power 5Gbps` as stated on their website: https://www.qnap.com/en/product/qna-uc5g1t)

### Description of your products

**Output `uname -a`**: `Linux Devenology 3.10.108 #41890 SMP Fri Jun 25 02:40:49 CST 2021 x86_64 GNU/Linux synology_avoton_415+`
**DSM version:** 7.0
**Ethernet Dongle:** QNAP QNA-UC5G1T USB 3.2 Gen 1 to 5GbE

### Description of your environment

[image]

### Output of ``dmesg`` command

**Truncated**
````
[1677604.343968] Module [usbnet] is removed.
[1677604.350342] Module [mii] is removed.
[1677608.632365] aqc111 3-1:1.0 eth2: register 'aqc111' at usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:5b:f9:a9
[1677608.645591] usbcore: registered new interface driver aqc111
[1677608.933589] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[1677616.371453] aqc111 3-1:1.0 eth2: Link Speed 5000, USB 3
[1677616.385760] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[1677638.498363] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[1677801.033352] Module [aqc111] is removed.
[1677801.038010] usbcore: deregistering interface driver aqc111
[1677801.044451] aqc111 3-1:1.0 eth2: unregister 'aqc111' usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[1677801.069042] Module [usbnet] is removed.
[1677801.075434] Module [mii] is removed.
[1677808.034358] aqc111 3-1:1.0 eth2: register 'aqc111' at usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:5b:f9:a9
[1677808.047595] usbcore: registered new interface driver aqc111
[1677808.341055] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[1677815.665630] aqc111 3-1:1.0 eth2: Link Speed 5000, USB 3
[1677815.679295] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[1677925.917274] Module [aqc111] is removed.
[1677925.921943] usbcore: deregistering interface driver aqc111
[1677925.928496] aqc111 3-1:1.0 eth2: unregister 'aqc111' usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[1677925.957980] Module [usbnet] is removed.
[1677925.964313] Module [mii] is removed.
[1677928.762776] type=1400 audit(1626814351.760:2): apparmor="DENIED" operation="capable" parent=1 profile="/usr/sbin/dhclient" pid=31097 comm="dhclient" pid=31097 comm="dhclient" capability=16  capname="sys_module"
[1677930.878783] type=1400 audit(1626814353.875:3): apparmor="DENIED" operation="capable" parent=1 profile="/usr/sbin/dhclient" pid=31836 comm="dhclient" pid=31836 comm="dhclient" capability=16  capname="sys_module"
[1677931.210130] aqc111 3-1:1.0 eth2: register 'aqc111' at usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:5b:f9:a9
[1677931.223370] usbcore: registered new interface driver aqc111
[1677931.517826] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[1677938.947827] aqc111 3-1:1.0 eth2: Link Speed 5000, USB 3
[1677938.962391] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[1678332.344855] aqc111 3-1:1.0 eth2: Link Speed 2500, USB 3
[1678335.714885] Module [aqc111] is removed.
[1678335.719576] usbcore: deregistering interface driver aqc111
[1678335.726081] aqc111 3-1:1.0 eth2: unregister 'aqc111' usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[1678335.753203] Module [usbnet] is removed.
[1678335.759525] Module [mii] is removed.
[... 142 lines cut ...]
eth2      Link encap:Ethernet  HWaddr 24:5E:BE:5B:F9:A9
          inet addr:192.168.2.210  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe5b:f9a9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:563798 errors:1 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:790571027 (753.9 MiB)  TX bytes:18363528 (17.5 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1651550 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1651550 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:145778818 (139.0 MiB)  TX bytes:145778818 (139.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

devedse, 2021-07-20:
Or would it simply be better / more stable to go with a 2.5gbit adapter?

bb-qq, 2021-07-21:
First of all, you should use the native version of iperf. The docker version consumes a lot of CPU for network processing, so it is not suitable for measuring the throughput of USB-based ethernet devices which is CPU-hungry.

You can get the native iperf from [here](http://www.jadahl.com/iperf-arp-scan/DSM_6.2/). Unfortunately, there is no package for DSM 7.0, but you can use it without any problem by extracting the binary inside.

Next, to get throughput in SMB close to iperf, you need to set the SMB version to v2 with LargeMTU or v3. See [here](https://kb.synology.com/en-us/DSM/help/DSM/AdminCenter/file_winmacnfs_win?version=7) for more details.

Lastly, assigning a different subnet to each NIC is recommended to prevent communication through unintended NICs. In this case, it is a good idea to assign multiple IP addresses to the PC as aliases.

devedse, 2021-07-21:
I think I reeeaaaally screwed up this time. I extracted the `iperf_avoton-6.2_3.7-1.spk` package and then did a copy of iperf3 to the /bin/ folder on my synology.
After trying it it showed me the following error:
```
iperf3: error while loading shared libraries: libiperf.so.0: cannot open shared object file: No such file or directory
```

So I thought, ok, then I'll copy over the /lib directory as well. The second I did this my ssh connection broke and now my NAS is completely unreachable.... shit...

devedse, 2021-07-21:
Due to loads of issues I've just reinstalled my NAS. How should I go about installing iperf3 natively without bricking my NAS again 🙂 ?

devedse, 2021-07-22:
I got a hint from someone else. Simply run this command:

```
LD_LIBRARY_PATH='/volume1/Unsorted/#SynologyStuff/iperf3' ./iperf3 -s
```

devedse, 2021-07-22:
I did some more tests today with the 5gbit adapter and these are the results:
[image]

It seems that MTU matters **A LOT** for the 5gbe adapter. Even though it makes no difference at all for the 1gbit integrated network ports. Also difference for this adapter on my specific NAS between native iperf3 and docker iperf3 seems to be minimal.

devedse, 2021-07-22:
I also did some tests with 10 threads in iperf3 (iperf3 docker):
[image]

And 10 threads in iperf3 native:
[image]

devedse, 2021-07-22:
To do some more testing I also ordered the CLUB 3D CAC-1420 (Type-A, confirmed working). Here's some testing results for that device. (I'll also add them to the respective github):

* Model(platform): `DS415+ (avoton)`
* DSM version:  `e.g. 7.0-41890`
* Driver version:  `e.g. 2.15.0-3`
* Ethernet device: `CLUB 3D CAC-1420 2.5GBIT`
* Working status: `Tbd`

| Threads | iperf3 type | MTU | Reversed? | Speed | Notes |
| -- | -- | -- | -- | -- | -- |
| 1 | native | 1500 | | 1.77 Gbits | |
| 10 | native | 1500 | | 1.83 Gbits | |
| 1 | docker | 1500 | | 943 Mbits | This one sometimes dropped to 0.00 temporary |
| 10 | docker | 1500 | | 1.76 Gbits | |
| 1 | native | 1500 | -R | 1.85 Gbits | -R seems to be more stable |
| 10 | native | 1500 | -R | 1.85 Gbits | -R seems to be more stable |
| 1 | docker | 1500 | -R | 1.86 Gbits | -R seems to be more stable |
| 10 | docker | 1500 | -R | 1.86 Gbits | -R seems to be more stable |

bb-qq, 2021-07-24:
Sorry I didn't give you more details about iperf. I'm relieved to hear that you were able to recover.

Could you try to connect the devices via Y-cable or self-powered hub with an AC adapter? It's unlikely this is the cause, but. insufficient bus power has been known to cause performance or stability issues.

devedse, 2021-07-24:
I've decided to return the AQC module and just focus on the 2.5gbit. I hope that's understandable 😄

bb-qq, 2022-01-10:
Since there is no way to further investigate this symptom, this issue is closed.

To other people: if you encounter a similar problem, please try increasing `vm.min_free_kbytes` as well.

* read current value: `sysctl -n vm.min_free_kbytes`
* write new value: `sysctl -w vm.min_free_kbytes=XXXX`

closed 2022-01-10 by bb-qq

biergaizi, 2024-10-11:
My observation is that USB network adapters can interact really badly with some USB Host Controllers and hubs. I have an AMD motherboard with native CPU USB ports and expanded USB ports provided by the A320 chipset (which is basically a custom ASMedia controller). The native CPU port has no problems at all. But if a USB NIC is plugged into the ASMedia port, sending a lot of traffic from another machine to that NIC can cause an immediate hardware lockup for RTL8156B (hardware buffer overrun?) adapters, or limited to a slow 300 Mbps for AQC111 adapters. Adding a USB 3 Hub in the middle can improve it somewhat, no more lockups or throttling, but it still can't go faster than 1 Gbps. The ASMedia port itself has no problems handling USB HDD writes at 400 MB/s.

Thus, I suspect performance problems are related to how USB transactions are handled in different host controllers, with subtle interoperability problems that can only be understood by USB stack gurus...