bb-qq/aqc111 #50: DS 918+ DSM 7.0 can't find /var/packages/r8152/target/r8152/spk_su

gzplca, 2021-07-26:
I'm using CableCreation USB 3.0 to 2.5 Gigabit LAN Ethernet Cable Adapter, it works fine when I'm using DSM 6. Today I updated to DSM 7, it doesn't work, then I follow the https://github.com/bb-qq/r8152 guide, first time it said Failed to repair "AQC111 driver". Then I used SSH terminal to run "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su". It told me don't have this file or folder. I check the folders, I have /var/packages/r8152/target/r8152/spk_su /opt/sbin/ folder but don't have spk_su folder or file, I think var/packages/r8152/target/r8152/spk_su /opt/sbin/ folder is created when I installed the driver under DSM6. Then I tried to remove the folder and tried again. The manual install didn't create any new folder. Not sure what I can do now.

gzplca, 2021-07-26:
Found the reason, the release page link in "Chose a driver package downloaded from the release page" is link to https://github.com/bb-qq/aqc111/releases but not https://github.com/bb-qq/r8152/releases/tag/2.15.0-3

closed 2021-07-26 by gzplca

bb-qq, 2021-08-22:
Thank you for pointing out it. 
It is already fixed.