bb-qq/aqc111 #51: DS420j + DSM

ffries, 2021-08-28:
Dear all,

First I would like to thank you for the hard work on bringing 5Gb lan connection to our Synology NAS. 
The Spec of the DS420j is 115Mb/s but it is way more on a 5Gb network (probably around 200 Mb/s).

sudo hdparm -Tt /dev/md1 
/dev/md1:
 Timing cached reads:   1428 MB in  2.00 seconds = 713.82 MB/sec
 Timing buffered disk reads: 610 MB in  3.00 seconds = 203.25 MB/sec

I can report success installing the latest driver on the DS420J.
The driver is loaded and eth1 is present. Unfortunately it does not show in the UI.

How can I add eth1 manually or in the UI?

Kind regards,
French Fries

ffries, 2021-08-31:
I configured the interface manually using an ssh shell:
ifconfig eth1 192.168.9.1/24 up

Then the interface showed-up in DSM7. Very strange. Maybe this should be added in the HOWTO.

Please leave this issue open as other can read it.

ffries, 2021-08-31:
Unfortunately, these settings do not survive a reboot. I am not sure that all services are listening and running on eth0/eth1. Any solution so far?

ffries, 2021-08-31:
Maybe a duplicate of this closed issue.

ffries, 2021-08-31:
if I run /usr/syno/bin/synopkg restart aqc111 manually everything is fine. 
So the seems to be a problem on boot running the aqc111 service.

bb-qq, 2021-09-24:
Could you try to increase the sleep-wait in "/var/packages/aqc111/scripts/start-stop-status"

bb-qq, 2022-01-10:
I released a new version that includes an improvement of startup handling.
https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-5

I'm closing this issue because it hasn't been updated in a long time, but feel free to reopen it if you need anything.

closed 2022-01-10 by bb-qq