bb-qq/aqc111 #54: Only 1000mbs on DSM 7.0.1 DS920+

BodhiSmoketiffa, 2021-11-07:
### Hello, I'm trying to use the QNA-UC5G1T 5Gbe Adapter to connect to my DS920+ but can't get more than 100mbs transfers regardless of what protocol I choose. I've tried SMB and iSCSI. I have Static IP for my ETH ports connected directly to network and Static IP for the QNAP Device and my PC NIC Card.  I know I should have never updated to DSM 7 but this is my first NAS and I goofed. Thanks in advance. I really appreciate any help at all. 

> Replace this line with your answer.

### Description of your products

> - DS920+
>     - and output of ``Linux TUN_MPS_NAS 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Li                                        nux synology_geminilake_920+
` command
>     - DSM version x.x
> - QNA-UC5G1T

### Description of your environment

> - connect directly to a PC 
>     - PC:Intel(R) Ethernet 10G 2P X550-t Adapter
 or chipset name
>     - Hub: model name
> - cable category CAT 6 

### Output of ``dmesg`` command

````
[    0.706625] pci_bus 0000:00: root bus resource [mem 0xfed01000-0xfed01fff window]
[    0.715001] pci_bus 0000:00: root bus resource [mem 0xfed03000-0xfed03fff window]
[    0.723375] pci_bus 0000:00: root bus resource [mem 0xfed06000-0xfed06fff window]
[    0.731750] pci_bus 0000:00: root bus resource [mem 0xfed08000-0xfed09fff window]
[    0.740126] pci_bus 0000:00: root bus resource [mem 0xfed80000-0xfedbffff window]
[    0.748502] pci_bus 0000:00: root bus resource [mem 0xfed1c000-0xfed1cfff window]
[    0.756877] pci_bus 0000:00: root bus resource [mem 0xfee00000-0xfeefffff window]
[    0.765254] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.771398] pci 0000:00:00.0: [8086:31f0] type 00 class 0x060000
[    0.771729] pci 0000:00:02.0: [8086:3185] type 00 class 0x030000
[    0.771750] pci 0000:00:02.0: reg 0x10: [mem 0xa0000000-0xa0ffffff 64bit]
[    0.771758] pci 0000:00:02.0: reg 0x18: [mem 0x90000000-0x9fffffff 64bit pref]
[    0.771764] pci 0000:00:02.0: reg 0x20: [io  0x4000-0x403f]
[    0.772101] pci 0000:00:0f.0: [8086:319a] type 00 class 0x078000
[    0.772149] pci 0000:00:0f.0: reg 0x10: [mem 0xa1312000-0xa1312fff 64bit]
[    0.772239] pci 0000:00:0f.0: PME# supported from D3hot
[    0.772533] pci 0000:00:12.0: [8086:31e3] type 00 class 0x010601
[    0.772558] pci 0000:00:12.0: reg 0x10: [mem 0xa1310000-0xa1311fff]
[    0.772567] pci 0000:00:12.0: reg 0x14: [mem 0xa131b000-0xa131b0ff]
[    0.772576] pci 0000:00:12.0: reg 0x18: [io  0x4080-0x4087]
[    0.772584] pci 0000:00:12.0: reg 0x1c: [io  0x4088-0x408b]
[    0.772593] pci 0000:00:12.0: reg 0x20: [io  0x4060-0x407f]
[    0.772601] pci 0000:00:12.0: reg 0x24: [mem 0xa1319000-0xa13197ff]
[    0.772636] pci 0000:00:12.0: PME# supported from D3hot
[    0.772884] pci 0000:00:13.0: [8086:31d8] type 01 class 0x060400
[    0.772953] pci 0000:00:13.0: PME# supported from D0 D3hot D3cold
[    0.773171] pci 0000:00:13.0: System wakeup disabled by ACPI
[    0.779369] pci 0000:00:13.2: [8086:31da] type 01 class 0x060400
[    0.779442] pci 0000:00:13.2: PME# supported from D0 D3hot D3cold
[    0.779656] pci 0000:00:13.2: System wakeup disabled by ACPI
[    0.785848] pci 0000:00:13.3: [8086:31db] type 01 class 0x060400
[    0.785913] pci 0000:00:13.3: PME# supported from D0 D3hot D3cold
[    0.786130] pci 0000:00:13.3: System wakeup disabled by ACPI
[    0.792337] pci 0000:00:14.0: [8086:31d6] type 01 class 0x060400
[    0.792410] pci 0000:00:14.0: PME# supported from D0 D3hot D3cold
[    0.792627] pci 0000:00:14.0: System wakeup disabled by ACPI
[    0.798819] pci 0000:00:14.1: [8086:31d7] type 01 class 0x060400
[    0.798897] pci 0000:00:14.1: PME# supported from D0 D3hot D3cold
[    0.799113] pci 0000:00:14.1: System wakeup disabled by ACPI
[    0.805330] pci 0000:00:15.0: [8086:31a8] type 00 class 0x0c0330
[    0.805360] pci 0000:00:15.0: reg 0x10: [mem 0xa1300000-0xa130ffff 64bit]
[    0.805418] pci 0000:00:15.0: PME# supported from D3hot D3cold
[    0.805638] pci 0000:00:15.0: System wakeup disabled by ACPI
[    0.811858] pci 0000:00:16.0: [8086:31ac] type 00 class 0x118000
[    0.811894] pci 0000:00:16.0: reg 0x10: [mem 0xa1313000-0xa1313fff 64bit]
[    0.811908] pci 0000:00:16.0: reg 0x18: [mem 0xa1314000-0xa1314fff 64bit]
[    0.812230] pci 0000:00:18.0: [8086:31bc] type 00 class 0x118000
[... 1756 lines cut ...]
          inet addr:10.8.32.9  Bcast:10.8.33.255  Mask:255.255.254.0
          inet6 addr: fe80::211:32ff:feca:fa1d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:104530 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44289 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:37637479 (35.8 MiB)  TX bytes:43931774 (41.8 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:CA:FA:1E
          inet addr:10.8.32.61  Bcast:10.8.33.255  Mask:255.255.254.0
          inet6 addr: fe80::211:32ff:feca:fa1e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:57294 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5468 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:25428462 (24.2 MiB)  TX bytes:4227709 (4.0 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2021-11-13:
When QNA-UC5G1T is connected to a PC and communicates in the same environment, is the performance OK?
I would like to confirm just in case because QNA-UC5G1T is a very sensitive device. (thermal throttling, auto-negotiation, etc.)

BodhiSmoketiffa, 2021-11-19:
Yes the preformance was as expected. I did some minor changes. I'm using a LUN that only connects via LAN 3(QNAP) and I was able to get the transfer speeds up too 400/mbs but that only lasts for 10 percent of the transfer. After that it goes back down to 100ish/mbs. I imagine this is due to thermal throttling?

bb-qq, 2021-11-21:
Yes, you are most likely experiencing thermal throttling. Disabling thermal throttling or enabling Low Power mode using ethtool may improve the symptoms. Please see the README for instructions.

BodhiSmoketiffa, 2021-11-21:
I input both the cmds from the readme using putty and it'll trasnfer at 1.5gbe to start, then go down to 300/mbs then back down to 100/mbs. How can you check to see if they were input correctly?

bb-qq, 2021-11-23:
If you didn't see any errors, the settings would have been reflected correctly.

I would like to ask aditional:

* Are you using a front USB port?
* Did you measure throughput using iperf?

bb-qq, 2024-06-15:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

What I wanted to confirm is now summarized on this page and I leave the link as a note.
https://github.com/bb-qq/r8152/wiki/Troubleshooting

closed 2024-06-15 by bb-qq