bb-qq/aqc111 #55: DS920+  Adapter won't automatically connect w/out manual restart after reboot

wdwms, 2021-11-27:
I've successfully installed the driver and have it working with a QNA-UC5G1T.  After a reboot (or hot unplug/replug) the adapter fails to connect. I have to manually start/stop the server in package manager, or issue  sudo /usr/syno/bin/synopkg restart aqc111 via ssh.

I can deal with the hot/swap not being available, however i'm toying w/the idea of putting 2-3 minute timer in the boot script and then executing sudo /usr/syno/bin/synopkg restart aqc111 in order to get the adapter up and running at boot

Can you advise of the right way to do this?

Linux Peabody 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_920+

wdwms, 2021-11-28:
Looks like Sleep 15 in the start script fixed this. I think you should add that as part of the install. Will do speed tests tomorrow with iperf3

bb-qq, 2022-01-09:
I released the [1.3.3.0-5](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-5) release that includes improvement of that wait by another way. Try it, please.

closed 2022-01-09 by bb-qq