bb-qq/aqc111 #56: Good downstream, poor upstream with DS920+

wdwms, 2021-11-28:
From DS920+ to my machine I get 1.1gbe, from the machine to the DS920+ I get 2.37gbe

DS920+ has a QNA-UC5G1T; main machine has a RealTech 2.5gbe NIC; only a 5gbe switch between the two.

Here is the iperf results from my DS920+ to my main machine, 

` iperf3 -c 192.168.2.2 -P 2
Connecting to host 192.168.2.2, port 5201
[  5] local 192.168.2.25 port 54880 connected to 192.168.2.2 port 5201
[  7] local 192.168.2.25 port 54882 connected to 192.168.2.2 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  58.4 MBytes   490 Mbits/sec    0    218 KBytes
[  7]   0.00-1.00   sec  56.0 MBytes   469 Mbits/sec    0    218 KBytes
[SUM]   0.00-1.00   sec   114 MBytes   959 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   1.00-2.00   sec  56.3 MBytes   472 Mbits/sec    0    218 KBytes
[  7]   1.00-2.00   sec  56.8 MBytes   477 Mbits/sec    0    218 KBytes
[SUM]   1.00-2.00   sec   113 MBytes   949 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   2.00-3.00   sec  56.3 MBytes   472 Mbits/sec    0    218 KBytes
[  7]   2.00-3.00   sec  56.8 MBytes   477 Mbits/sec    0    218 KBytes
[SUM]   2.00-3.00   sec   113 MBytes   949 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   3.00-4.00   sec  56.5 MBytes   474 Mbits/sec    0    218 KBytes
[  7]   3.00-4.00   sec  56.5 MBytes   474 Mbits/sec    0    218 KBytes
[SUM]   3.00-4.00   sec   113 MBytes   948 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   4.00-5.00   sec  56.2 MBytes   472 Mbits/sec    0    218 KBytes
[  7]   4.00-5.00   sec  56.9 MBytes   477 Mbits/sec    0    218 KBytes
[SUM]   4.00-5.00   sec   113 MBytes   949 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   5.00-6.00   sec  56.8 MBytes   477 Mbits/sec    0    218 KBytes
[  7]   5.00-6.00   sec  56.5 MBytes   474 Mbits/sec    0    218 KBytes
[SUM]   5.00-6.00   sec   113 MBytes   951 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   6.00-7.00   sec  56.4 MBytes   473 Mbits/sec    0    218 KBytes
[  7]   6.00-7.00   sec  56.6 MBytes   475 Mbits/sec    0    218 KBytes
[SUM]   6.00-7.00   sec   113 MBytes   948 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec  56.7 MBytes   476 Mbits/sec    0    218 KBytes
[  7]   7.00-8.00   sec  56.6 MBytes   475 Mbits/sec    0    218 KBytes
[SUM]   7.00-8.00   sec   113 MBytes   950 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   8.00-9.00   sec  56.6 MBytes   475 Mbits/sec    0    218 KBytes
[  7]   8.00-9.00   sec  56.6 MBytes   475 Mbits/sec    0    218 KBytes
[SUM]   8.00-9.00   sec   113 MBytes   949 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec  56.6 MBytes   475 Mbits/sec    0    218 KBytes
[  7]   9.00-10.00  sec  56.7 MBytes   476 Mbits/sec    0    218 KBytes
[SUM]   9.00-10.00  sec   113 MBytes   951 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   567 MBytes   475 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   566 MBytes   475 Mbits/sec                  receiver
[  7]   0.00-10.00  sec   566 MBytes   475 Mbits/sec    0             sender
[  7]   0.00-10.00  sec   565 MBytes   474 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  1.11 GBytes   950 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec  1.11 GBytes   949 Mbits/sec                  receiver
`

From main machine to DS920+
`S:\iperf-3.1.3-win64>iperf3.exe -c 192.168.2.20 -P 2
Connecting to host 192.168.2.20, port 5201
[  4] local 192.168.2.2 port 50126 connected to 192.168.2.20 port 5201
[  6] local 192.168.2.2 port 50127 connected to 192.168.2.20 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   144 MBytes  1.20 Gbits/sec
[  6]   0.00-1.00   sec   140 MBytes  1.17 Gbits/sec
[SUM]   0.00-1.00   sec   283 MBytes  2.37 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[... 1194 lines cut ...]
[32m[   85.132219] [0mfuse init (API version 7.23)
[32m[   85.683594] [0mSynotify use 16384 event queue size
[32m[   85.688698] [0mSynotify use 16384 event queue size
[32m[   87.156781] [0m[33mip_tables[0m: (C) 2000-2006 Netfilter Core Team
[32m[   87.177991] [0m[33maudit[0m: type=1325 audit(1638071208.189:2): table=filter family=2 entries=0
[32m[   87.200671] [0mnf_conntrack version 0.5.0 (65536 buckets, 262144 max)
[32m[   87.420756] [0m[33maudit[0m: type=1325 audit(1638071208.431:3): table=filter family=2 entries=4
[32m[   87.440869] [0m[33maudit[0m: type=1325 audit(1638071208.452:4): table=filter family=2 entries=8
[32m[   87.623329] [0m[33maudit[0m: type=1325 audit(1638071208.634:5): table=filter family=2 entries=9
[32m[   91.271188] [0mSynotify use 16384 event queue size
[32m[   96.620407] [0m[33mIPv6[0m: ADDRCONF(NETDEV_UP): eth2: link is not ready
[32m[  104.224967] [0m[33maqc111 2-1:1.0 eth2[0m: Link Speed 5000, USB 3
[32m[  104.236202] [0m[33mIPv6[0m: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[32m[  109.398603] [0m[33maudit[0m: type=1325 audit(1638071231.354:6): table=nat family=2 entries=0
[32m[  127.989969] [0m[33maudit[0m: type=1325 audit(1638071249.948:7): table=filter family=2 entries=10
[32m[  128.004587] [0m[33maudit[0m: type=1325 audit(1638071249.963:8): table=filter family=2 entries=9
[32m[  128.030017] [0m[1mModule [xt_recent] is removed. [0m
[32m[  128.055874] [0m[1mModule [xt_iprange] is removed. [0m
[32m[  128.072472] [0m[1mModule [xt_limit] is removed. [0m
[32m[  128.087740] [0m[1mModule [xt_state] is removed. [0m
[32m[  128.111401] [0m[1mModule [xt_multiport] is removed. [0m
[32m[  128.139287] [0m[1mModule [xt_LOG] is removed. [0m
[32m[  128.180453] [0m[33maudit[0m: type=1325 audit(1638071250.139:9): table=filter family=2 entries=8
[32m[  174.544440] [0mSynotify use 16384 event queue size
[01;32mtmccartn@Peabody[00m:[01;34m/[00m$ `

bb-qq, 2022-01-04:
There are no performance issues with the DS920+ generation hardware, so I suspect that there is something wrong with your network configuration.

So, could you please tell us more about the network configuration of how the PC and DS920+ are connected?

If you are using both built-in NICs and USB NICs, I am especially curious whether each subnet is split to avoid routing issues.

bb-qq, 2022-01-09:
Also, could you try to increase `vm.min_free_kbytes`?

* read: `sysctl -n vm.min_free_kbytes`
* write: `sysctl -w vm.min_free_kbytes=XXXX`

Brandoskey, 2022-02-26:
I seem to be having the same issue on a ds918+

Running iperf3 from the ds918 to a ds1821 with a synology 10gbe network card I get:

`[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   740 MBytes   620 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   740 MBytes   618 Mbits/sec                  receiver
[  7]   0.00-10.00  sec   737 MBytes   618 Mbits/sec    0             sender
[  7]   0.00-10.04  sec   736 MBytes   615 Mbits/sec                  receiver
[  9]   0.00-10.00  sec   720 MBytes   604 Mbits/sec    0             sender
[  9]   0.00-10.04  sec   720 MBytes   601 Mbits/sec                  receiver
[ 11]   0.00-10.00  sec   696 MBytes   584 Mbits/sec    0             sender
[ 11]   0.00-10.04  sec   696 MBytes   582 Mbits/sec                  receiver
[ 13]   0.00-10.00  sec   704 MBytes   591 Mbits/sec    0             sender
[ 13]   0.00-10.04  sec   704 MBytes   589 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  3.51 GBytes  3.02 Gbits/sec    0             sender
[SUM]   0.00-10.04  sec  3.51 GBytes  3.00 Gbits/sec                  receiver`

Running iperf3 from the ds918 to the ds1821 with the -R (reverse) flag i get:

`[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   521 MBytes   437 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   515 MBytes   432 Mbits/sec                  receiver
[  7]   0.00-10.00  sec   436 MBytes   365 Mbits/sec    0             sender
[  7]   0.00-10.00  sec   430 MBytes   361 Mbits/sec                  receiver
[  9]   0.00-10.00  sec   524 MBytes   440 Mbits/sec    0             sender
[  9]   0.00-10.00  sec   518 MBytes   435 Mbits/sec                  receiver
[ 11]   0.00-10.00  sec   446 MBytes   374 Mbits/sec    1             sender
[ 11]   0.00-10.00  sec   440 MBytes   369 Mbits/sec                  receiver
[ 13]   0.00-10.00  sec   443 MBytes   372 Mbits/sec    0             sender
[ 13]   0.00-10.00  sec   437 MBytes   367 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  2.32 GBytes  1.99 Gbits/sec    1             sender
[SUM]   0.00-10.00  sec  2.29 GBytes  1.96 Gbits/sec                  receiver`

Testing from another machine to the ds1821 with a 10gbe connection I get 10gbe both ways

All 3 machines run through a unifi flex xg 10gbe switch capable of multi gig (1/2.5/5/10) The ds918 reports 5000mbit full duplex

File transfers over smb and NFS show similar results

bb-qq, 2022-03-06:
@Brandoskey, have you tried to configure MTU to 9k?

It is known that some platforms do not perform well enough due to USB host controllers or other causes, but at least geminilake should be able to provide adequate performance.

> Is there a better device to use than the QNAP?

As far as I know, QNAP is the better device in terms of heat dissipation among the products using the AQC111U.

Brandoskey, 2022-03-06:
> @Brandoskey, have you tried to configure MTU to 9k?
>
> It is known that some platforms do not perform well enough due to USB host controllers or other causes, but at least geminilake should be able to [...]

It definitely appears to be a hardware limitation. Amazon reviews aren't good and I can't find any solid data showing up and down speeds being symmetrical.

Either that or I just got a bad unit.

I've tried it on my desktop and my ds918 and the ds918 actually performed better. When connected to my desktop I tried adjusting nearly every driver setting and never got better than the 2gbit I get when connected to the ds918

bb-qq, 2024-06-15:
I had not able to suggest increasing the MTU values of all communicating devices, so I'm leaving a link as a note.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#increase-mtu

Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

closed 2024-06-15 by bb-qq