bb-qq/aqc111 #58: DS918+ & Qnap UC51T keeps disconnecting

T13nou, 2021-12-06:
### Description of the problem

Hi, I own a DS918+ with a QNAP QNA-UC5G1T

The adaptater is pretty hot so I've runned the command for thermal throttling.

In any case, same result, the connection drops very often and I need to manually restart the drive from the package interface. 

### Description of your products

> NAS : Synology DS918+
> Output "uname -a" :  Linux DISKSTATION 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
> DSM version 7.0.1-42218
> Dongle : QNAP QNA-UC5G1T

### Description of your environment

> Dongle directly connected to the front port of the NAS
> NAS connected to a Qnap switch
> Jumbo frames not enabled
> All Cables CAT7

### Output of ``dmesg`` command

````
[470936.755006] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470936.761719] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470936.768430] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470936.775139] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470936.785163] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470936.791900] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.459842] net_ratelimit: 74 callbacks suppressed
[470945.465297] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.472021] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.478746] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.494474] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.501206] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.508324] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.515054] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.522127] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.528863] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470945.536021] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.646622] net_ratelimit: 4415 callbacks suppressed
[470952.652272] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.659002] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.667939] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.674676] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.681402] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.688133] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.695752] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.702489] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.709215] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470952.716282] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470961.555953] net_ratelimit: 71 callbacks suppressed
[470961.561407] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470961.568120] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470961.574831] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470961.581546] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.525885] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.532616] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.540046] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.546771] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.553495] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.560809] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.567542] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.575916] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.582645] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470975.589776] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[470986.749325] net_ratelimit: 74 callbacks suppressed
[470986.754776] aqc111 2-1:1.0 eth2: kevent 2 may have been dropped
[... 1390 lines cut ...]
          RX bytes:1458316403 (1.3 GiB)  TX bytes:8205626818 (7.6 GiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:A1:4D:A4  
          inet addr:169.254.76.235  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr D2:C0:41:81:DA:B1  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:218936 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2284395 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:176333609 (168.1 MiB)  TX bytes:333540100 (318.0 MiB)

````

patrickenfuego, 2021-12-11:
I am having the same issue with a nearly identical configuration - the only difference from @T13nou is that I'm running DSM 6.2. The driver works well for a while and then fails, requiring me to either reboot the NAS or restart the driver package if I'm able to connect to the NAS via a different NIC (sometimes I can, sometimes not).

Globetrotter27, 2021-12-26:
Same here. Ds620Slim and this QNAP adapter, DSM 7. 
Can work steady for 7 days and can disconnect after several hours, no regularity detected. Also keeping onboard 1G connection as a backup.

Trying to resolve the issue, I have installed additional USB fan blowing on adapter and switch (also very hot). Now they are both ice cold, so temperature is not the cause here.

wdwms, 2021-12-26:
Ditto here on a DS920+; i'm now rebooting every monday as a precaution.. It starts to fail after about a week.
My DS418play+ doesn't exhibit the problem.

sparkplug37, 2021-12-27:
My DS918+ running DSM 7.0 via a Sonnet Solo5G also drops the USB Ethernet connection once  every two weeks or so. I generally shut down and restart the driver package and its all good to go. I've not been able to determine a cause so far but I only use this connection for moving things around from one computer. I use the onboard ethernet for all other devices on my network to connect to so it does not get much traffic. BTW I've not had much luck using the front USB (very slow speeds) so always use the rear USB port.

patrickenfuego, 2021-12-27:
@sparkplug37 that's interesting, I'll move mine to the rear and see if there is an improvement. I've been using the front as I believe that's what @bb-qq recommended? Maybe I'm wrong.

How is everyone handling failover? I tried to do an active/passive link aggregation with the onboard NIC, but it didn't work (no errors were thrown, but the interface for the 5G adapter completely disappeared until I removed the agg). I'm trying to find a way to have It failover to the 1G NIC if the adapter/driver fails because I need close to 24/7 uptime. I'm using the QNAP adapter.

sparkplug37, 2021-12-27:
I tested both front and rear USB and got about .5gbe on the front usb and over 2gbe front the rear tested with iperf3 over a 10gbe switch. I also liked the rear fans cooling the adapter as an added bonus. 

I have no failover as I just notice it’s down and use the onboard Ethernet to get back in. It seems 1 in 8 failures means a driver restart is not enough so a box reboot gets it back.

sparkplug37, 2021-12-29:
So my ethernet adaptor stopped again this morning and I caught this call trace output from dmesg along with many 'aqc111 2-2:1.0 eth2: kevent 2 may have been dropped' messages. I  do not know if its helpful to anyone much more in the know or even if its related to the issue.

[210375.271752] ------------[ cut here ]------------
[210375.277025] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x241/0x250()
[210375.286573] NETDEV WATCHDOG: eth2 (aqc111): transmit queue 0 timed out
[210375.293973] Modules linked in: aqc111(OE) usbnet(OE) mii(OE) nfnetlink xfrm_user xfrm_algo fuse xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack xt_addrtype 8021q vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) udf isofs loop cifs synoacl_vfs(PO) tun nf_conntrack_ipv6 ip6table_filter ip6_tables ipt_MASQUERADE xt_REDIRECT nf_nat_masquerade_ipv4 xt_nat iptable_nat nf_nat_ipv4 nf_nat_redirect nf_nat xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 iptable_filter ip_tables x_tables btrfs ecryptfs zstd_decompress zstd_compress xxhash raid456
[210375.373680]  async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper syno_hddmon(P) openvswitch gre nf_defrag_ipv6 nf_conntrack apollolake_synobios(PO) hid_generic usbhid hid usblp zram i915 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cfbfillrect cfbcopyarea cfbimgblt drm drm_panel_orientation_quirks iosf_mbi fb fbdev video backlight button uhci_hcd ehci_pci ehci_hcd usb_storage r8168(O) i2c_algo_bit igb(O) e1000e(O) sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd sha256_generic ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance
[210375.451700]  acpi_cpufreq processor cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel etxhci_hcd xhci_pci xhci_hcd usbcore usb_common [last unloaded: mii]
[210375.466332] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE   4.4.180+ #42218
[210375.474984] Hardware name: Synology DS918+/DS918+, BIOS M.034 11/11/2019
[210375.482568]  0000000000000000 ffff88047fc03dc0 ffffffff812ef89d 0000000000000009
[210375.490944]  ffff88047fc03e08 ffff88047fc03df8 ffffffff8105255f 0000000000000000
[210375.499318]  ffff880440916000 0000000000000000 0000000000000001 ffff88047fc11300
[210375.507690] Call Trace:
[210375.510515]  <IRQ>  [<ffffffff812ef89d>] dump_stack+0x4d/0x70
[210375.517046]  [<ffffffff8105255f>] warn_slowpath_common+0x7f/0xb0
[210375.523854]  [<ffffffff810525d9>] warn_slowpath_fmt+0x49/0x50
[210375.530366]  [<ffffffff814f4691>] dev_watchdog+0x241/0x250
[210375.536591]  [<ffffffff814f4450>] ? qdisc_rcu_free+0x40/0x40
[210375.543009]  [<ffffffff814f4450>] ? qdisc_rcu_free+0x40/0x40
[210375.549430]  [<ffffffff810b0735>] call_timer_fn+0x35/0x150
[210375.555653]  [<ffffffff814f4450>] ? qdisc_rcu_free+0x40/0x40
[210375.562075]  [<ffffffff810b0cf8>] run_timer_softirq+0x178/0x260
[210375.568788]  [<ffffffff810bf1f8>] ? clockevents_program_event+0xc8/0x120
[210375.576375]  [<ffffffff81056d5c>] __do_softirq+0xdc/0x300
[210375.582502]  [<ffffffff810b3cf6>] ? hrtimer_interrupt+0xb6/0x180
[210375.589309]  [<ffffffff810570dc>] irq_exit+0x9c/0xb0
[210375.594941]  [<ffffffff81035223>] smp_apic_timer_interrupt+0x43/0x60
[210375.602128]  [<ffffffff8157a9db>] apic_timer_interrupt+0x9b/0xa0
[210375.608935]  <EOI>  [<ffffffff8100e1ad>] ? default_idle+0x2d/0x100
[210375.615946]  [<ffffffff8100efb0>] arch_cpu_idle+0x10/0x20
[210375.622075]  [<ffffffff8108eb68>] default_idle_call+0x28/0x40
[210375.628592]  [<ffffffff8108ed25>] cpu_startup_entry+0x1a5/0x340
[210375.635301]  [<ffffffff81574442>] rest_init+0x72/0x80
[210375.641040]  [<ffffffff818f1e92>] start_kernel+0x3e8/0x3f5
[210375.647263]  [<ffffffff818f12f3>] x86_64_start_reservations+0x2a/0x2c
[210375.654553]  [<ffffffff818f13fb>] x86_64_start_kernel+0x106/0x113
[210375.661479] ---[ end trace 424dcc8a7b706a2e ]---

This adaptor drop seemed to coincide with me disconnecting an IP camera connected to a different switch, as I had been using the interface, disconnected the camera (3 minutes max) and to come back to see I could no longer ping the adaptors IP.

After I restarted the driver all was good again.

wdwms, 2022-01-01:
Feeling like we're abandoned here.. I don't think the developer has posted in months.. :(

bb-qq, 2022-01-04:
Did you try disabling thermal throttling and enabling  Low Power 5G?
https://github.com/bb-qq/aqc111#private-flags

> I'll move mine to the rear and see if there is an improvement. I've been using the front as I believe that's what @bb-qq recommended? Maybe I'm wrong.

In my environment (DS918+), the front port works more stably.
However, there seem to be individual differences between hardware lot and models, and some environments seem to be more stable with the rear port.

sparkplug37, 2022-01-04:
Yes thermal throttling is enabled and here are the current config:

Private flags for eth2:
Low Power 5G      : off
Thermal throttling: on

The interface does not seem to get overly hot either. I did try enabling low power mode but this causes the speed to drop off below the internal LAN ports.

bb-qq, 2022-01-04:
> I did try enabling low power mode but this causes the speed to drop off below the internal LAN ports.

How about enabling low power mode with MTU 9000? I tried the same setting in my environment (QNA-UC5G1T / DS918+), and the performance did not drop drastically. So if this configuration is also usable for you, I would like to ask you to check whether there is any difference in stability.

sparkplug37, 2022-01-07:
So I will give the MTU 9000 and low power mode a go. Though an MTU of 9000 is not great as not all things on my network like it.

As for bandwidth I get the following results with iPerf3

standard MTU, normal power and thermal throttling - 2 - 2.4Gbits/sec
standard MTU, low power and thermal throttling - <0.5 Gbits/sec
9000 MTU, normal power and thermal throttling - 3 - 3.4Gbits/sec
9000 MTU, low power and thermal throttling - 1.6 - 2.2Gbits/sec

bb-qq, 2022-01-09:
I understand that the MTU 9000 setting is difficult to use in your environment, but did enabling "Low Power Mode" change the stability of your system? I'd like to know the results because knowing this will help me isolate the problem.

---
Just FYI:

> aqc111 2-2:1.0 eth2: kevent 2 may have been dropped

kevent 2(`EVENT_RX_MEMORY`) means failure of memory allocations for received data(skb). This can be ignored, but many errors lead to degraded performance.
Increasing `vm.min_free_kbytes` can help to decrease the number of errors under low memory environments, but it can not be eliminated completely.

If you try to change this setting, use these commands:

* read: `sysctl -n vm.min_free_kbytes`
* write: `sysctl -n vm.min_free_kbytes=XXXX`

sparkplug37, 2022-01-09:
So I have been running the low power mode and 9000 MTU without an interface drop for the past few days. I will continue and report back. I have also doubled the vm.min_free_kbytes from the initial 65536 value and see if the warnings lessen. 

Many thanks

sparkplug37, 2022-01-11:
FYI I still have not had an interface drop since changing to low power mode and using a MTU of 9000. 

I'll report back again in a few days time.

sparkplug37, 2022-01-17:
So still not adaptor disconnects since changing both to low power mode and 9000MTU.

wdwms, 2022-01-18:
Only solution I have right now is to shutdown and restart my system at 3am every 3 days. :(

wdwms, 2022-01-19:
Would like to try it, what are the exact commands to run, and after running them do they "stick" after a reboot?

sparkplug37, 2022-01-19:
> Would like to try it, what are the exact commands to run, and after running them do they "stick" after a reboot?

Just ssh/putty into the box using a command like 
  ssh <username>@<IP address> -p <port> 
so:
  ssh fred@192.168.1.25 -p 22

Once logged in type to enable low power mode:

  sudo ethtool --set-priv-flags eth2 "Low Power 5G" on

Then change the MTU in GUI.

Globetrotter27, 2022-01-19:
> Would like to try it, what are the exact commands to run, and after running them do they "stick" after a reboot?

Yesterday decided to give it another try. But ALWAYS was using MTU 9000 from the start, and I think experimented with low power also when the issues started.

For the moment, 24 hours looks OK. But I definitely rememeber, there were couple of times when link disconnected under heavy load (like I was moving on purpose 100s of GB to desktop)

sparkplug37, 2022-01-20:
> Would like to try it, what are the exact commands to run, and after running them do they "stick" after a reboot?

FYI  It appears the low power mode flag did not survive the OS update though the MTU change did.

sparkplug37, 2022-01-20:
So I rebooted my box yesterday to do the OS update and on power up the MTU was still set though the low power option was not and the driver has already died. So I have set it back to low power and we shall see how long it works for.

bb-qq, 2022-01-21:
Thank you for your testing.

From this result, I am suspecting this is a power consumption issue. So if you do not prefer to set MTU to 9000 with low power mode, it might be worth trying to use Y-cable or self-powered USB hub with an AC adapter, but I'm not certain.

The reason for my assumption is that one of the users had observed voltage drops of USB under heavy load using a USB analyzer and reported that it was resolved by using a Y-cable.
https://github.com/bb-qq/aqc111/issues/36#issue-812672752

I don't know if the voltage drop is due to the load on the host side such as CPU/HDD, or because the dongle consumes more power, or both. But I think it's possible to relate since the QNA-UC5G1T consumes 4.5W of power, which is a lot for a USB device.

sparkplug37, 2022-01-21:
Thats really interesting so I have just ordered a USB 3 y cable, with luck should arrive later today and I'll let you know if it helps. Many thanks for the suggestion.

sparkplug37, 2022-01-22:
So I got a y splitter cable and connected the power to an external USB power supply, I restarted the NAS, so lost the low power setting so the speed was great, but the the driver has died twice in less then 24 hours. So I have reverted to the low power mode to ensure its solid.

bb-qq, 2022-01-23:
I'm sorry to hear that. Sadly there seems to be no further solution that I can suggest.

My DS918+ is working stably, but I don't know what's different between the problematic environment and my environment.
It could be due to individual hardware differences, or it could be due to the fact that I only have three HDDs and have margin of power. (Even if that is the case, this seems difficult to confirm.)

Globetrotter27, 2022-01-23:
Same here. Tried using connection through USB hub with external power (5V 1A), no luck. Connection dropped within minutes after starting transfers. BTW using 5 HDDs + 1 SSD in my chassis. Maybe some time later will try experiment with another PS - will 2A power will do the magic? Now have to revert to Low power, which gives 2-2,5 GBPS with 1 stream SMB.

Knowing this limitation earlier, I would probably go with 2,5G adapter and switch. Here I expected 3,5 G throughput.

bb-qq, 2022-01-24:
> will 2A power will do the magic?

Since the USB device descriptor claims that power consumption is 896mA, a 5V-1A power supply should be enough. But if the device consumes more power due to some kind of glitch, there is the possibility of improving the situation.

If it is a problem under high load, it would be helpful if you could try setting the MTU to 16k and see if there is a difference in how often the problem occurs. Increasing MTU will not further improve the throughput, but it will reduce the number of USB packets, which should lower the processing load on the device a bit.

It can only be set from CUI:
```
sudo ip link set eth2 mtu 16334
```

Since this setting consumes more memory, [min_free_kbytes](https://github.com/bb-qq/aqc111/issues/58#issuecomment-1008239739) should be larger.
Also, if the SMB Protocol Version is not set to v3, please set that as well.

bb-qq, 2022-01-26:
I have created a special version that reduces the queue size for USB transfers to suppress the bursty memory allocation.

I don't know if this will stabilize the device, but it is expected to reduce the frequency of memory allocation failure errors. I ran CristalDiskMark on my environment, and the errors seem to have decreased somewhat. Try it if you have time.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-5/aqc111-apollolake-1.3.3.0-6_rx_urb_queue_test.spk

sparkplug37, 2022-01-27:
Many thanks for the updated driver, I will try it out in a few days time after I finish my current tests.

So a few weeks back I noticed my SSD read cache (2 x 128GB) had stopped working and was no longer being used. I kept the drives in the unit as it was disabled and thought nothing of it. But a few days ago I investigated a bit more to find one drive was being seen, so after a bit of shutting down the NAS and swapping the drives around I can see only one drive worked. So I removed the dead drive, re-enabled the read cache on 1 x 128gb drive and ever since my ethernet connection has not dropped.

Before I could get it go wrong by doing : 
  iperf3 -c <NAS IP address> -t 4000

BTW I have a docker iperf3 server running on the NAS.

Since then I have repeated this command from my Mac, using a thunderbolt 10G Nic, ten or more times without any issues so far and I have not reverted to low power mode though I still use a 9000 MTU for both Mac and NAS.

I will continue my tests and report back in a few days time and I'm also using the front USB port of the NAS as it was the easiest to get to when I was swapping drives around and as its been working well ever since I have not bothered to change it.

FYI my NAS spec is:
DS918+
4 x IronWolf 14TB drives (ST14000VN0008) - Noisy as hell, they clatter (80's HD style) even on idle without any services running and not that fast.
1 x Transcend 128GB SSD (TS128GMTE110S) in slot 1 - read cache enabled
16GB memory (2 x 8GB)
1 x Sonnet Solo5g - USB3 to 5 Gigabit Ethernet (5GbE) Adapter

sparkplug37, 2022-02-02:
A little catchup, so I moved my NAS (I'm moving home soon so I'm trying to tidy things up) so at the same time updated the driver to your test release above. With that said before and since the update I have not had a single drop. I have been really pushing the NAS and as its not on low power mode and running 9000 MTU so I get 3.5Gbits/sec using iPerf3 so I really do not believe I was seeing either a power issue or throughput issue.

So it seems removing an unused, but problematic SSD, fixed my issues.

Many thanks for all you help.

bb-qq, 2022-02-05:
@sparkplug37, thank you for the status update. It looks like the problem has been resolved.

@wdwms and @Globetrotter27, If you still need some help, can you create a separate ticket? I would like to track the reports separately, because if symptoms seem to be the same, the environment and cause may not. 

@T13nou. have you tried tweaks or the test version of the driver mentioned in this thread? Let me know how it goes now.

iCassius, 2022-02-23:
I havd same issus
tried front and back USB port , always lost  net connect after few hours 
but i can ping to gateway and using "route" command with very slow.

`[16355.553603] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x241/0x250()
[16355.563071] NETDEV WATCHDOG: eth2 (aqc111): transmit queue 0 timed out
[16355.570395] Modules linked in: aqc111(OE) usbnet(OE) mii(OE) nfnetlink xfrm_user xfrm_algo pci_st                                                                                            ub vfio_pci vfio_virqfd vfio_iommu_type1 vfio vhost_net(O) kvm_intel kvm irqbypass fuse xt_ipvs ip_v                                                                                            s_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack xt_addrtype                                                                                             nf_conntrack_ipv6 ip6table_filter ip6_tables ipt_MASQUERADE xt_REDIRECT nf_nat_masquerade_ipv4 xt_n                                                                                            at iptable_nat nf_nat_ipv4 nf_nat_redirect nf_nat xt_recent xt_iprange xt_limit xt_state xt_tcpudp x                                                                                            t_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 iptable_filter ip_tables x_tables 8021q vhost_sc                                                                                            si(O) vhost(O) tun tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core                                                                                            _multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rod                                                                                            sp_ep(O) udf
[16355.650112]  isofs loop synoacl_vfs(PO) btrfs ecryptfs zstd_decompress zstd_compress xxhash raid4                                                                                            56 async_raid6_recov async_memcpy async_pq async_xor xor async_tx raid6_pq leds_lp3943 aesni_intel g                                                                                            lue_helper lrw gf128mul ablk_helper syno_hddmon(P) apollolake_synobios(PO) hid_generic usbhid hid us                                                                                            blp i915 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cfbfillrect cfbcopyarea cfbimg                                                                                            blt drm drm_panel_orientation_quirks iosf_mbi fb fbdev video backlight button uhci_hcd ehci_pci ehci                                                                                            _hcd usb_storage r8168(O) i2c_algo_bit igb(O) e1000e(O) sg dm_snapshot dm_bufio crc_itu_t crc_ccitt                                                                                             psnap p8022 llc hfsplus md4 hmac sit tunnel4 openvswitch gre nf_defrag_ipv6 ipv6 nf_conntrack zram f                                                                                            lashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd sha256_ge                                                                                            neric ecb
[16355.727952]  aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_perf                                                                                            ormance acpi_cpufreq processor cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel etxhci_hcd xh                                                                                            ci_pci xhci_hcd usbcore usb_common [last unloaded: mii]
`

bb-qq, 2022-03-13:
@iCassius, did you try enabling Low-Power mode? I appreciate it if you create a separate issue.

mcnahum, 2022-04-18:
Hello,
I'm on a DS 720+ and the issue is the same ...
I tried to install the beta driver but it seams to not be compatible with my NAS
I did the "sudo ethtool --set-priv-flags eth2 "Low Power 5G" on" for now and I will see if it's more stable.

Will you do the beta driver who should fix the memory leak for my device?

ihvar, 2022-04-19:
Hi!
On my Synology DS418play adapter keeps disconnecting. "Low Power 5G", "MTU" settings don't work. Every time connection is lost, only power-off-on procedure makes QNAP working...

bb-qq, 2022-05-01:
I suspect that in cases where the symptoms do not improve in LowPower mode, memory consumption is not the cause of the problem, but the cause is also the type of device and the cable that connects to that device.
(I too encountered the same symptoms when using a different USB cable or using an L-shaped USB adapter.)

However, I am not sure because I have not been able to sort out that information. If anyone with the same symptoms could create an independent issue and fill in the information according to the template, it would help me in my investigation.

bb-qq, 2022-08-11:
In priority to stability, the settings discussed here have been applied by default in recent releases.

Since there are so many mixed reports on this issue that it is difficult to track the status individually, we would like to close this issue.
If you still have stability issues using the latest drivers, please create a new issue separately. It would also be useful for other users.

closed 2022-08-11 by bb-qq