bb-qq/aqc111 #61: DS1019+ - UC5G1T suddenly writes slower

emanresu2, 2022-01-27:
### Description of the problem
Not quite sure if this is even related to the driver.
Initially I had the Club 3D 2.5G adapter (with r8152-apollolake-2.15.0-5.spk) and it ran without problems. Reading and writing reached around 280 MB/sec.
Yesterday I got my hands on a QNA-UC5G1T and installed it just to see how much of a difference it makes (using aqc111-apollolake-1.3.3.0-5.spk). I can't quite remember but I'm pretty sure the write speed increased to about 360 MB/sec. Reading went up to 390 MB/sec.
Today I noticed that the write speed went down to around 220 MB/sec (reading is still at 390). I rebooted NAS, PC and router (just in case), reinstalled the driver, used the other USB port and checked if the Club 3D adapter still performs as before (it does).
The only thing I didn't test was replacing the USB C to A cable because I noticed that I don't have one with USB 3.

In short:
2.5G: 280/280
5G yesterday: 360/390
5G today: 220/390

I have no idea what changed.

### Description of your products

DS1019+
Linux DS1019 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_1019+
DSM 7.0.1-42218 Update 2
QNAP QNA-UC5G1T

### Description of your environment

DS1019+ is directly connected to an AQtion AQC107 (10G, onboard) with a CAT 8.1 cable.

### Output of ``dmesg`` command

````
[   26.909814] sd 3:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[   26.909896] sd 3:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   26.986772]  sdb: sdb1 sdb2 sdb3
[   26.990982] sd 3:0:0:0: [sdb] Attached SCSI removable disk
[   26.999489] Freeing unused kernel memory: 916K
[   27.004469] Write protecting the kernel read-only data: 8192k
[   27.011235] Freeing unused kernel memory: 508K
[   27.017545] Freeing unused kernel memory: 300K
[   27.128551] ACPI: bus type USB registered
[   27.133084] usbcore: registered new interface driver usbfs
[   27.139254] usbcore: registered new interface driver hub
[   27.145234] usbcore: registered new interface driver ethub
[   27.151391] usbcore: registered new device driver usb
[   27.191011] xhci_hcd 0000:00:15.0: xHCI Host Controller
[   27.197121] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 1
[   27.406533] xhci_hcd 0000:00:15.0: hcc params 0x200077c1 hci version 0x100 quirks 0x01109010
[   27.415964] xhci_hcd 0000:00:15.0: cache line size of 64 is not supported
[   27.416472] hub 1-0:1.0: USB hub found
[   27.420716] hub 1-0:1.0: 8 ports detected
[   27.425256] host 0000:00:15.0 port 1 has Vbus GPIO#13 with polarity ACTIVE_HIGH
[   27.433786] host 0000:00:15.0 port 2 has Vbus GPIO#11 with polarity ACTIVE_HIGH
[   27.442812] xhci_hcd 0000:00:15.0: xHCI Host Controller
[   27.448663] xhci_hcd 0000:00:15.0: new USB bus registered, assigned bus number 2
[   27.456992] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   27.466427] hub 2-0:1.0: USB hub found
[   27.470654] hub 2-0:1.0: 7 ports detected
[   27.475192] host 0000:00:15.0 port 1 has Vbus GPIO#13 with polarity ACTIVE_HIGH
[   27.483730] host 0000:00:15.0 port 2 has Vbus GPIO#11 with polarity ACTIVE_HIGH
[   27.518840] apollolake_synobios: loading out-of-tree module taints kernel.
[   27.526531] apollolake_synobios: module license 'Synology Inc.' taints kernel.
[   27.534599] Disabling lock debugging due to kernel taint
[   27.551482] synobios open /dev/ttyS1 success
[   27.567499] correction with 0x00
[   27.572199] 2022-1-27 15:26:14 UTC
[   27.576001] synobios: load, major number 201
[   27.580772] Brand: Synology
[   27.583890] Model: DS-1019+
[   27.587009] This is default settings: set group disks wakeup number to 1, spinup time deno 1
[   27.596454] synobios cpu_arch proc entry initialized
[   27.602001] synobios crypto_hw proc entry initialized
[   27.607644] synobios syno_platform proc entry initialized
[   27.622414] Module [apollolake_synobios] is removed.
[... 1768 lines cut ...]

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:BD:6F:B2
          inet addr:169.254.191.175  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:65:53:D7
          inet addr:172.16.254.1  Bcast:172.16.254.3  Mask:255.255.255.252
          inet6 addr: fe80::265e:beff:fe65:53d7/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:17518553 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1643314 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:24784211026 (23.0 GiB)  TX bytes:12135845378 (11.3 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2022-02-05:
How did you measure the throughput?
Also, can you try setting the MTU to 9000 to see if that improves the throughput?

emanresu2, 2022-02-07:
To measured the speed I just started copying a large file (always the same file) to/from the NAS via a mapped network drive in Windows and looked at the speed it settles at after a few seconds. It's not very scientific.
I did find a program called "NAS performance tester" which basically does the same but slightly fancier.

I set the MTU to 9000 and that did something. Here are the results reported by that NAS tester:

Club 3D. The writing got slower but reading faster.

Running a 2000MB file write on X: 5 times...
Iteration 1:    295,22 MB/sec
Iteration 2:    290,58 MB/sec
Iteration 3:    200,15 MB/sec
Iteration 4:    194,64 MB/sec
Iteration 5:    178,86 MB/sec
**Average (W):    231,89 MB/sec**

Running a 2000MB file read on X: 5 times...
Iteration 1:    305,12 MB/sec
Iteration 2:    306,10 MB/sec
Iteration 3:    301,41 MB/sec
Iteration 4:    306,42 MB/sec
Iteration 5:    307,53 MB/sec
**Average (R):    305,32 MB/sec**

QNAP. Writing is roughly back to where it was.

Running a 2000MB file write on X: 5 times...
Iteration 1:    409,45 MB/sec
Iteration 2:    406,10 MB/sec
Iteration 3:    341,45 MB/sec
Iteration 4:    308,99 MB/sec
Iteration 5:    294,06 MB/sec
**Average (W):    352,01 MB/sec**

Running a 2000MB file read on X: 5 times...
Iteration 1:    425,19 MB/sec
Iteration 2:    443,75 MB/sec
Iteration 3:    395,01 MB/sec
Iteration 4:    447,17 MB/sec
Iteration 5:    432,61 MB/sec
**Average (R):    428,75 MB/sec**

Unfortunately the QNAP adapter works somewhat unstable. I did a stress test and at around 7 TB of reading/writing I went and  downloaded a thing (25 MB/sec or less) to the network drive and it just went unresponsive. Any program that tried to access the drive just froze.
The DiskStation still worked normally over the 1G port, LAN 3 still says connected and the logs don't mention anything.
Only the Windows Event Viewer says "{Delayed Write Failed} Windows was unable to save all the data for the file \\172.16.254.1\home\Files\Downloads\Unconfirmed 64677.crdownload; the data has been lost. This error may be caused by network connectivity issues. Please try to save this file elsewhere."

In further tests it would also happen after just a few GB without additional load. It seems to happen every few minutes now if too much data is being transferred.
Getting it to work again is a bit weird too. In rare cases the network drive becomes available again after like 2 minutes. Sometimes just restarting my PC helps and other times I have to restart the DiskStation. It seems random.
I already tried a completely fresh install of Windows 11 to rule out problems caused by any program but it didn't help.

Also tried both USB ports and disabling the SSD read/write cache.

The speed also varies quite a bit. Here is another QNAP test I just took:

Running a 2000MB file write on X: 5 times...
Iteration 1:    409,73 MB/sec
Iteration 2:    284,62 MB/sec
Iteration 3:    212,69 MB/sec
Iteration 4:    207,02 MB/sec
Iteration 5:    247,85 MB/sec
**Average (W):    272,38 MB/sec**

Running a 2000MB file read on X: 5 times...
Iteration 1:    447,42 MB/sec
Iteration 2:    435,77 MB/sec
Iteration 3:    448,78 MB/sec
Iteration 4:    432,71 MB/sec
Iteration 5:    441,15 MB/sec
**Average (R):    441,17 MB/sec**

My network knowledge is kind of limited so I don't know if that's expected or not. Maybe the DiskStation was busy with something else, though I wouldn't know what. Or maybe it's the program.

bb-qq, 2022-03-06:
I guess that instability of performance is caused by storage and cache. So could you try to measure pure network throughput using iperf?
http://www.jadahl.com/iperf-arp-scan/DSM_7.0/

bb-qq, 2022-08-11:
I close this issue due to no response. Please feel free to re-open.

closed 2022-08-11 by bb-qq