bb-qq/aqc111 #69: DS1618+ DSM7.1 UC5G1T

jackhudson2001, 2022-05-01:
DS1618+ upgraded to DSM7.1 using the qnap UC5G1T adaptor it now looses connection after about 15 mins of use.
was working fine before this latest upgrade.

bb-qq, 2022-05-04:
With one or both of the following commands applied, does the frequency of problems change?

```
sudo /sbin/sysctl -w vm.min_free_kbytes=1024000
```

```
sudo ethtool --set-priv-flags eth4 "Low Power 5G" on
````

jackhudson2001, 2022-05-04:
sudo /sbin/sysctl -w vm.min_free_kbytes=1024000 this command seems better, i'll let you know further on

andmar197264, 2022-05-25:
Hello! I confirm the problem of the qnap 5G adapter on DSM 7.1! On the previous version of DSM 7.01 it worked like a Swiss watch 👍 I have a DS720+

andmar197264, 2022-05-25:
Are these commands suitable for Gemini?

andmar197264, 2022-05-25:
Do I still need to use the commands? $ ethtool --set-priv-flags eth1 "Thermal throttling" off

$ ethtool --set-priv-flags eth1 "Low Power 5G" on

andmar197264, 2022-05-25:
Synology_NAS:~$ ifconfig
bond0     Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          inet addr:192.168.1.80  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fec9:6ed8/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:11321232 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10807780 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6308242590 (5.8 GiB)  TX bytes:89775668192 (83.6 GiB)

docker0   Link encap:Ethernet  HWaddr 02:42:CD:3E:7E:6F
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:cdff:fe3e:7e6f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:473716 errors:0 dropped:0 overruns:0 frame:0
          TX packets:341243 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12427523955 (11.5 GiB)  TX bytes:6457393065 (6.0 GiB)

dockerc5b Link encap:Ethernet  HWaddr BA:F3:FF:24:E4:D3
          inet6 addr: fe80::b8f3:ffff:fe24:e4d3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:291611 errors:0 dropped:0 overruns:0 frame:0
          TX packets:152783 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:12422136395 (11.5 GiB)  TX bytes:1080788039 (1.0 GiB)

dockerdfb Link encap:Ethernet  HWaddr 5E:01:2E:D2:37:05
          inet6 addr: fe80::5c01:2eff:fed2:3705/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:182105 errors:0 dropped:0 overruns:0 frame:0
          TX packets:189049 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:12019584 (11.4 MiB)  TX bytes:5376684403 (5.0 GiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:84508 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1095246 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5070480 (4.8 MiB)  TX bytes:325015691 (309.9 MiB)
          Interrupt:97 base 0xe000

eth1      Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:11236724 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9712534 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6303172110 (5.8 GiB)  TX bytes:89450652501 (83.3 GiB)
          Interrupt:96 base 0xc000

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:52:5C:AA
          inet addr:192.168.1.200  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe52:5caa/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:4790598 errors:1 dropped:0 overruns:0 frame:0
          TX packets:3064067 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7451656603 (6.9 GiB)  TX bytes:10097664312 (9.4 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4034051 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4034051 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1052316620 (1003.5 MiB)  TX bytes:1052316620 (1003.5 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.19.0.156  P-t-P:10.19.0.156  Mask:255.255.0.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:1371 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2200 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:214479 (209.4 KiB)  TX bytes:211939 (206.9 KiB)

andmar197264, 2022-05-25:
Reply to dmeg

[252038.496731] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252038.503612] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252039.155933] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252043.571998] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252043.578717] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252043.585426] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252043.592135] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252043.598844] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252119.962095] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252119.968819] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252123.075059] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252123.962855] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252123.969575] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252124.303508] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252124.310227] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252125.959601] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252125.966320] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252125.973029] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252125.979737] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252125.986446] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252126.959494] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252129.369605] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252129.376325] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252129.957544] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252140.002163] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252140.008873] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252141.536581] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252141.543307] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252141.550024] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252141.556726] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252194.740166] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252194.746887] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252220.292017] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252220.298737] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252220.905446] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252220.912165] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252221.904453] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252221.911180] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252221.917892] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.500750] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.507472] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.514181] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.520890] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.527599] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.534309] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.541017] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.547864] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.554580] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252226.561292] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252231.721073] net_ratelimit: 3 callbacks suppressed
[252231.726438] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252231.733149] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252231.739860] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252232.898672] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252236.938909] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252236.945621] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252263.882143] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252266.879838] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252271.876829] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252272.876667] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252272.883408] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252272.890121] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252273.876145] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252274.876011] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252275.875774] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252279.873076] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252290.866463] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[252290.873184] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[... 126 lines cut ...]
[253624.098926] audit: type=1325 audit(1653463353.409:3078): table=filter family=2 entries=52
[253624.115029] audit: type=1325 audit(1653463353.425:3079): table=filter family=2 entries=53
[253624.130980] audit: type=1325 audit(1653463353.441:3080): table=filter family=2 entries=54
[253626.516894] aqc111 2-2:1.0 eth2: Link Speed 5000, USB 3
[253626.578634] audit: type=1325 audit(1653463355.889:3081): table=filter family=2 entries=55
[253630.767264] audit_printk_skb: 165 callbacks suppressed
[253630.773184] audit: type=1325 audit(1653463360.078:3137): table=filter family=2 entries=55
[253630.789031] audit: type=1325 audit(1653463360.100:3138): table=filter family=2 entries=44
[253630.804853] audit: type=1325 audit(1653463360.115:3139): table=filter family=2 entries=48
[... cut ...]
[253667.293681] audit: type=1325 audit(1653463396.609:3206): table=filter family=2 entries=59
[253667.310249] audit: type=1325 audit(1653463396.625:3207): table=filter family=2 entries=58
[253667.525703] audit: type=1325 audit(1653463396.841:3208): table=filter family=2 entries=57
[253667.542076] audit: type=1325 audit(1653463396.857:3209): table=filter family=2 entries=58
[253696.848798] audit: type=1325 audit(1653463426.169:3210): table=filter family=2 entries=59
[253696.866024] audit: type=1325 audit(1653463426.186:3211): table=filter family=2 entries=58

andmar197264, 2022-05-25:
@Synology_NAS:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00911F20345A74)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 09I03493)

bb-qq, 2022-05-28:
Executing `sudo /sbin/sysctl -w vm.min_free_kbytes=1024000` reduces the frequency with which the warning `kevent 2 may have been dropped` is output. I wasn't sure if this would improve stability, but @jackhudson2001 reports that it seems to do, so you may want to try this setting.

Once I have proof that this setting is effective, I will try to set it automatically on the driver's side. 
@jackhudson2001, what is your current situation?

bb-qq, 2022-08-11:
I close this issue due to no response.
The setting discussed in this issue is applied by default in the latest release. You need not configure this setting anymore.

closed 2022-08-11 by bb-qq