bb-qq/aqc111 #71: RS819 - QNA-UC5G1T - SMB Transfert from NAS disconnect it

Gustour, 2022-05-26:
### Description of the problem

When copying files FROM the NAS to my computer using SMB service (network drive mounted on computer), it opens the copy windows, but never start, and SMB crash, even with the build in NIC. I have to stop SMB (uncheck in the parameters) and restart it.

Copy from computer TO NAS works (1gb max, my network is 2.5bg so i have to find out why).

Browsing NAS webui thru Lan3 works.

Download Station using QNA-UC5G1T (Lan3) works (1gb max right now cause it's my isp speed).

Streaming with VideoStation/Infuse/VLC thru Lan3 works.

Everything looks great just this SMB copy FROM the NAS.

### Description of your products

> - RS819
>     - and output of ``uname -a`` command (i'll update it when i'll be home)
>     - DSM 7.0.1-42218 Update 3
> - QNA-UC5G1T

### Description of your environment

> - Using a hub
>     - PC: XIAOLO RTL8156B (https://www.amazon.fr/gp/product/B09DVRKFF5/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&th=1)
>     - Hub: QNAP QSW-2104-2T
> - cable category Cat.8

### Output of ``dmesg`` command

````
(i'll update it when i'll be home)
````

### Output of ``lsusb`` command

````
(i'll update it when i'll be home)
````

### Output of ``ifconfig -a`` command

````
(i'll update it when i'll be home)
````

bb-qq, 2022-05-28:
Could you try these commands?
```
sudo /sbin/sysctl -w vm.min_free_kbytes=1024000
sudo ethtool --set-priv-flags eth2 "Low Power 5G" on

Gustour, 2022-05-28:
Thanks for the answer. I did what you ask.
Same result, but smb didn't crash, i could still acces with the in build NIC.
I had to reboot the NAS in order to have the Lan3 back (reboot the driver didn't work, re install the driver didn't work).
I did a second try and the result was same.
Any other idea ?

bb-qq, 2022-08-02:
The above commands are not persistent over system reboots. 
If you see a difference by running these commands, try the latest release. This release automatically executes a series of commands.
https://github.com/bb-qq/r8152/releases/tag/2.15.0-9

Gustour, 2022-08-03:
Ok i'll give it a try today, but i think you did a wrong copy/paste, i'll try this version :
https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-7

Gustour, 2022-08-08:
Hi, can't install 1.3.3.0-8 too. Same problem, it ask to repair, but fail. I have a bond too.

bb-qq, 2022-08-09:
Oh sorry, I forgot to inform you about the new driver.

Could you paste the output of these commands using `1.3.3.0-8`?

```
sudo /var/packages/aqc111/scripts/start-stop-status stop
sudo /var/packages/aqc111/scripts/start-stop-status start

bb-qq, 2022-08-10:
```
insmod: ERROR: could not insert module /var/packages/aqc111/target/aqc111/mii.ko: Invalid module format
```

Hmm... This log indicates that loading `mii.ko` was failed due to invalid file content, but this file is identical to the one in 1.3.3.0-5. It's weird.

Could you try these steps?

1. Uninstall the package
2. Reboot system
3. Re-install the package (1.3.3.0-9)

If you face the same result:

4. run "sha1sum /var/packages/aqc111/target/aqc111/mii.ko"

Gustour, 2022-08-10:
ok, still not working, so i did what you wanted (but on 1.3.3.0-8, can't find -9)

````
root@NAS:~# sha1sum /var/packages/aqc111/target/aqc111/mii.ko
47d93f2740d1364b23b041c067e3990706c1c9d5  /var/packages/aqc111/target/aqc111/mii.ko
````

bb-qq, 2022-08-11:
The file seems to be fine.

Could you please paste the output of the `dmesg` again? Maybe there is some additional log output at the moment the module fails to load.

Gustour, 2022-08-11:
hope that help :

<details>
  <summary>dmesg</summary>

````
root@NAS:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.180+ (root@build16) (gcc version 7.5.0 (GCC) ) #42218 SMP Mon Oct 18 19:16:01 CST 2021
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] cma: Reserved 32 MiB at 0x000000007e000000
[    0.000000] On node 0 totalpages: 524288
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv0.1 detected in firmware.
[    0.000000] psci: Conflicting PSCI version detected.
[    0.000000] PERCPU: Embedded 15 pages/cpu @ffffffc07df78000 s24472 r8192 d28776 u61440
[    0.000000] pcpu-alloc: s24472 r8192 d28776 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: ip=off console=ttyS0,115200 root=/dev/md0 rw syno_hdd_detect=4,5,6,7 syno_hdd_enable=55,86,82,81 HddHotplug=4 syno_usb_vbus_gpio=23@xhci-hcd.2.auto@0,22@xhci-hcd.5.auto@0 syno_hw_version=RS819 hd_power_on_seq=4 ihd_num=4 netif_num=2 swiotlb=1 ahci_remap=4>0:0>4 audio_version=1012363 syno_fw_version=M.707 U-boot=2012.07-gfcbc9a2 (Dec 05 2018 - 17:24:57)
[    0.000000] SYNO GPIO hdd detect pin: 4 5 6 7
[    0.000000] SYNO GPIO hdd enable pin: 55 86 82 81
[    0.000000] Support HDD Hotplug.
[    0.000000] USB Vbus GPIO Control:
[    0.000000]  Gpp#23 - Host: xhci-hcd.2.auto      - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000]  Gpp#22 - Host: xhci-hcd.5.auto      - Port:1 - Polarity: ACTIVE_HIGH
[    0.000000] Synology Hardware Version: RS819
[    0.000000] Internal HD num: 4
[    0.000000] Internal netif num: 2
[    0.000000] SYNO: ahci remap initialized
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] software IO TLB: mapped [mem 0x7dc30000-0x7dc70000] (0MB)
[    0.000000] Memory: 2005140K/2097152K available (5156K kernel code, 304K rwdata, 1808K rodata, 256K init, 448K bss, 59244K reserved, 32768K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc0000000 - 0xffffffbdc2000000   (    32 MB actual)
                   fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
                   PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
                   modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
                   memory  : 0xffffffc000000000 - 0xffffffc080000000   (  2048 MB)
                     .init : 0xffffffc00094e000 - 0xffffffc00098e000   (   256 KB)
                     .text : 0xffffffc000280000 - 0xffffffc00094db24   (  6967 KB)
                     .data : 0xffffffc00099c000 - 0xffffffc0009e8140   (   305 KB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] Architected cp15 timer(s) running at 27.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x63a1e71a3, max_idle_ns: 440795203123 ns
[    0.000006] sched_clock: 56 bits at 27MHz, resolution 37ns, wraps every 4398046511093ns
[    0.000086] [RTK-TIMER0] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000098] [RTK-TIMER0] clocksource 0 register HZ
[    0.000108] clocksource: rtk_timer0_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 70787423951 ns
[    0.000159] [RTK-TIMER0] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000168] [RTK-TIMER1] set mode: CLOCK_EVT_MODE_SHUTDOWN
[    0.000177] [RTK-TIMER1] clocksource 1 register HZ
[    0.000186] clocksource: rtk_timer1_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 579169832313 ns
[    0.000510] [RTK_MEM_REMAP] initialized
[    0.000786] Calibrating delay loop (skipped), value calculated using timer frequency.. 54.00 BogoMIPS (lpj=270000)
[    0.000800] pid_max: default: 32768 minimum: 301
[... 1154 lines cut ...]
[70629.974109] audit: type=1325 audit(1660211569.270:131): table=filter family=2 entries=9
[70629.999016] audit: type=1325 audit(1660211569.300:132): table=filter family=2 entries=8
[70630.053107] Module [xt_recent] is removed.
[70630.150453] Module [xt_iprange] is removed.
[70630.200205] Module [xt_limit] is removed.
[70630.260529] Module [xt_state] is removed.
[70630.339950] Module [xt_tcpudp] is removed.
[70630.399901] Module [xt_multiport] is removed.
[70630.498116] Module [xt_LOG] is removed.
[70630.570946] Module [iptable_filter] is removed.
[70666.343966] audit: type=1325 audit(1660211605.641:133): table=filter family=2 entries=0
[70666.539292] audit: type=1325 audit(1660211605.841:134): table=filter family=2 entries=4
[... cut ...]
[70666.870957] audit: type=1325 audit(1660211606.171:136): table=filter family=2 entries=9
[70666.895982] audit: type=1325 audit(1660211606.201:137): table=filter family=2 entries=8
[70666.950416] Module [xt_recent] is removed.
[70667.069848] Module [xt_iprange] is removed.
[70667.129810] Module [xt_limit] is removed.
[70667.180016] Module [xt_state] is removed.
[70667.219487] Module [xt_tcpudp] is removed.
[70667.269397] Module [xt_multiport] is removed.
[70667.357842] Module [xt_LOG] is removed.
[70667.460123] Module [iptable_filter] is removed.
````
</details>

bb-qq, 2022-08-20:
Unfortunately, nothing in dmesg was outputting any logs that might be related to the problem.

Lastly, with the problematic package installed, could you please try to remove the script that changes the memory settings with the following command and reboot the system?
```
sudo rm /var/packages/aqc111/scripts/apply-memory-setting
```
This should make the situation exactly the same as the previous driver, but if you don't see any change, I have no idea.

Gustour, 2022-08-28:
same issue, doesn't install, ask if i want to repair. 

[image]

Gustour, 2022-08-31:
I am still on DSM 7.0.1-42218 Update 3 maybe that is the problem. But i dodn't know if your driver works with last available update 7.1-42661 Update 4.
Should I upgrade and give a try ?

bb-qq, 2022-09-06:
I still can't figure out why 1.3.3.0-5 works but the newer version doesn't work at all.
Perhaps the behavior is the same when DSM is upgraded.

Until I figure out the cause, I think it is best to install 1.3.3.0-5 and run the command written in https://github.com/bb-qq/aqc111/issues/71#issuecomment-1140167329 on every boot. (This command has no effect after rebooting.)

alanyenhk, 2022-09-15:
@bb-qq 
I have met the same problem but I found a way to get it work.

First, I removed the driver in DSM and reinstall the version 1.3.3.0-5 with provided steps.
Then, after confirmed the adapter has been shown on control panel, just update the driver to 1.3.3.0-8 in DSM. 

BOOM! the latest driver installed successfully.

I don't know why it's worked but hope the solution can help. I will have further tests to try the new driver.
Anyway, @bb-qq Thank you for your work!!! Wish you all the best!

[image]
[image]

Gustour, 2022-10-12:
Sorry, I had only few time those last days, and tried to install 1.3.3.0-8 again only yesterday.

@alanyenhk i tried what you did, but still can't manage to install any version over -5 :(

Gustour, 2022-10-14:
1.3.3.0-9 installed without problem ... We don't know why, but this install version works.

I'll try if my bug is still present.

Thanks

bb-qq, 2022-12-04:
The "can't repair" issue was fixed in the [1.3.3.0-10 release](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-10)

However, the original stability issue seems to be caused by rtd1296 SoC.
I decided to merge topics on rtd1296 into the issue [#275](https://github.com/bb-qq/r8152/issues/275) in the r8152 project.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq

bb-qq, 2023-01-28:
Are you using Safari?
There have been reports of problems with Safari regarding package installation.