bb-qq/aqc111 #72: Driver 1.3.3.0-5 DSM-7.0

joseefmpcc71, 2022-05-27:
### Description of the problem

> When I put the command: sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su
by SSH in my nas DS720+, It said invalid option - - '/' Why? .

### Description of your products

> - Synology NAS DS720+
>     - DSM version 7.0.1-42218 Update 3
> - QNA-UC5G1T

bb-qq, 2022-05-28:
What happens if I explicitly specify /bin?

`sudo /bin/install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su`

joseefmpcc71, 2022-05-28:
Hello,
First thanks you for your answer,
I could install the code but I have this erro yet.
Do you know what is the solution?
Regards

jsiryana, 2022-06-02:
also having the same issue

bb-qq, 2022-07-31:
Could you paste the outputs of these commands?

* `/bin/install --version`
* `/bin/install --help`

bb-qq, 2023-03-19:
I am closing this issue as I have not heard back in a long time. Feel free to re-open it if you have any questions.

closed 2023-03-19 by bb-qq