bb-qq/aqc111 #73: interface on DS416slim always show 'disconnected'

gzm55, 2022-06-07:
### Description of the problem

The status of the interface is  always 'disconnected'.

### Description of your products

- NAS model: DS416slim
    - and output of ``uname -a``: ``Linux DS416slim 3.10.108 #42661 SMP Fri Apr 1 15:29:19 CST 2022 armv7l GNU/Linux synology_armada38x_ds416slim``
    - DSM version 7.1
- the product name of the ethernet dongle: QNA-UC5G1T

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
- connection:
  - eth0 of NAS is connected to a router, PC is also connected to the same router
  - eth2 (QNA-UC5G1T) is connected to three different routers with DHCP lan.
- cable category Cat.7

### Output of ``dmesg`` command

````
[3020202.296623] Module [aqc111] is removed.
[3020202.300844] usbcore: deregistering interface driver aqc111
[3020202.306695] aqc111 2-1:1.0 eth2: unregister 'aqc111' usb-f10f0000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[3020202.357846] Module [usbnet] is removed.
[3020202.368165] Module [mii] is removed.
[3020216.304007] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-f10f0000.usb3-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:53:9f:ae
[3020216.316005] usbcore: registered new interface driver aqc111
[3020216.851199] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f0000.usb3) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 0AI18104)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
  |__3-1         051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS RS 550G FW:857.L7 .I USB FW:L7   4B2007P37173  )
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
|__usb5          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd f1058000.usb) hub
````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 00:11:32:6B:7B:FD
          inet addr:192.168.11.6  Bcast:192.168.11.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe6b:7bfd/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:949631704 errors:0 dropped:0 overruns:0 frame:0
          TX packets:522043090 errors:0 dropped:130 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:3375636135 (3.1 GiB)  TX bytes:4871150096 (4.5 GiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:6B:7B:FD
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:651870655 errors:0 dropped:0 overruns:0 frame:0
          TX packets:435214821 errors:0 dropped:130 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:863270749 (823.2 MiB)  TX bytes:1560452824 (1.4 GiB)
          Interrupt:193

eth1      Link encap:Ethernet  HWaddr 00:11:32:6B:7B:FD
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:297761049 errors:0 dropped:0 overruns:0 frame:0
          TX packets:86828269 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:2512365386 (2.3 GiB)  TX bytes:3310697272 (3.0 GiB)
          Interrupt:194

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:53:9F:AE
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

gzm55, 2022-06-10:
@bb-qq how can i debug this issue?

bb-qq, 2022-06-14:
When the dongle is connected to a PC, does it work correctly as expected? I would like to confirm this is not auto-negotiation or other issues of ethernet connectivity.

gzm55, 2022-06-14:
> When the dongle is connected to a PC, does it work correctly as expected? I would like to confirm this is not auto-negotiation or other issues of ethernet connectivity.

Using on my macbook, due to the mac driver problem, it can be detected and work as a 1G NIC, and connect to the switch.

When used on the NAS, I tried the different ports, such as a fixed 1G, a fixed 100M, an auto-nego 10/5/2.5G port, and get the same problem.

bb-qq, 2022-08-01:
I have no idea what is causing the problem. Perhaps the device is underpowered and a self-powered USB hub to which an AC adapter can be connected might help, but I do not have confidence.

bb-qq, 2022-08-11:
The latest driver might improve the situation since it includes some minor changes.