bb-qq/aqc111 #76: capped at 300 Mbits/sec with DSM 7.1+ QNA-UC5G1T

ahmadnassri, 2022-07-23:
### Description of the problem

brand new DS620slim, installed driver successfully, only getting ~300 Mbits/sec

### Description of your products

> - `DS620slim`
>     - `Linux Synology 4.4.180+ #42661 SMP Mon Jun 27 15:05:48 CST 2022 x86_64 GNU/Linux synology_apollolake_620slim`
>     - DSM 7.1-42661 Update 3
> - QNA-UC5G1T by QNAP

### Description of your environment

> - Synology -> Unifi UDM Pro -> PC
>     - PC: Asus XG-C100C 10G Network Adapter PCI-E x4 Card 
>     - Hub: UDM Pro
> - cable category: Cat.6e

### Output of ``dmesg`` command

<details>
<summary>Details</summary>

```

[   19.170583] md/raid1:md0: active with 5 out of 6 mirrors
[   19.197967] usb 1-3: new high-speed USB device number 2 using xhci_hcd
[   19.339235] usb 1-3: not running at top speed
[   19.712969] usb 1-3: new high-speed USB device number 2 using xhci_hcd
[   19.854249] usb 1-3: not running at top speed
[   20.227991] usb 1-3: new high-speed USB device number 2 using xhci_hcd
[   20.369269] usb 1-3: not running at top speed
[   20.393847] md0: detected capacity change from 0 to 2549940224
[   20.503001] usb 1-4: new high-speed USB device number 3 using xhci_hcd
[   22.109311] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   22.326482] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   24.540284] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   24.571784] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   24.780819] loop: module loaded
[   24.866247] EXT4-fs (loop0): couldn't mount as ext3 due to feature incompatibilities
[   24.963938] EXT4-fs (loop0): mounted filesystem with ordered data mode. Opts: (null)
[   29.734187] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   29.785752] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   30.280653] EXT4-fs (md0): couldn't mount as ext3 due to feature incompatibilities
[   30.335627] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: barrier=1
[   30.828775] systemd[1]: systemd 219 running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS -ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN)
[   30.849218] systemd[1]: Detected architecture x86-64.
[   30.865502] systemd[1]: Set hostname to <Synology>.
[   32.484370] systemd[1]: Cannot add dependency job for unit synoperfeventd.service, ignoring: Unit synoperfeventd.service is masked.
[   32.497891] systemd[1]: Cannot add dependency job for unit pkg-MailServer_dhparam.service, ignoring: Unit pkg-MailServer_dhparam.service failed to load: No such file or directory.
[   32.515858] systemd[1]: Cannot add dependency job for unit pkg-install-high-priority-builtin.service, ignoring: Unit pkg-install-high-priority-builtin.service failed to load: No such file or directory.
[   32.535916] systemd[1]: Cannot add dependency job for unit pkgctl-dsm6-ui-compatible.service, ignoring: Unit pkgctl-dsm6-ui-compatible.service failed to load: No such file or directory.
[   32.554432] systemd[1]: Cannot add dependency job for unit pkgctl-ExifTool.service, ignoring: Unit pkgctl-ExifTool.service failed to load: No such file or directory.
[   32.570986] systemd[1]: Cannot add dependency job for unit pkgctl-SynologyMoments.service, ignoring: Unit pkgctl-SynologyMoments.service failed to load: No such file or directory.
[   32.588883] systemd[1]: Cannot add dependency job for unit pkg-install-low-priority-builtin.service, ignoring: Unit pkg-install-low-priority-builtin.service failed to load: No such file or directory.
[   32.608714] systemd[1]: Cannot add dependency job for unit pkgctl-PhotoStation.service, ignoring: Unit pkgctl-PhotoStation.service failed to load: No such file or directory.
[   32.626034] systemd[1]: Cannot add dependency job for unit pkgctl-Node.js_v8.service, ignoring: Unit pkgctl-Node.js_v8.service failed to load: No such file or directory.
[   32.644929] systemd[1]: Reached target Paths.
[   32.653806] systemd[1]: Starting Paths.
[   32.658135] systemd[1]: Reached target Timers.
[   32.667190] systemd[1]: Starting Timers.
[   32.671942] systemd[1]: Created slice Root Slice.
[   32.681595] systemd[1]: Starting Root Slice.
[   32.686527] systemd[1]: Created slice ActiveBackup-GSuite's slice.
[   32.699466] systemd[1]: Starting ActiveBackup-GSuite's slice.
[   32.706014] systemd[1]: Created slice Package Center Slice.
[   32.717596] systemd[1]: Starting Package Center Slice.
[   32.723469] systemd[1]: Created slice BTRFS Optimization.
[   32.734650] systemd[1]: Starting BTRFS Optimization.
[   32.740324] systemd[1]: Created slice HybridShare's slice.
[... 880 lines cut ...]
[SUM]   6.00-7.00   sec  42.0 MBytes   352 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   7.00-8.00   sec  21.1 MBytes   177 Mbits/sec    0    219 KBytes
[  7]   7.00-8.00   sec  21.2 MBytes   178 Mbits/sec    0    221 KBytes
[SUM]   7.00-8.00   sec  42.3 MBytes   355 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   8.00-9.00   sec  20.9 MBytes   175 Mbits/sec    0    219 KBytes
[  7]   8.00-9.00   sec  21.0 MBytes   176 Mbits/sec    0    221 KBytes
[SUM]   8.00-9.00   sec  41.9 MBytes   352 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[  5]   9.00-10.00  sec  21.1 MBytes   177 Mbits/sec    0    219 KBytes
[  7]   9.00-10.00  sec  21.1 MBytes   177 Mbits/sec    0    221 KBytes
[SUM]   9.00-10.00  sec  42.3 MBytes   354 Mbits/sec    0
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   209 MBytes   175 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   208 MBytes   175 Mbits/sec                  receiver
[  7]   0.00-10.00  sec   213 MBytes   179 Mbits/sec    0             sender
[  7]   0.00-10.00  sec   213 MBytes   179 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec   422 MBytes   354 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec   421 MBytes   353 Mbits/sec                  receiver

iperf Done.
```
</details>

bb-qq, 2022-07-30:
Could you try this command?:  `sudo sysctl -w vm.min_free_kbytes=524288`
And, could you please check if the throughput changes when Synology is directly connected(without the switch) to a PC?

wysuperfly, 2023-08-22:
I have the same problem.  It shows "Link Speed 1000, USB 2" , i don't know why it is not USB 3

bb-qq, 2023-09-10:
Regarding the issue of being recognized as a USB 2.0 device, I updated [the troubleshooting page](https://github.com/bb-qq/r8152/wiki/Troubleshooting#check-the-protocol-version-of-the-usb-connection). Check it, please.

bb-qq, 2024-06-15:
The cause is that the device is not recognized by USB 3.0. Please refer to [the page](https://github.com/bb-qq/r8152/wiki/Troubleshooting#check-the-protocol-version-of-the-usb-connection) for a solution.

Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

closed 2024-06-15 by bb-qq