bb-qq/aqc111 #77: 1.3.3.0-6 DSM-7.0 System failed to start [AQC111 driver].

sparkplug37, 2022-07-31:
### Description of the problem

On a DS918+ Appollolake version 1.3.3.0-6 DSM-7.0 of the driver refuses to run after installation. After installation, I run the driver, and I get the error 'System failed to start [AQC111 driver].'. 

I can uninstall 1.3.3.0-6 and install 1.3.3.0-5 without issue. The old driver works as expected. 

I rebooted the NAS and reinstalled it from scratch without luck.

### Description of your products

DS918+
Linux DS918 4.4.180+ #42661 SMP Mon Jun 27 15:05:48 CST 2022 x86_64 GNU/Linux synology_apollolake_918+
DSM 7.1-42661 Update 3
Sebrent USB 5G NT-ss5G

### Description of your environment

Connected directly to switch
Aquantia AQtion
Netgear 10 switch XSS508M
Cat6

### Output of ``dmesg`` command

dmesg too long for a comment.

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008A2CABE41617)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         2eca:c101:0101 00  3.20 5000MBit/s 896mA 1IF  (Aquantia Aquantia 5G USB Ethernet Adapter 000000E1)

### Output of ``ifconfig -a`` command

docker0   Link encap:Ethernet  HWaddr 02:42:39:22:BF:DF  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:39ff:fe22:bfdf/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:84699 errors:0 dropped:0 overruns:0 frame:0
          TX packets:83753 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4480269 (4.2 MiB)  TX bytes:116040299 (110.6 MiB)

docker821 Link encap:Ethernet  HWaddr 1E:E5:66:F6:87:5B  
          inet6 addr: fe80::1ce5:66ff:fef6:875b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:91 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:11364 (11.0 KiB)

docker2d4 Link encap:Ethernet  HWaddr CE:36:94:0B:46:0A  
          inet6 addr: fe80::cc36:94ff:fe0b:460a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:27 errors:0 dropped:0 overruns:0 frame:0
          TX packets:123 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1634 (1.5 KiB)  TX bytes:13266 (12.9 KiB)

docker33c Link encap:Ethernet  HWaddr 9E:32:13:BA:BE:28  
          inet6 addr: fe80::9c32:13ff:feba:be28/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:84725 errors:0 dropped:0 overruns:0 frame:0
          TX packets:83778 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:5667483 (5.4 MiB)  TX bytes:116041825 (110.6 MiB)

docker69a Link encap:Ethernet  HWaddr DA:55:B5:0E:11:B9  
          inet6 addr: fe80::d855:b5ff:fe0e:11b9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:91 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:11364 (11.0 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AD  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:518271 errors:0 dropped:0 overruns:0 frame:0
          TX packets:320459 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:549262638 (523.8 MiB)  TX bytes:38926907 (37.1 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AE  
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:24:27:88:36:C1  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:16845 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16845 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:2129303 (2.0 MiB)  TX bytes:2129303 (2.0 MiB)

ovs-syste Link encap:Ethernet  HWaddr 4A:BF:85:F5:46:91  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AD  
          inet addr:172.16.1.25  Bcast:172.16.7.255  Mask:255.255.248.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:390059 errors:0 dropped:0 overruns:0 frame:0
          TX packets:316279 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:535347160 (510.5 MiB)  TX bytes:38651027 (36.8 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AE  
          inet addr:169.254.18.190  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:45.74.0.41  P-t-P:45.74.0.41  Mask:255.255.255.224
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:100250 errors:0 dropped:0 overruns:0 frame:0
          TX packets:100010 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:125192104 (119.3 MiB)  TX bytes:5567051 (5.3 MiB)

bb-qq, 2022-07-31:
Is your issue fixed by this test driver?
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-6/aqc111-apollolake-1.3.3.0-7.spk

sparkplug37, 2022-07-31:
Well, the outcome is the same in that it fails to run, but I do not get the error it failed to run pop up. In Log Center I can still see the error "System failed to start [AQC111 driver]."

bb-qq, 2022-07-31:
Could attach the output of `dmesg` as a file?

sparkplug37, 2022-07-31:
[dmesg.txt](https://github.com/bb-qq/aqc111/files/9228888/dmesg.txt)
file uploaded.

bb-qq, 2022-07-31:
Thank you.

Hmm, it seems the driver runs correctly...
Could you try another command to debug?

1. Run `sudo vi /var/packages/aqc111/scripts/start-stop-status`
2. In the editor, change `set -eu` to `set -eux` and save.
3. Run `sudo /var/packages/aqc111/scripts/start-stop-status start` and attach the output.

sparkplug37, 2022-07-31:
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ test -e /var/packages/aqc111/scripts/apply-memory-setting
+ sh /var/packages/aqc111/scripts/apply-memory-setting
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ wc -l
++ grep /usb
+ initial_count=1
+ test -e /var/packages/aqc111/target/aqc111/mii.ko
+ /sbin/insmod /var/packages/aqc111/target/aqc111/mii.ko
insmod: ERROR: could not insert module /var/packages/aqc111/target/aqc111/mii.ko: File exists

bb-qq, 2022-08-01:
Thanks for your help.
Hmm, it seems that there are still some modules that have been loaded previously...

Next, could you try these steps?

1. Install the revised version: https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-6/aqc111-apollolake-1.3.3.0-7_2.spk
2. Reboot the system
3. If the driver does not work, attach the output again of this command: `sudo /var/packages/aqc111/scripts/start-stop-status reload `

sparkplug37, 2022-08-01:
Still no joy so here is the output as requested, thanks.

[output.txt](https://github.com/bb-qq/aqc111/files/9231304/output.txt)

sparkplug37, 2022-08-01:
So oddly, it's showing up in the control panel as an interface, and I can ping it from elsewhere, but before, it did not. Sorry I had not noticed this difference. In the package centre, it still shows as not running and lists repair or uninstall for the package.

bb-qq, 2022-08-01:
Hmm....

Does the driver still work with this command?

1. Run `sudo /var/packages/aqc111/scripts/start-stop-status stop`
2. Run `sudo /var/packages/aqc111/scripts/start-stop-status start`

If so, could you try this?

1. Run `sudo /var/packages/aqc111/scripts/start-stop-status stop`
2. Push Run button of AQC111 Driver in the package center

sparkplug37, 2022-08-01:
So the stop command stops the driver, and my ping fails. The start does not start the driver, and this is the output:
`Dave@DS918:~$ sudo /var/packages/aqc111/scripts/start-stop-status start
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ '[' -e /var/packages/aqc111/scripts/apply-memory-setting ']'
+ sh /var/packages/aqc111/scripts/apply-memory-setting
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=1
+ '[' -e /var/packages/aqc111/target/aqc111/mii.ko ']'
+ /sbin/insmod /var/packages/aqc111/target/aqc111/mii.ko
insmod: ERROR: could not insert module /var/packages/aqc111/target/aqc111/mii.ko: File exists`

The reload command does get it running again, though.

Once stopped, the package centre still only gives me the option to repair or uninstall and not to run it. I have quit package centre and this caused it to say getting status, but this still only gave me the repair and uninstall options.

bb-qq, 2022-08-01:
It seems "stop" scirpt did not unload modules correctly.
Could you please post the output of this stop command as well?
`sudo /var/packages/aqc111/scripts/start-stop-status stop`

sparkplug37, 2022-08-01:
sudo /var/packages/aqc111/scripts/start-stop-status stop 
Output:

++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='docker0
docker0238f36
docker8b2c1b5
dockerbd01079
dockerddf8c3e
eth0
eth1
eth2
lo
ovs_eth0
ovs_eth1
ovs_eth2
ovs-system
sit0
syno_ovs_bonds
tun0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ docker0 = \l\o ]]
+ [[ docker0 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ docker0238f36 = \l\o ]]
+ [[ docker0238f36 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ docker8b2c1b5 = \l\o ]]
+ [[ docker8b2c1b5 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ dockerbd01079 = \l\o ]]
+ [[ dockerbd01079 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ dockerddf8c3e = \l\o ]]
+ [[ dockerddf8c3e == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ eth0 = \l\o ]]
+ [[ eth0 == \d\o\c\k\e\r* ]]
+ [[ eth0 == \o\v\s* ]]
+ [[ eth0 == \s\y\n\o* ]]
+ [[ eth0 == \s\i\t* ]]
+ [[ eth0 == \t\a\p* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
+ interface_has_target_driver=false
++ echo 'lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep aqc111
+ '[' '!' -z '' ']'
+ '[' false = true ']'
+ for interface_name in ${all_interface_names}
+ [[ eth1 = \l\o ]]
[... 89 lines cut ...]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ sit0 = \l\o ]]
+ [[ sit0 == \d\o\c\k\e\r* ]]
+ [[ sit0 == \o\v\s* ]]
+ [[ sit0 == \s\y\n\o* ]]
+ [[ sit0 == \s\i\t* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ syno_ovs_bonds = \l\o ]]
+ [[ syno_ovs_bonds == \d\o\c\k\e\r* ]]
+ [[ syno_ovs_bonds == \o\v\s* ]]
+ [[ syno_ovs_bonds == \s\y\n\o* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ tun0 = \l\o ]]
+ [[ tun0 == \d\o\c\k\e\r* ]]
+ [[ tun0 == \o\v\s* ]]
+ [[ tun0 == \s\y\n\o* ]]
+ [[ tun0 == \s\i\t* ]]
+ [[ tun0 == \t\a\p* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/tun0/device/driver
ls: cannot access '/sys/class/net/tun0/device/driver': No such file or directory
+ driver_location=

bb-qq, 2022-08-01:
> ls: cannot access '/sys/class/net/tun0/device/driver': No such file or directory

I added a workaround for this error to this driver.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-6/aqc111-apollolake-1.3.3.0-7_4.spk

Could you paste the output of `sudo /var/packages/aqc111/scripts/start-stop-status stop` again?

sparkplug37, 2022-08-01:
On installation the driver still failed but here is the output:

++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='docker0
docker0238f36
docker8b2c1b5
dockerbd01079
dockerddf8c3e
eth0
eth1
lo
ovs_eth0
ovs_eth1
ovs-system
sit0
syno_ovs_bonds
tun0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ docker0 = \l\o ]]
+ [[ docker0 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ docker0238f36 = \l\o ]]
+ [[ docker0238f36 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ docker8b2c1b5 = \l\o ]]
+ [[ docker8b2c1b5 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ dockerbd01079 = \l\o ]]
+ [[ dockerbd01079 == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ dockerddf8c3e = \l\o ]]
+ [[ dockerddf8c3e == \d\o\c\k\e\r* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ eth0 = \l\o ]]
+ [[ eth0 == \d\o\c\k\e\r* ]]
+ [[ eth0 == \o\v\s* ]]
+ [[ eth0 == \s\y\n\o* ]]
+ [[ eth0 == \s\i\t* ]]
+ [[ eth0 == \t\a\p* ]]
+ [[ eth0 == \t\u\n* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/eth0/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
+ interface_has_target_driver=false
++ echo 'lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep aqc111
+ '[' '!' -z '' ']'
+ '[' false = true ']'
+ for interface_name in ${all_interface_names}
+ [[ eth1 = \l\o ]]
+ [[ eth1 == \d\o\c\k\e\r* ]]
+ [[ eth1 == \o\v\s* ]]
+ [[ eth1 == \s\y\n\o* ]]
+ [[ eth1 == \s\i\t* ]]
+ [[ eth1 == \t\a\p* ]]
+ [[ eth1 == \t\u\n* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/eth1/device/driver
+ driver_location='lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
+ interface_has_target_driver=false
++ echo 'lrwxrwxrwx 1 root root 0 Aug  1 07:00 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/igb'
++ grep aqc111
+ '[' '!' -z '' ']'
+ '[' false = true ']'
+ for interface_name in ${all_interface_names}
+ [[ lo = \l\o ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ ovs_eth0 = \l\o ]]
+ [[ ovs_eth0 == \d\o\c\k\e\r* ]]
+ [[ ovs_eth0 == \o\v\s* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ ovs_eth1 = \l\o ]]
+ [[ ovs_eth1 == \d\o\c\k\e\r* ]]
+ [[ ovs_eth1 == \o\v\s* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ ovs-system = \l\o ]]
+ [[ ovs-system == \d\o\c\k\e\r* ]]
+ [[ ovs-system == \o\v\s* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ sit0 = \l\o ]]
+ [[ sit0 == \d\o\c\k\e\r* ]]
+ [[ sit0 == \o\v\s* ]]
+ [[ sit0 == \s\y\n\o* ]]
+ [[ sit0 == \s\i\t* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ syno_ovs_bonds = \l\o ]]
+ [[ syno_ovs_bonds == \d\o\c\k\e\r* ]]
+ [[ syno_ovs_bonds == \o\v\s* ]]
+ [[ syno_ovs_bonds == \s\y\n\o* ]]
+ continue
+ for interface_name in ${all_interface_names}
+ [[ tun0 = \l\o ]]
+ [[ tun0 == \d\o\c\k\e\r* ]]
+ [[ tun0 == \o\v\s* ]]
+ [[ tun0 == \s\y\n\o* ]]
+ [[ tun0 == \s\i\t* ]]
+ [[ tun0 == \t\a\p* ]]
+ [[ tun0 == \t\u\n* ]]
+ continue
+ /sbin/rmmod /var/packages/aqc111/target/aqc111/aqc111.ko
rmmod: ERROR: Module aqc111 is not currently loaded

BTW Afterwards, I did the reload command but the interface did not come up this time.

sparkplug37, 2022-08-01:
Interesting, using the control panel I clicked the repair option and now it says its running and a few seconds later I have a working interface.

bb-qq, 2022-08-01:
That was good. If you uninstall the package and install it again, does it work ok? If no problem, I will officially release this version.

sparkplug37, 2022-08-01:
Uninstalled and manually installed the package, and it worked as expected.

I then tested the start and stop functionality in control panel and these worked as expected also.

I'm getting approx 2.3Gbits/sec using iPerf3, which is about the same as the previous working driver.

Many thanks

closed 2022-08-01 by sparkplug37

andmar197264, 2022-08-02:
Hello ! On my DS720+ (aqc111-geminilake-1.3.3.0-7.spk ) I have exactly the same type of error as the person above (Does not start reporting an error (
Output of lsusb command

Synology_NAS:~$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00911F20345A74)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 09I03493)

ifconfig -a
 ifconfig -a
bond0     Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          inet addr:192.168.1.80  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fec9:6ed8/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:38847 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24530 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:28298169 (26.9 MiB)  TX bytes:3113420 (2.9 MiB)

docker0   Link encap:Ethernet  HWaddr 02:42:25:22:F8:DE
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:25ff:fe22:f8de/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

docker221 Link encap:Ethernet  HWaddr 92:0E:66:4C:03:6B
          inet6 addr: fe80::900e:66ff:fe4c:36b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:876 (876.0 B)

docker705 Link encap:Ethernet  HWaddr A6:53:00:B9:F5:C7
          inet6 addr: fe80::a453:ff:feb9:f5c7/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:438 (438.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:105 errors:0 dropped:0 overruns:0 frame:0
          TX packets:211 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6300 (6.1 KiB)  TX bytes:40809 (39.8 KiB)
          Interrupt:97 base 0xe000

eth1      Link encap:Ethernet  HWaddr 00:11:32:C9:6E:D8
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:38742 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24319 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:28291869 (26.9 MiB)  TX bytes:3072611 (2.9 MiB)
          Interrupt:96 base 0xc000

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:52:5C:AA
          inet addr:192.168.1.200  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe52:5caa/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:4751 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2333 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1663348 (1.5 MiB)  TX bytes:5107576 (4.8 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:22328 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22328 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2531054 (2.4 MiB)  TX bytes:2531054 (2.4 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

andmar197264, 2022-08-02:
Yes, of course - I need a couple of minutes!

andmar197264, 2022-08-02:
ynology_NAS:~$ sudo /var/packages/aqc111/scripts/start-stop-status stop
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='bond0
bonding_masters
docker0
docker3c7123a
dockerf3bb143
eth0
eth1
eth2
lo
sit0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ bond0 = \l\o ]]
+ [[ bond0 == \d\o\c\k\e\r* ]]
+ [[ bond0 == \o\v\s* ]]
+ [[ bond0 == \s\y\n\o* ]]
+ [[ bond0 == \s\i\t* ]]
+ [[ bond0 == \t\a\p* ]]
+ [[ bond0 == \t\u\n* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/bond0/device/driver
ls: cannot access '/sys/class/net/bond0/device/driver': No such file or directory
+ driver_location=

andmar197264, 2022-08-02:
ynology_NAS:~$ sudo /var/packages/aqc111/scripts/start-stop-status start
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ case $1 in
+ '[' -e /var/packages/aqc111/scripts/apply-memory-setting ']'
+ sh /var/packages/aqc111/scripts/apply-memory-setting
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=1
+ '[' -e /var/packages/aqc111/target/aqc111/mii.ko ']'
+ /sbin/insmod /var/packages/aqc111/target/aqc111/mii.ko
insmod: ERROR: could not insert module /var/packages/aqc111/target/aqc111/mii.ko: File exists

andmar197264, 2022-08-02:
aqc111-geminilake-1.3.3.0-5.spk

works as a Swiss watch)

andmar197264, 2022-08-02:
Возможно что-то сделать с этой проблемой ? Или все же оставаться на 5 релизе ? Спасибо !

bb-qq, 2022-08-02:
Does this package solve your issue? 
Note: Reboot is required after installation.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-7/aqc111-geminilake-1.3.3.0-8pre.spk

andmar197264, 2022-08-02:
can be installed over version 5?  Or complete removal?  Thanks

bb-qq, 2022-08-02:
> can be installed over version 

Yes.

andmar197264, 2022-08-02:
I'm in a hurry to please you with good news) After installing over version 5, your version 8 installed without problems and works stably! But haven't rebooted DSM yet (

andmar197264, 2022-08-02:
after the reboot - it started and works normally))) Thank you so much for your work! You are smart!!!