bb-qq/aqc111 #79: Gathering compatibility/stability reports (aqc111)

bb-qq, 2022-08-11:
(This issue has the same purpose as [the issue in r8152](https://github.com/bb-qq/r8152/issues/135). Even though the hardware is different, the report will be helpful to you.)

If you can help me create the matrix, **reply to this issue using following the template below**. *For bug reports and troubleshooting questions, please create a separate ticket* to get support.

If you have previously reported it, it would be appreciated if you could report it anew again. We would especially like to know how stable it is running in the latest environment.

## Template

> * Model(platform): `e.g. DS918+ (apollolake)`
> * DSM version:  `e.g. 7.0-41890`
> * Driver version:  `e.g. 1.3.3.0-8`
> * Ethernet device: `e.g. QNAP QNA-UC5G1T`
> * Working status: `e.g. Stable`
>   * (If you have done any special testing, please write the details here.)
> * Maximum Throughput: `e.g. 3.34 Gbps @ MTU 9000 in iperf3, 434MB/s @ MTU 9000 in SMB3`

### Working status

To determine the working status, please use the following example as a basis. 

* **Not work**
  * Failed to install the driver or the device is not recognized correctly
* **Work**
  * The devices seem to work correctly
  * No major issue with large file transfer(at least 5min)
* **Stable**
  * No major issue over 2 weeks
  * No major issue with large file transfer(at least 30min, speed degradation due to thermal throttling may occur)
* **Unstable**
  * Disconnection or other major issues are observed under long-term use or high load condition

It will help other users if you write detailed information such as what kind of tests you did and what results were observed, and how long the product has been in use.

YVES59, 2022-09-01:
Model (platform): DS418play (apollolake)
DSM version: 7.1.1-42951
Driver version: 1.3.3.0-8
Ethernet device: Sabrent  NT-SS5G
Working status: stable
        
Work: the device seems to work correctly
No major issue with large file transfer (at least 5min), MTUs are at 9000
Stable, no major issue over 1 week
No major issue with large file transfer (at least 30min, speed degradation due to thermal throttling does not occur) at MTU 9000

BUT 
- device is very, very hot (no throttle however)
- after a Synology NAS reboot, the network connection is lost, mandatory to go back to automatic DNS parameters, then switch to my static address, network  (I did not check the efficacy of stopping/starting the driver)

Many thanks to our favorite developer, bb-qq

Greetings from France to Japan and, of course, elsewhere 😊👌

YVES59, 2022-09-01:
stopping/starting the driver ----> no restore network connection 🫥

bdruth, 2022-09-04:
* Model (platform): DS920+ (geminilake)
* DSM version: 7.1-42661 Update 4
* Driver version: 1.3.3.0-8
* Ethernet device: Sabrent NT-SS5G
* Working status: Unstable

Seems to disconnect indeterminately ... I've setup an unbind/bind task to re-attach it daily (6am), but even then, today, it was disconnected when I went to use it around 2pm. When it's disconnected, it has a self-assigned IP, other than that, I just see the `kevent 2 may have been dropped` in `dmesg` output.

lankynibbs, 2022-10-08:
Model: DS1019+ (Apollolake)
DSM 7.1-42661 Update 4
Driver version: 1.3.3.0-8
Ethernet device: TRENDNet TUC-ET5G
Working Status: Unstable

Using a Sokiwi powered hub to attach USB-C cable.  Plugged into rear USB of the NAS.  Connected to a Netgear GS110EMX switch, MTU 9000.  I have a case fan blowing air through the unit to help cool it.  For a week, it was able to transfer 430MBs SMB3 traffic to my PC (my test file is a 13gig video).  Now SMB transfers max out at 220MBs.  I have turned off thermal throttling and low power mode.  Seems to be running at 2.5g speeds, even though the NAS reports it as a 5000 Mbps, full duplex interface.  Disconnecting the network cable sometimes crashes DSM!  (Have to login again)  I can edit video over this interface... that seems less intensive than a file transfer at full speed.  When the unit gets hot, it will slow down or stop.

>iperf3 -c 192.168.3.9
Connecting to host 192.168.3.9, port 5201
[  4] local 192.168.3.8 port 55571 connected to 192.168.3.9 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   134 MBytes  1.12 Gbits/sec
[  4]   1.00-2.00   sec   100 MBytes   838 Mbits/sec
[  4]   2.00-3.00   sec   163 MBytes  1.37 Gbits/sec
[  4]   3.00-4.00   sec   150 MBytes  1.26 Gbits/sec
[  4]   4.00-5.00   sec   149 MBytes  1.26 Gbits/sec
[  4]   5.00-6.01   sec   106 MBytes   888 Mbits/sec
[  4]   6.01-7.00   sec   134 MBytes  1.13 Gbits/sec
[  4]   7.00-8.00   sec   138 MBytes  1.16 Gbits/sec
[  4]   8.00-9.00   sec   142 MBytes  1.19 Gbits/sec
[  4]   9.00-10.00  sec   143 MBytes  1.20 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.33 GBytes  1.14 Gbits/sec                  sender
[  4]   0.00-10.00  sec  1.33 GBytes  1.14 Gbits/sec                  receiver

iperf Done.

Thanks for making this synology package, it is a shame the unit can get overheated!

lartiguebe, 2022-10-24:
Just bought the Sabrent but on your release page I don't see any R1000 architecture.
I have the new DS1522+, do you think you can generate the good spk and I can test it ? Thanks !

ofbarea, 2022-11-27:
Model(platform): DS220+ (geminilake)
DSM version: 7.1.1-42962
Driver version: 1.3.3.0-10
Ethernet device: SABRENT NT-SS5G
Working status: Stable, MUST use the front USB port and need to enable Low-Power mode
Maximum Throughput: 101.0 MB/s when copy very large files using FastCopy @ MTU 1500 in SMB3. 2.5gbps switch port.

Note: NAS has a couple of 6TB WD Red SMR.

leonnite, 2023-02-10:
Model(platform): DS1019+ (apollolake)
DSM version:  7.1.1-42962
Driver version:  1.3.3.0-10

All tests were done with the adapter behind rear fans as they get hot, especially the Sabrent.
Sabrent appears to get hotter quicker, I suspect a better design for heat transfer to the case.
QNAP unit has a lower quality cable which caused issues, may have also been faulty.
My router's 10G ports have been tested to have less than 10G speeds so ymmv and you will likely get better speeds.

Ethernet device: QNAP QNA-UC5G1T
Working status: Stable (with after market cable)
All Tests: Thermal Throttling Off, Low Power Off
Original Cable, Rear Port: Unstable
CableCreation Cable, Rear Port: Stable
Maximum Throughput: 2.6 Gbps @ MTU 1500 in iperf3

Ethernet device: Sabrent NT-SS5G
Working status: Stable
All Tests: Thermal Throttling Off, Low Power Off
Original Cable, Rear Port: Seems Stable (short test)
CableCreation 0.5m Cable, Rear Port: Seems Stable (short test)
CableCreation 0.5m Cable, Front Port: Stable (currently in use)
Maximum Throughput: 3.0 Gbps @ MTU 1500 in iperf3, 3.2 Gbps @ MTU 1500 in iperf3 -R

When rebooting the router it fails to get an ip,
I've had to work around it with ip -o -4 addr list to match the ip then ifconfig down, ifconfig up.

sgreadly, 2023-03-19:
Model(platform): DS918+ (apollolake)
DSM version:  DSM 7.1.1-42962 Update 4
Driver version:  RTL8152/8153 v2.16.3-3
Ethernet device: Plugable USBC-E2500 2.5Gb - https://www.amazon.com.au/gp/product/B084L4JL9K
Working status: Stable

Accepted connection from 192.168.1.87, port 59111
[  5] local 192.168.1.6 port 5201 connected to 192.168.1.87 port 59112
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   205 MBytes  1.72 Gbits/sec
[  5]   1.00-2.00   sec   239 MBytes  2.00 Gbits/sec
[  5]   2.00-3.00   sec   195 MBytes  1.64 Gbits/sec
[  5]   3.00-4.00   sec   189 MBytes  1.59 Gbits/sec
[  5]   4.00-5.00   sec   187 MBytes  1.57 Gbits/sec
[  5]   5.00-6.00   sec   273 MBytes  2.29 Gbits/sec
[  5]   6.00-7.00   sec   257 MBytes  2.16 Gbits/sec
[  5]   7.00-8.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   8.00-9.00   sec   232 MBytes  1.95 Gbits/sec
[  5]   9.00-10.00  sec   209 MBytes  1.75 Gbits/sec
[  5]  10.00-10.01  sec  1.10 MBytes  1.80 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.01  sec  2.22 GBytes  1.90 Gbits/sec                  receiver

Stable with no issues other than installation needing 2 reboots. 

Reference: https://github.com/bb-qq/aqc111/issues/95

jrwhitehead, 2023-04-19:
Model(platform): DS420+ geminilake
DSM version: 7.1.1-42962 Update 4
Driver version: 1.3.3.0-11
Ethernet device: Sabrent NT-SS5G
Working status: Works
Maximum Throughput: ~3.25 Gbits/sec @ MTU 9000 in iperf3

NOTES: Only tested with low power mode enabled on NIC (Sabrent NT-SS5G) on PC.

BioTeq, 2023-05-03:
Model(platform): DS423+ geminilake
DSM version: DSM 7.1.1-42962 Update 5 (I'm planning 7.2 RC update soon)
Driver version: 1.3.3.0-11
Ethernet device: Sabrent NT-SS5G
Working status: Works and is completely stable, I've unplugged all other cables, been running for a couple of days now, no disconnections.
Maximum Throughput: max 3.2 Gbits/sec @ MTU 9000 in iperf3

NOTES: No settings changed except MTU - for this I've also opened an issue (#110) - network interface allows a max of 1500 for the MTU, no way to set 9000. I had to set the jumbo frames via the terminal and create a triggered boot task as it resets every time I restart.

jdpdata, 2023-08-20:
Model (platform): DS918+ (apollolake)
DSM version: 7.2-64570 Update 3
Driver version: 1.3.3.0-12
Ethernet device: Sabrent NT-SS5G
Working status: stable

Work: the device seems to work correctly
No major issue with large file transfer (at least 5min), MTUs are at 9000
Getting sustained 340+ MB/s via SMB, iperf 3.52 GB/s
Test machine is Asus Zenbook Space Edition with Thunderbolt 3 to 10GbE RJ45 (Model: TH-S3EA) connected via Cat 6 using SFP+ module UACC-CM-RJ45-MG plugged into UniFI USW-Aggregation.

sgreadly, 2023-11-16:
Followup to https://github.com/bb-qq/aqc111/issues/79#issuecomment-1475405133

Model(platform): DS918+ (apollolake)
DSM version: DSM 7.2.1-69057 Update 2
Driver version: RTL8152/8153 v2.17.1-1
Ethernet device: Plugable USBC-E2500 2.5Gb - https://www.amazon.com.au/gp/product/B084L4JL9K
Working status: Stable

Upgrading from 17.1.1 > 17.1.2 broke the driver install. Uninstalling & installing newer version fixed that.