bb-qq/aqc111 #80: Only 1000Mbps with QNA-UC5G1T with DS220+

rmihai81, 2022-08-24:
### Description of the problem

> Replace this line with your answer.

### Description of your products

> - DS220+
>     - and output of ``uname -a`` command
>     - DSM version 7.1
> - QNA-UC5G1T

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
>     - Hub: model name
> - cable category Cat.6 

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2022-10-09:
Could you try restarting the driver a few times and see if the results change?

Also, what happens if the dongle is connected to the PC without changing the environment such as switches and cables?

icestraw, 2023-02-24:
I have similar problem on DS220+ and aqc111

**Description of the problem**
When using iperf speed testing, random connection problems occured including 3 different scenarios:
   1. connection on 1000Mbps and last for a while (a minute or so), then it starts hanging. Should not be happened because it's negotiation speed is larger.
   2. connection larger than 1000Mbps and last for seconds, then hanging forever.
   3. when sending from nas to client pc, hanging forever as soon as iperf start.
Problems are occured randomly but there're often some pattens:
   1 mostly occured on my Another PC with Windows 11/ Ubuntu 22.
   2/3 mostly occcured on my MacBook with M1 chip, Ventura.

I have tried using a powered usb hub, adjusting low power 5G/termal control flag, swithing usb port from back to the front, switching computers, etc. Neither of them works.
The cable and the clinet should not be a considered issue since I've tried connection between two PC, and everything is OK.
Restarting the network multiple times doesn't change the result. The connection gets recovered, but the problems still exist once I'm using that connection again.
I'm directly connect my client devices to the nas. with two usb network dongles, on the nas it's aqc111, on the client pc it's r8152, so theoretically it should have 2500Mbps speed. The mtu has been set to 1500. When set to 9000 (9014 on windows) the connection is even slower (and the bug persists), that's strange.

**Description of your products**
DS220+
and output of uname -a command
Linux DS220P 4.4.180+ #42962 SMP Fri Feb 10 16:34:59 CST 2023 x86_64 GNU/Linux synology_geminilake_220+

**dmesg:**
`
[20197.502892] Module [xt_state] is removed.
[20197.525089] Module [xt_multiport] is removed.
[20197.545935] Module [xt_LOG] is removed.
[20197.586700] audit: type=1325 audit(1677186072.586:52): table=filter family=2 entries=8
[22585.050722] aqc111 2-1.1:1.0 eth2: Link Speed 2500, USB 3
[22608.598893] aqc111 2-1.1:1.0 eth2: Link Speed 2500, USB 3
[22615.085292] audit: type=1325 audit(1677188490.505:53): table=filter family=2 entries=4
[22615.100106] audit: type=1325 audit(1677188490.520:54): table=filter family=2 entries=8
[22615.199143] audit: type=1325 audit(1677188490.619:55): table=filter family=2 entries=9
[22620.446956] audit: type=1325 audit(1677188495.868:56): table=filter family=2 entries=10
[22620.462122] audit: type=1325 audit(1677188495.883:57): table=filter family=2 entries=9
[22620.488284] Module [xt_recent] is removed.
[22620.510083] Module [xt_iprange] is removed.
[22620.525843] Module [xt_limit] is removed.
[22620.541893] Module [xt_state] is removed.
[22620.564003] Module [xt_multiport] is removed.
[22620.586237] Module [xt_LOG] is removed.
[22620.627935] audit: type=1325 audit(1677188496.049:58): table=filter family=2 entries=8
[22676.986978] audit: type=1325 audit(1677188552.418:59): table=filter family=2 entries=4
[22677.001672] audit: type=1325 audit(1677188552.432:60): table=filter family=2 entries=8
[22677.102147] audit: type=1325 audit(1677188552.533:61): table=filter family=2 entries=9
[22682.339882] audit: type=1325 audit(1677188557.771:62): table=filter family=2 entries=10
[22682.354746] audit: type=1325 audit(1677188557.786:63): table=filter family=2 entries=9
[22682.380865] Module [xt_recent] is removed.
[22682.401309] Module [xt_iprange] is removed.
[22682.418174] Module [xt_limit] is removed.
[22682.433459] Module [xt_state] is removed.
[22682.455296] Module [xt_multiport] is removed.
[22682.477076] Module [xt_LOG] is removed.
[22682.518103] audit: type=1325 audit(1677188557.950:64): table=filter family=2 entries=8
[22719.428448] aqc111 2-1.1:1.0 eth2: Link Speed 2500, USB 3
[22734.697611] audit: type=1325 audit(1677188610.138:65): table=filter family=2 entries=4
[22734.712359] audit: type=1325 audit(1677188610.153:66): table=filter family=2 entries=8
[22734.812130] audit: type=1325 audit(1677188610.253:67): table=filter family=2 entries=9
[22740.051586] audit: type=1325 audit(1677188615.493:68): table=filter family=2 entries=10
[22740.066431] audit: type=1325 audit(1677188615.508:69): table=filter family=2 entries=9
[22740.092681] Module [xt_recent] is removed.
[22740.112297] Module [xt_iprange] is removed.
[22740.128301] Module [xt_limit] is removed.
[22740.143392] Module [xt_state] is removed.
[22740.165319] Module [xt_multiport] is removed.
[22740.187296] Module [xt_LOG] is removed.
[22740.229871] audit: type=1325 audit(1677188615.671:70): table=filter family=2 entries=8
[23725.596016] aqc111 2-1.1:1.0 eth2: Link Speed 2500, USB 3
[23741.041321] audit: type=1325 audit(1677189616.657:71): table=filter family=2 entries=4
[23741.056466] audit: type=1325 audit(1677189616.672:72): table=filter family=2 entries=8
[23741.154983] audit: type=1325 audit(1677189616.770:73): table=filter family=2 entries=9
[... 210 lines cut ...]
          RX bytes:9666508 (9.2 MiB)  TX bytes:6558718 (6.2 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:F4:06:AA
          inet addr:169.254.179.173  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 00:E0:4C:7E:E4:06
          inet addr:10.0.4.4  Bcast:10.0.4.255  Mask:255.255.255.0
[... cut ...]
          RX packets:259 errors:0 dropped:0 overruns:0 frame:0
          TX packets:291 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:31448 (30.7 KiB)  TX bytes:43593 (42.5 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
`

bb-qq, 2023-03-19:
If you swap the aqc111 adapter and the r8152 adapter and use aqc111 on the PC side, does it work stably?

bb-qq, 2024-06-15:
Since there has been no response for a long time, this issue is closed. Please feel free to re-open it if you have any questions.

What I wanted to confirm is now summarized on this page and I leave the link as a note.
https://github.com/bb-qq/r8152/wiki/Troubleshooting

closed 2024-06-15 by bb-qq