bb-qq/aqc111 #82: Model (platform): DS418play (apollolake)

YVES59, 2022-09-01:
Model (platform): DS418play (apollolake)
DSM version: 7.1.1-42951
Driver version: 1.3.3.0-8
Ethernet device: Sabrent  NT-SS5G
Working status: stable
        
Work: the device seems to work correctly
No major issue with large file transfer (at least 5min), MTUs are at 9000
Stable, no major issue over 1 week
No major issue with large file transfer (at least 30min, speed degradation due to thermal throttling does not occur) at MTU 9000

BUT 
- device is very, very hot (no throttle however)
- after a Synology NAS reboot, the network connection is lost, mandatory to go back to automatic DNS parameters, then switch to my static address, network  (I did not check the efficacy of stopping/starting the driver)

Many thanks to our favorite developer, bb-qq

Greetings from France to Japan and, of course, elsewhere 😊👌

_Originally posted by @YVES59 in https://github.com/bb-qq/aqc111/issues/79#issuecomment-1234353253_

YVES59, 2022-10-12:
after a Synology NAS reboot, the network connection is lost, mandatory to go back to automatic DNS parameters, then switch to my static address, network (I did not check the efficacy of stopping/starting the driver) : any new driver to solve this problem ?

Many thanks to bb-qq for helping us, as usually 👌👌

bb-qq, 2022-10-12:
The release [1.3.3.0-9](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-9) should fix your issue.

YVES59, 2022-10-12:
1.3.3.0-9 DSM-7.x (reuploaded) : TOP, a reboot is now keeping all the acq (Sabrent NT-SS5G) IP network parameters, congrats to bb-qq and, again, many thanks 👌

Greetings to Japan from France ⛩️

YVES59, 2022-10-12:
Many thanks for your very prompt response. Have a good evening, best regards

Yves LEMOINE Ph.D. Lille University Professor Emeritus

NB In the past (in the 80’s years), I used to collaborate with Japan, I managed a biotech contract with Nippon Zeon Company Ltd Kawasaky 210, I often went to Japan and enjoyed it !!!!
For your information, one of my publication linked to this collaboration with Japanese scientists is enclosed !!

De : bb-qq
Envoyé le :mercredi 12 octobre 2022 11:48
À : bb-qq/aqc111
Cc : YVES59; Mention
Objet :Re: [bb-qq/aqc111] Model (platform): DS418play (apollolake) (Issue #82)

The release 1.3.3.0-9 should fix your issue.