bb-qq/aqc111 #84: DSM 7.1.1-42962 Update 1 - DSM7-aqc111-apollolake-1.3.3.0-8.spk stopped working! Reset IP adddress

jmrickerby, 2022-10-12:
Updated the DS718+ to "DSM 7.1.1-42962 Update 1" today, and the AQC111 1.3.3.0-8 driver appeared to have stopped working.  Within DSM the driver appears to be running and the Control Panel -> Network -> Network Interface shows the configured interface, but I could not even ping the devices IP. 

**uname -a:**
`Linux Alexandria 4.4.180+ #42962 SMP Wed Sep 21 10:56:47 CST 2022 x86_64 GNU/Linux synology_apollolake_718+`

**dmesg:**
```
[19321.733918] aqc111 2-3:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:4e:29:97
[19321.746909] usbcore: registered new interface driver aqc111
[19322.040724] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[19329.596830] aqc111 2-3:1.0 eth2: Link Speed 5000, USB 3
[19329.609584] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
```

**lsusb:**
```
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-3         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 04I00545)
```

**ifconfig -a:**
```
eth2      Link encap:Ethernet  HWaddr 24:5E:BE:4E:29:97
          inet6 addr: fe80::265e:beff:fe4e:2997/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:471 errors:1 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:37573 (36.6 KiB)  TX bytes:656 (656.0 `B)
```

Ah-ha! While the DSM Network Interface shows the correct IP address, ifconfig doesn't shop a configured IPv4 address. I went back to the DSM Network Interface window, edited the network adapter and set a different IP address, then presto, back in business:
```
eth2      Link encap:Ethernet  HWaddr 24:5E:BE:4E:29:97
          inet addr:172.16.16.12  Bcast:172.16.16.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe4e:2997/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:747 errors:1 dropped:0 overruns:0 frame:0
          TX packets:23 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:66546 (64.9 KiB)  TX bytes:2256 (2.2 KiB)
```

bb-qq, 2022-10-12:
The release [1.3.3.0-9](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-9) should fix this issue.

duqcyxwd, 2022-10-21:
@jmrickerby I have same problem here but I lost my NAS UI and ssh access as well. Do you have any suggestion to recover my NAS?

jmrickerby, 2022-10-21:
Just to clarify, if you connect one of the built-in Ethernet ports (LAN interfaces) on the NAS, you are still unable to connect to the device?

You might want to download [Synology Assistant](https://www.synology.com/en-us/support/download/DS718+?version=7.1#utilities) to see if it can find the NAS on your network.

duqcyxwd, 2022-10-21:
@jmrickerby  Thank you for your reply. 

I think my original statement is not correct, I got my NAS UI and ssh connection back 40 minutes after the first installation. I am trying the release 1.3.3.0-9 to see the different. During the installation, I have same symptom, no new ssh connection and no access to NAS UI. However, my existing SSH session is still alive but with 30 seconds delay.  I have no idea what is going on in my NAS. 

I also have iperf server in my NAS and I can still using iperf cli to test my connection speed. The ssh command is just hanging there without any error message and it is using same ip as my iperf client.

jmrickerby, 2022-10-21:
Can you clarify, do you have both a built-in 1Gbps and a USB connected multi-gigabit interface connecting your NAS to you LAN switch? Both interfaces have static IP addresses? Personally, I use the NAS 1 Gbps interface for everything but SMB traffic.

duqcyxwd, 2022-10-21:
@jmrickerby no, when this happens, I have two build-in 1Gbps ethernet cable connected with link aggregate enabled. At that time, I haven't plug my ethernet adapter into NAS yet.  I believe my problem, including no access to NAS UI and slow ssh access, only happens during the installation of the driver. 

My NAS comes back after a hard reboot.  The new release 1.3.3.0-9 doesn't same to solve this installation problem as well. I have also tried the stable release [1.3.3.0-5 DSM-7.x](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-5) and installation just went smoothly. My NAS modele is DS720+ and this maybe a model specific issue. 

I finally got my adapter driver working now. It turns out I don't nee this aqc111 driver but need [r8152](https://github.com/bb-qq/r8152). Big relief!