bb-qq/aqc111 #85: When using virtual machine manager/open vSwitch, QNAP adator doesn't work as gateway

jameslashmar, 2022-11-03:
Hello :)

This has driven me crazy for some time.  I run a virtual machine running deadline to manage renders on my local network.  I usually have a cable plugged into one of the normal eth ports on my Synology 920+ as a backup, then the QNAP 5gbit adaptor in the front usb port.

No matter what i do in synology, if i try to set the QNAP as the priority over the other LAN connection, it doesn't work if i have the virtual machine manager installed which also installs open vSwitch.

If i uninstall this, all works fine.

Why is this an issue?  Well i don't have full control of the network as it is in an office.  I have ports forwarded to a specific IP that is mapped to my QNAP MAC address.  This also means that i cannot access DSM this way.  Also it means that i can only access shares via \\nasName\ rather than via IP.  Using the hostname defaults to the slower connection, even if i have 'multiple gateways' enabled.  Therefore, even though the system can see the QNAP adaptor with the driver, something about the open vSwitch just means that connections to the NAS via it, are ignored.

bb-qq, 2022-11-09:
Could you try this?
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-9/aqc111-geminilake-1.3.3.0-10test.spk

I am not sure what is going on in your environment, but at least there was a problem with the QNAP adapter not saving the settings properly when it was set as the priority default gateway, and I have taken action to address that issue.

jameslashmar, 2022-12-01:
Sorry I got caught up in a project and couldn't test this.

Is this also included in the latest build 1.3.3.0-10 DSM-7.x (reuploaded)?

Thanks.  I super appreciate your commitment to this!

jameslashmar, 2022-12-01:
I tested the build you sent and it absolutely works! This has plagued me
forever!  I'm going to stay on this build for now, but if you can let me
know if i can update to the latest and if that would break it, i'd
appreciate it.  Thanks again 🙏🏼

> Message ID: ***@***.***>
>>
>

bb-qq, 2022-12-04:
The driver I published on the release page is identical to the one I uploaded here. You should be able to update it without any problems, and there is no change after updating it at this time.

closed 2024-06-15 by bb-qq